ghdl/ghdl #741: Building with vendor libraries

kayb94, 2019-01-22:
Hi,

I'm trying to study VHDL without an FPGA board. I'm very happy, you made ghdl. I have some lecture material including code, but sadly I wasn't able to compile it yet, but I have read the entire documentation of ghdl. Obviously I need 'lpm', because the error message says:
```
$ ghdl -a *.vh*
e_lpm_count.vhd:40:9:error: cannot find resource library "lpm"
ghdl:error: compilation error
```
I installed Quartus-Lite into a VM, so /opt/altera/ is now populated and did this in the project directory:
```
$ sudo /usr/lib/ghdl/vendors/compile-altera.sh --altera
```
Compilation then still fails with the same error. I have the following object-files:
```
$ find -name '*.o'
./altera/altera/v93/altera_europa_support_lib.o
./altera/altera/v93/altera_standard_functions.o
./altera/altera/v93/altera_primitives.o
./altera/altera/v93/altera_syn_attributes.o
./altera/altera/v93/altera_primitives_components.o
./altera/altera/v93/alt_dspbuilder_package.o
./altera/lpm/v93/220model.o
./altera/lpm/v93/220pack.o
./altera/altera_mf/v93/altera_mf_components.o
./altera/altera_mf/v93/altera_mf.o
./altera/altera_lnsim/v93/altera_lnsim_components.o
./altera/sgate/v93/sgate.o
./altera/sgate/v93/sgate_pack.o
./e_hex7seg.o
```
The project directory itself looks like this:
```
$ ls
altera       e_hex7seg.vhd    e_my_LPMcounter.vhd  work-obj93.cf
e_hex7seg.o  e_lpm_count.vhd  simulation
```
I'm sure, it's just a small option to ghdl I'm missing, but I can't figure out since I'm just starting with ghdl and VHDL... I have documented what I have done so far and plan write something about it in the ArchWiki.

Would be very thankful for any help!

Best Regards
Kay

kayb94, 2019-01-22:
Okay, so it went like always: No matter how much time I have spent trying to figure something out; as soon as I write down the question somewhere and post it, I find it out myself quickly afterwards. xD

For anyone struggling after me (in the same directory as all the commands from above):
```
$ ghdl -a -P=altera/ *.vh*
$ ghdl -e -P=altera/ --ieee=synopsys e_my_LPMcounter
```
("e_my_LPMcounter" is the entity in the example project)

I often read on various pages I should use the workdir option, however that didn't "work out" for me.

Hope this helps anyone.

closed 2019-01-22 by kayb94

Paebbels, 2019-01-23:
GHDL is shipped with vendor library compile scripts including Altera/Intel. The latest set of scripts is on branch Paebbels:paebbels/reworked-windows-compile-scripts and can be seen in MR #693.