ghdl/ghdl #742: Recursive function call.

gbd360, 2019-01-26:
**Description**
Analysis of recursive function in package causes ghdl to crash.  Maybe related to issue #41  

**Expected behaviour**
Recursive function call should function properly.

**Context**
******************** GHDL Bug occurred ***************************
Please report this bug on https://github.com/ghdl/ghdl/issues
GHDL release: 0.36-dev (20181129-161-g3c30e3b4) [Dunoon edition]
Compiled with GNAT Version: 6.5.0
Target: x86_64-linux-gnu
In directory: /test/
Command line:
/opt/ghdl/bin/ghdl1-llvm -g -lv --std=08 --work=work --workdir=./build/test_work -P/opt/ghdl/lib/ghdl/ieee/v08/ -P/opt/ghdl/lib/ghdl/ -c -fpic -o ./build/test_work/test_pkg.o test_pkg.vhdl
Exception TYPES.INTERNAL_ERROR raised
Exception information:
raised TYPES.INTERNAL_ERROR : trans-chap3.adb:49
Call stack traceback locations:
0x5b56c2 0x5ba8c7 0x5bac08 0x5bb0ae 0x5c8a9b 0x53a2c3 0x53b984 0x53df72 0x52c960 0x597b94 0x5948d8 0x52cc95 0x51df9d 0x51d963 0x51da86 0x51d896 0x51da86 0x51d9ec 0x51c9fb 0x521abe 0x52cac4 0x52c108 0x5889ae 0x5a24cd 0x5a2677 0x588f9a 0x58904a 0x5a24dc 0x5a26ae 0x5cef36 0x5b19ab 0x5d02b1 0x5663f9 0x5e1e44 0x5e26df 0x44ed74 0x408688 0x7f2236ea0b95 0x407a58 0xfffffffffffffffe
******************************************************************

- OS: Ubuntu 18.04
- Origin: git hash 3c30e3b4
 

**Additional context**
Recursion in original function has been confirmed in modelsim 10.4a and riviera_pro 2017.02.

**How to reproduce?**
Simply run ghdl analysis on mwe attached.
ghdl -a -g -lv --std=08 --work=work --workdir=./build/ghdl_work test_pkg.vhdl

**Files**
test_pkg.vhdl

**Checklist**
Before submitting your issue, please review the following checklist:

- [ x] Add `GHDL Bug occurred` log block
- [ x] Add a MWE
- [ x] Try the latest version

gbd360, 2019-01-26:
[test_pkg.zip](https://github.com/ghdl/ghdl/files/2799111/test_pkg.zip)

julian-becker, 2019-09-02:
I have encountered the same or a similar issue:

> ******************** GHDL Bug occurred ***************************
Please report this bug on https://github.com/ghdl/ghdl/issues
GHDL release: 0.37-dev (v0.36-673-gbe3a72d2) [Dunoon edition]
Compiled with GNAT Version: 8.3.0
Target: x86_64-linux-gnu
/work/ModelSources/
Command line:
/usr/local/bin/ghdl1-llvm --workdir=/work/ModelSources/vunit_out/ghdl/libraries/testlib --work=testlib --std=08 -P/work/ModelSources/vunit_out/ghdl/libraries/vunit_lib/ -P/work/ModelSources/vunit_out/ghdl/libraries/osvvm/ -P/work/ModelSources/vunit_out/ghdl/libraries/testlib/ -P/usr/local/lib/ghdl/synopsys/v08/ -P/usr/local/lib/ghdl/ -c -fpic -o /work/ModelSources/vunit_out/ghdl/libraries/testlib/myentity_tb.o /work/ModelSources/myentity_tb.vhd
Exception TYPES.INTERNAL_ERROR raised
Exception information:
raised TYPES.INTERNAL_ERROR : trans-chap3.adb:49
Call stack traceback locations:
0x61015d 0x614cd4 0x614ff7 0x61544e 0x62200a 0x5bda34 0x5bdfd7 0x5be0d6 0x5c7de2 0x5b227a 0x6324eb 0x577473 0x63858f 0x639c34 0x63b4fd 0x409f99 0x7f92ddc82099 0x409178 0xfffffffffffffffe
******************************************************************

Is there any workaround for this issue?

tgingold, 2019-09-02:
No known workaround.  I have to investigate.

julian-becker, 2019-09-02:
@tgingold FYI, in my case, I could boil down the issue to the following entity which fails to compile:

    library ieee;
    use ieee.std_logic_1164.all;

    entity crashing_entity is
    end crashing_entity;

    architecture crashing_arch of crashing_entity is

      type byte_array is array (natural range <>) of std_logic_vector(7 downto 0);
      type message_array is array (natural range <>) of byte_array;
      ------------------------------------------------------------------------------
      -- the following compiles fine:
      -- type byte_array is array (natural range 0 to 1) of std_logic_vector(7 downto 0);
      -- type message_array is array (natural range 0 to 1) of byte_array;

      constant messages : message_array := (
        (X"00", X"00"), 
        (X"00", X"00")
      );

    begin

    end crashing_arch;

tgingold, 2019-09-02:
Ah thanks.  That's very useful.  In particular it is not really a function call issue!

dalex78, 2026-09-03:
# Issue #742 — internal error (`trans-chap3.adb:49`) on a nested unconstrained array constant

Analysis by an AI agent. No longer reproducible; recommend merging the regression test and closing.

## The reproducer

The report was filed as "recursive function in package crashes analysis" and attached `test_pkg.zip`. That framing turned out to be wrong.
The reduction needs no attachment and no recursion at all:

```vhdl
library ieee;
use ieee.std_logic_1164.all;

entity crashing_entity is
end crashing_entity;

architecture crashing_arch of crashing_entity is

  type byte_array is array (natural range <>) of std_logic_vector(7 downto 0);
  type message_array is array (natural range <>) of byte_array;

  constant messages : message_array := (
    (X"00", X"00"),
    (X"00", X"00")
  );

begin

end crashing_arch;
```

`message_array` is an unconstrained array whose *element* subtype is itself unconstrained, so the element bounds exist only in the aggregate. Constraining either level made it analyse, which is what julian-becker reported.

The original `test_pkg.vhdl` is the same shape reached the long way round: `t_cfixed_row_vector` is an unconstrained array of a record with two `sfixed` (unbounded) elements, and `to_string` recurses over slices of it.

Both crashed identically:

```
raised TYPES.INTERNAL_ERROR : trans-chap3.adb:49
```

## What that line is

`trans-chap3.adb:49` is inside `Get_Composite_Type_Layout`, which walks the chain that tells a subtype where its layout lives:

```ada
   function Get_Composite_Type_Layout (Info : Type_Info_Acc) return Mnode is
   begin
      if Info.S.Subtype_Owner /= null then
         ...
```

An unbounded element subtype does not own its layout — it borrows it from whatever contains it, through `Info.S.Subtype_Owner` and, for a record element, `Info.S.Owner_Field`. In the versions that crashed, a subtype created for the element of a nested unconstrained array reached this function with neither an owner nor a layout of its own, and fell through to the `Error_Kind` at the end.

## Status on master

**The crash is gone.** Against master at `47b155fc8`, both the reduction and the reporter's original `test_pkg.vhdl` analyse cleanly, and the reduction also elaborates and simulates, on mcode, llvm and the LLVM JIT backend.

It was fixed incidentally, by the rework that gave unbounded subtypes the `Subtype_Owner` / `Owner_Field` ownership chain that `Get_Composite_Type_Layout` reads today — there is no commit in the history that mentions this issue number, and the exact commit could not be pinned by bisection because the 2019-era sources no longer build with a current GNAT (real compile errors in `grt-signals.adb` and elsewhere, not just warnings).

## What this change adds

Nothing in `src/`. `testsuite/gna/issue742` pins both shapes so they cannot regress:

- `test_pkg.vhdl` — the reporter's original package, analysed as-is;
- `crash.vhdl` — julian-becker's reduction, analysed, elaborated and simulated.

Recommend closing #742 as fixed.

closed 2026-09-04 by tgingold