ghdl/ghdl #743: raised CONSTRAINT_ERROR : grt-processes.adb:770 overflow check failed

twilco, 2019-01-27:
**Description**
I encountered a bug when running my test bench.

**Expected behaviour**
The test bench runs without error.

**Context**
```
GHDL release: 0.35 (v0.35-dirty) [Dunoon edition]
Compiled with GNAT Version: GPL 2017 (20170515-63)
Target: x86_64-apple-darwin14.5.0
In directory: /Users/twilco/projects/vhdl-sandbox/led-blinker/
Command line:
ghdl -r led_blink_tb
Exception CONSTRAINT_ERROR raised
Exception information:
raised CONSTRAINT_ERROR : grt-processes.adb:770 overflow check failed
Load address: 0x10c11a000
Call stack traceback locations:
0x10c177d12 0x10c178446 0x10c1786f6 0x10c1788f0 0x10c11c858 0x10c1b2b7c 0x10c3e9dd1 0x10c3d6de4 0x10c3c99b5 0x10c3f2cd0 0x10c11c48c
```

- OS: MacOS Mojave
- Origin:
  - [x] Released binaries. Tarball: https://github.com/ghdl/ghdl/releases/download/v0.35/ghdl-0.35-mcode-macosx.tgz

**Additional context**
Add any other context about the problem here. If applicable, add screenshots to help explain your problem.

**How to reproduce?**
Tell us how to reproduce this issue. Please provide a Minimal Working Example (MWE), that is compatible with [issue-runner](https://github.com/1138-4EB/issue-runner). With sample code it's easier to reproduce the bug and it's much faster to fix it. For example:

```
#>> led_blink.vhd
library ieee;
use ieee.std_logic_1164.all;
use ieee.numeric_std.all;

entity led_blink is
  port (
    i_clock : in std_logic;
    i_enable : in std_logic;
    i_switch_1 : in std_logic;
    i_switch_2 : in std_logic;
    o_led_drive: out std_logic
  );
end led_blink;

architecture rtl of led_blink is

  -- Constants to create the needed frequencies.
  -- Formula is: (clock_rateMHz / output_freqHz * percentage_of_time_on% duty cycle)
  -- So for 100Hz with 25MHz clock: 25MHz / 100Hz * 50% duty cycle === 125000
  constant CNT_TO_100HZ : natural := 125000;
  constant CNT_TO_50HZ : natural := 250000;
  constant CNT_TO_10HZ : natural := 1250000;
  constant CNT_TO_1HZ : natural := 12500000;

  -- These signals will be the counters:
  signal cnt_100hz : natural range 0 to CNT_TO_100HZ;
  signal cnt_50hz : natural range 0 to CNT_TO_50HZ;
  signal cnt_10hz : natural range 0 to CNT_TO_10HZ;
  signal cnt_1hz : natural range 0 to CNT_TO_1HZ;

  -- These signals will toggle at the frequencies needed:
  signal toggle_100hz : std_logic := '0';
  signal toggle_50hz : std_logic := '0';
  signal toggle_10hz : std_logic := '0';
  signal toggle_1hz : std_logic := '0';

  -- One bit select wire.
  signal led_select : std_logic;

begin

  -- All processes toggle the signal at a specific frequency.
  -- They all run continuously, even if they are not the selected frequency.

  p_100hz : process (i_clock) is
  begin
    if rising_edge(i_clock) then
      if cnt_100hz = CNT_TO_100HZ - 1 then -- -1, since counter starts at 0
        toggle_100hz <= not toggle_100hz;
        cnt_100hz <= 0;
      else
        cnt_100hz <= cnt_100hz + 1;
      end if;
    end if;
  end process p_100hz;

  p_50hz : process (i_clock) is
  begin
    if rising_edge(i_clock) then
      if cnt_50hz = CNT_TO_50HZ - 1 then -- -1, since counter starts at 0
        toggle_50hz <= not toggle_50hz;
        cnt_50hz <= 0;
      else
        cnt_50hz <= cnt_50hz + 1;
      end if;
    end if;
  end process p_50hz;

  p_10hz : process (i_clock) is
  begin
    if rising_edge(i_clock) then
      if cnt_10hz = CNT_TO_10HZ - 1 then -- -1, since counter starts at 0
        toggle_10hz <= not toggle_10hz;
        cnt_10hz <= 0;
      else
        cnt_10hz <= cnt_10hz + 1;
      end if;
    end if;
  end process p_10hz;

  p_1hz : process (i_clock) is
  begin
    if rising_edge(i_clock) then
      if cnt_1hz = CNT_TO_1HZ - 1 then -- -1, since counter starts at 0
        toggle_1hz <= not toggle_1hz;
        cnt_1hz <= 0;
      else
        cnt_1hz <= cnt_1hz + 1;
      end if;
    end if;
  end process p_1hz;

  -- create multiplexor based on inputs to select correct output
  led_select <= toggle_100hz when (i_switch_1 = '0' and i_switch_2 = '0') else
                toggle_50hz when (i_switch_1 = '0' and i_switch_2 = '1') else
                toggle_10hz when (i_switch_1 = '1' and i_switch_2 = '0') else
                toggle_1hz;

  o_led_drive <= led_select and i_enable;
end rtl;

#>> led_blinker_tb.vhd
library ieee;
use ieee.std_logic_1164.all;
use ieee.numeric_std.all;

entity led_blink_tb is
end led_blink_tb;

architecture behave of led_blink_tb is

  -- 40ns is 25MHz
  constant CLOCK_PERIOD : time := 40 ns;

  signal s_clock : std_logic := '0';
  signal s_enable : std_logic := '0';
  signal s_switch_1 : std_logic := '0';
  signal s_switch_2 : std_logic := '0';
  signal s_led_drive : std_logic;

  component led_blink is
    port (
      i_clock : in std_logic;
      i_enable : in std_logic;
      i_switch_1 : in std_logic;
      i_switch_2 : in std_logic;
      o_led_drive : out std_logic
    );
  end component led_blink;

begin

  UUT : led_blink
    port map (
      i_clock => s_clock,
      i_enable => s_enable,
      i_switch_1 => s_switch_1,
      i_switch_2 => s_switch_2,
      o_led_drive => s_led_drive
    );

  clk_gen : process is
  begin
    wait for CLOCK_PERIOD / 2;
    s_clock <= not s_clock;
  end process clk_gen;

  process
  begin
    s_enable <= '1';

    s_switch_1 <= '0';
    s_switch_2 <= '0';
    wait for 0.2 sec;

    s_switch_1 <= '0';
    s_switch_2 <= '1';
    wait for 0.2 sec;

    s_switch_1 <= '1';
    s_switch_2 <= '0';
    wait for 0.2 sec;

    s_switch_1 <= '1';
    s_switch_2 <= '1';
    wait for 2 sec;
    report "finished 1 and 1";

  end process;
end behave;

#>> sim.sh
ghdl -a led_blinker.vhd
ghdl -a led_blinker_tb.vhd
ghdl --elab-run led_blinker_tb

#>> run.sh
docker run --rm -tv /$(pwd):/src:z -w //src ghdl/ghdl:buster-mcode sh -c ./sim.sh

#>> end
```

Note that `run.sh` is used to execute `sim.sh` inside a docker container. Please, put your commands in `sim.sh` and just copy `run.sh` from the example. Using `ghdl/ghdl:*` docker images to run the MWEs ensures that the latest available GHDL is used.

**Checklist**
Before submitting your issue, please review the following checklist:

- [x] Add `GHDL Bug occurred` log block
- [x] Add a MWE
- [x] Try the latest version

diogratia, 2019-01-28:
The elaboration and simulation target name should be led_blink_tb. You're not using IEEE package numeric_std. 

Seems wasteful to have four 25 MHz counters to various intervals (100 Hz, 50 Hz, 10 Hz, 1 Hz) when the all start in sync. Simulating for 1.05 sec produced a 3/4 GB  GHW waveform dump file. Consider using a pre-scalar counter to reduce the number of events occurring.

There's nothing in your integer (natural) math that would overflow which sort of leaves simulation time which is a 32 bit value for the mcode version.

The latest [release](https://github.com/ghdl/ghdl/releases/latest)  is [ghdl-20181129-macosx-mcode.tgz](https://github.com/ghdl/ghdl/releases/download/20181129/ghdl-20181129-macosx-mcode.tgz), which still uses 32 bit time.  

See ALDEC's [Time Calculations in VHDL](https://www.aldec.com/en/support/resources/documentation/articles/1165):

> IThe question we should ask is: can we use 32-bit numbers to store time values in the simulator, no matter what units are used? Let's see:  
>  
>-  An hour has 3.6x10^^18 femtoseconds.  
>  
>- The maximal value that can be expressed in 32-bit signed integers is 2^^31-1 = 2147483647 = 2.15x10^^9.  
>  
>It means that 32 bits is not enough to cover all predefined time units. We would run out of range after reaching 2.15 microseconds! That is the main reason why simulators are not using 32-bit integers to represent time. 

You're likely better off using an llvm or gcc version of ghdl, both use 64 bit values for simulation time.

Setting the resolution limit on the command line was target platform specific in ghdl-0.35 and not supported for ghdl-0.35-mcode-macosx. It is supported in the latest release with the command line argument --time-resolution=UNIT.

See IEEE Std 1076-2008 5.2.4.2 Predefined physical types, with a settable resolution limit via the command line ghdl supports "for a longer period of simulated time, with reduced accuracy". The ability to do so is implementation defined.

Setting to resolution limit to unit ns would get you to 35.788+ minutes of simulation time on an mcode version with setting the resolution limit enabled.

The latest ghdl MacOS mcode release completes with your testbench simulation stopped by stop-time (ghdl -a --time-resolution=ns for both entities VHDL files, ghdl -r led_blink_tb --stop-time=1sec).

I haven't checked to see whether the 'image  attribute returns a time value in resolution limit units for type time. (It's a requirement.)

Paebbels, 2025-04-27:
@tgingold anything to do for this one?

tgingold, 2025-04-27:
Cannot reproduce.  Probably fixed.

closed 2025-04-27 by tgingold