ghdl/ghdl #744: Precompiled LLVM version for MacOS

twilco, 2019-01-27:
Thanks for making this awesome tool!

**Is your feature request related to a problem? Please describe.**
It would be nice if there were a precompiled LLVM or GCC version of GHDL for MacOS in addition to the already available `mcode` version.  The `mcode` version seems to go quite slowly even in simple designs.  I know I can build from source...but it would be nice if I didn't have to :).  Linux and Windows both have precompiled LLVM versions.

https://ghdl.readthedocs.io/en/latest/getting/Releases.html

Arcnor, 2021-02-01:
Hi @eine , can you point me / describe what the current status of this is? If it's a matter of modifying Travis / AppVeyor / GitHub actions to build with LLVM I might be able to open a PR for it, but I'm a bit confused as to what the current status of things are.

Thanks in advance.

eine, 2021-02-01:
Hola @Arcnor!

The current status is that we do test GHDL on macOS, but only with mcode backend and with synthesis features disabled: https://github.com/ghdl/ghdl/blob/master/scripts/ci-run.sh#L400-L401. We generate a zipfile and upload it to nightly assets: https://github.com/ghdl/ghdl/releases/tag/nightly.

- In #867, synthesis features are enabled. As you can see, it fails with `ld: library not found for -lgnat-2019`: https://github.com/ghdl/ghdl/pull/867/checks?check_run_id=1805280440#step:4:896. However, users have reported building it successfully locally. Hence, it must be some issue with the setup in CI.
  - https://github.com/eine/ghdl/commits/osx-hdiutil is an attempt to use hdiutil for extracting the installer of GNAT, instead of using 7z.
- After fixing #867, it would be interesting to add an LLVM build. See issues labeled [OS: Mac OS](https://github.com/ghdl/ghdl/labels/OS%3A%20Mac%20OS) (open and closed ones). For instance #1368.
- Currently, we use a single version of macOS only (`macos-latest`, see https://github.com/actions/virtual-environments/#available-environments). It would be desirable to test on Catalina and Big Sur.
- Homebrew Formulae exists for GHDL (https://formulae.brew.sh/cask/ghdl). It might be interesting to use one of those in our CI workflow, instead of the custom shell scripts. See [ktemkin/homebrew-oss-fpga](https://github.com/ktemkin/homebrew-oss-fpga) and [hdl/packages](https://github.com/hdl/packages). For reference, Windows builds/tests are based on PKGBUILD recipes.

Anyway, the most straightforward approach for testing LLVM backend on macOS in CI would be modifying the following:

- https://github.com/ghdl/ghdl/blob/master/scripts/macosx/install-ada.sh
- https://github.com/ghdl/ghdl/blob/master/scripts/ci-run.sh#L399-L402
- https://github.com/ghdl/ghdl/blob/master/scripts/ci-run.sh#L427-L430

Then, you can open a PR and we can discuss how to adapt it for having two or more macOS jobs.

The main reason for the current status on macOS is that none of the usual maintainers/developers/contributors has both a mac and expertise in CI. Therefore, it is very cumbersome for us to try debugging these issues.

Let us know if you need any further clarification/help.

eine, 2021-02-02:
Thanks to @Arcnor, pre-built LLVM tarballs for macOS are available in pre-release nightly: https://github.com/ghdl/ghdl/releases/tag/nightly. Those are updated after each successful CI run of branch `master`.

closed 2021-02-02 by eine

tgingold, 2021-02-02:
That's great!