ghdl/ghdl #747: GHDL bug with invalid code

svtesan, 2019-02-03:
ghdl -a --workdir=work a.vhdl
a.vhdl:8:10:warning: declaration of "a" hides entity "a" [-Whide]
a.vhdl:21:7:warning: declaration of "a" hides signal "a" [-Whide]
a.vhdl:21:10:warning: declaration of "b" hides signal "b" [-Whide]
a.vhdl:21:13:warning: declaration of "ci" hides signal "ci" [-Whide]
a.vhdl:21:17:warning: declaration of "s" hides signal "s" [-Whide]
a.vhdl:21:20:warning: declaration of "co" hides signal "co" [-Whide]
a.vhdl:37:17: prefix type is not constrained
a.vhdl:38:15: conversion not allowed between not closely related types
sem_selected_name(2): cannot handle IIR_KIND_TYPE_CONVERSION (a.vhdl:38:15)

******************** GHDL Bug occurred ***************************
Please report this bug on https://github.com/ghdl/ghdl/issues
GHDL release: 0.36-dev (20181129) [Dunoon edition]
Compiled with GNAT Version: 7.3.0
Target: i686-w64-mingw32
In directory: C:\Users\user\Documents\Play\vhdl\scratch\
Command line:
c:\Users\user\Documents\Apps\GHDL\bin\ghdl.exe -a --workdir=work a.vhdl
Exception TYPES.INTERNAL_ERROR raised
Exception information:
raised TYPES.INTERNAL_ERROR : errorout.adb:39
Call stack traceback locations:
0x5d1a7b 0x5aeb5e 0x5af3e2 0x5b0eb6 0x5b0e42 0x5b10fc 0x5b1cd6 0x5b20ec 0x5b23e5 0x5deb97 0x5e4ee2 0x5a86a7 0x5f9f91 0x58ee78 0x6a0623 0x402635 0x401389 0x75a28482 0x77423ab6 0x77423a86
******************************************************************

Paebbels, 2019-02-03:
Please provide a minimal and complete code example.

svtesan, 2019-02-03:
Attached the fa.vhdl, fa_tb.vhdl files.

To reproduce:
```
ghdl -a --workdir=work fa.vhdl
ghdl -e --workdir=work fa
ghdl -a --workdir=wor
[fa.zip](https://github.com/ghdl/ghdl/files/2825502/fa.zip)
k fa_tb.vhdl

GHDL Version:
GHDL 0.36-dev (20181129) [Dunoon edition]
 Compiled with GNAT Version: 7.3.0
 mcode code generator
Written by Tristan Gingold.

Copyright (C) 2003 - 2015 Tristan Gingold.
GHDL is free software, covered by the GNU General Public License.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
On my machine, I get
fa_tb.vhdl:20:7:warning: declaration of "a" hides signal "a" [-Whide]
fa_tb.vhdl:20:10:warning: declaration of "b" hides signal "b" [-Whide]
fa_tb.vhdl:20:13:warning: declaration of "ci" hides signal "ci" [-Whide]
fa_tb.vhdl:20:17:warning: declaration of "s" hides signal "s" [-Whide]
fa_tb.vhdl:20:20:warning: declaration of "co" hides signal "co" [-Whide]
fa_tb.vhdl:35:28: prefix type is not constrained
fa_tb.vhdl:36:26: conversion not allowed between not closely related types
sem_selected_name(2): cannot handle IIR_KIND_TYPE_CONVERSION (fa_tb.vhdl:36:26)

******************** GHDL Bug occurred ***************************
Please report this bug on https://github.com/ghdl/ghdl/issues
GHDL release: 0.36-dev (20181129) [Dunoon edition]
Compiled with GNAT Version: 7.3.0
Target: i686-w64-mingw32
In directory: C:\Users\user\vhdl\scratch\
Command line:
c:\Users\user\Apps\GHDL\bin\ghdl.exe -a --workdir=work fa_tb.vhdl
Exception TYPES.INTERNAL_ERROR raised
Exception information:
raised TYPES.INTERNAL_ERROR : errorout.adb:39
Call stack traceback locations:
0x5d1a7b 0x5aeb5e 0x5af3e2 0x5b0eb6 0x5b0e42 0x5b10fc 0x5b1cd6 0x5b20ec 0x5b23e5 0x5deb97 0x5e4ee2 0x5a86a7 0x5f9f91 0x58ee78 0x6a0623 0x402635 0x401389 0x75a28482 0x77423ab6 0x77423a86
******************************************************************
```
[fa.zip](https://github.com/ghdl/ghdl/files/2825503/fa.zip)

Paebbels, 2019-02-03:
sorry, but the zip contains only a `fa.vhdl`, the `fa_tb` file is a `*.vcd` file.

You can edit you last comment, delete the attached file and attache a new one.

tgingold, 2019-02-06:
I can reproduce the crash; I am checking whether this could be valid.

tgingold, 2019-02-06:
No, it cannot.

commit 2c5f8d35 (2019-02-07): vhdl: do not crash on selected by name type conversion.

Fix #747
closed 2019-02-07 by tgingold