ghdl/ghdl #749: GHDL bug report cannot handle IIR_KIND_SUBSTRACTION_OPERATOR

TonyU1234, 2019-02-07:
**Description**
I am newbie on GHDL. After elaborate for all file involve, when I launch these commands:
ghdl -a --std=08 uart_to_spi_bridge_tb.vhd
ghdl -e --std=08 uart_to_spi_bridge_tb
I get the error: 
eval_pos: cannot handle IIR_KIND_SUBSTRACTION_OPERATOR (uart_to_spi_bridge_tb.vhd:197:87)

**Expected behaviour**
GHDL should elaborate and give back the simulation file.

**Context**
Please report this bug on http://gna.org/projects/ghdl
GHDL release: GHDL 0.33 (20150921) [Dunoon edition]
Compiled with GNAT Version: GPL 2014 (20140331)
In directory: J:\My Documents\GHDL_simulation\uart2spi\
Command line:
C:\eda\ghdl\ghdl-0.33\bin\ghdl.exe -e --std=08 MSSnum_uart_to_spi_bridge_tb
Exception TYPES.INTERNAL_ERROR raised
Exception information:
Exception name: TYPES.INTERNAL_ERROR
Message: errorout.adb:63

- OS: Win10
- Origin:
  - [ ] Package manager. Repo:
  - [x] Released binaries. Tarball: ghdl-0.33-win32.zip
  - [ ] Built from sources. Commit SHA:

**Additional context**
Add any other context about the problem here. If applicable, add screenshots to help explain your problem.

**How to reproduce?**
Tell us how to reproduce this issue. Please provide a Minimal Working Example (MWE), that is compatible with [issue-runner](https://github.com/1138-4EB/issue-runner). With sample code it's easier to reproduce the bug and it's much faster to fix it. For example:

```
#>> uart_to_spi_bridge_tb.vhd
signal adc_rx_data      	: STD_LOGIC_VECTOR(test_d_width-1 downto 0) := (others => '0'); 
signal adc_busy         	: STD_LOGIC := '0';  											
signal test_uart_tx_data	: STD_LOGIC_VECTOR(test_d_width-1 downto 0); 
signal test_uart_rx	        : STD_LOGIC;	
signal test_uart_rx_data   : STD_LOGIC_VECTOR(test_d_width-1 downto 0);

#>> sim.sh
ghdl -a --std=08 uart_to_spi_bridge_tb.vhd
ghdl -e --std=08 uart_to_spi_bridge_tb

#>> end
```

**Files**
A list of relevant files for this issue. Large files can be uploaded one-by-one or in a tarball/zipfile. See [1138-4EB/issue-runner#parser](https://github.com/1138-4EB/issue-runner#parser).
[image]

**Checklist**
Before submitting your issue, please review the following checklist:

- [x] Add `GHDL Bug occurred` log block
- [ ] Add a MWE
- [ ] Try the latest version

1138-4EB, 2019-02-07:
Relase `ghdl-0.33-win32.zip` is more than three years and a half old. Might we know what are you using that version instead of e.g. `ghdl-20181129-mingw32-mcode.zip`?

tgingold, 2019-02-07:
A reproducer is really needed to investigate the issue.

TonyU1234, 2019-02-14:
Ok, I will try to use it.

TonyU1234, 2019-02-15:
Ok, everything is fine with the new version of ghdl.

closed 2019-02-15 by Paebbels