ghdl/ghdl #750: Linking with C error and segmentation fault

umarcor, 2019-02-07:
I am trying to build a design with GHDL, including two C files and some VUnit libraries. I am doing it in a [PYNQ-Z1](https://reference.digilentinc.com/reference/programmable-logic/pynq-z1/start) board with [this](http://files.digilent.com/Products/PYNQ/pynq_z1_v2.3.zip) SD card image, which is based on Ubuntu 18.04. I am using `gnat-8` and `llvm-6.0`. GHDL was built from sources with `-fPIC`:

``` bash
git clone https://github.com/ghdl/ghdl
cd ghdl
sed -i.bak 's/\(^make\) \(-j$(nproc) 2>make_err\.log\)/\1 LIB_CFLAGS="$LIB_CFLAGS" OPT_FLAGS="$OPT_FLAGS" \2/g' ./dist/travis/build.sh
rm ./dist/travis/build.sh.bak
OPT_FLAGS=-fPIC LIB_CFLAGS=-fPIC ./dist/travis/build.sh -b llvm-6.0 -p ghdl-llvm-fPIC
```

The structure of the files is the following:

- `app/`
  - `main.c`
- `srcs/`
  - `uut.vhd`
- `test/`
  - `c/`
    - `pkg_c.vhd`
    - `tb_c_uut.vhd`
    - `wrapper.c`
  - `py/`
    - `sim.py`
    - `tb_py_axisbuffer.vhd`
    - `vunit_out/`

First, I compile VUnit:

``` bash
cd vhdl/test/py
python3 sim.py --compile
cd -
```

Then, I analyze the VHDL sources:

``` bash
mkdir -p build/obj
cd build/obj

for f in srcs/uut test/c/pkg_c test/c/tb_c_uut; do
  ghdl -a -fPIC --std=08 -P../../vhdl/test/py/vunit_out/ghdl/libraries/vunit_lib "../../vhdl/$f.vhd"
done
```

As explained in [Linking with Ada](https://ghdl.readthedocs.io/en/latest/references/ImplementationOfVHDL.html#linking-with-ada), I bind the simulation unit:

``` bash
ghdl --bind -fPIC --std=08 -P../../vhdl/test/py/vunit_out/ghdl/libraries/vunit_lib <unit>
```

I also build the wrapper with GCC:

``` bash
gcc -fPIC -rdynamic -c ../../vhdl/test/c/wrapper.c -o wrapper.o
```

Last, I try to put everything together:

``` bash
gcc ../../app/main.c -Wl,wrapper.o -Wl,`ghdl --list-link <unit>`
```

and it fails:

``` bash
../../vhdl/test/py/vunit_out/ghdl/libraries/vunit_lib/memory_pkg-body.o: In function `vunit_lib__memory_pkg__set_expected_integer':
ortho:(.text+0x1898): undefined reference to `std__standard__integer_vector_BUILDER'
../../vhdl/test/py/vunit_out/ghdl/libraries/vunit_lib/memory_pkg-body.o: In function `vunit_lib__memory_pkg__serialize':
ortho:(.text+0x3d4c): undefined reference to `std__standard__integer_vector_BUILDER'
../../vhdl/test/py/vunit_out/ghdl/libraries/vunit_lib/codec-2008.o: In function `vunit_lib__codec_2008_pkg__decodeO1':
ortho:(.text+0x244): undefined reference to `std__standard__boolean_vector_BUILDER'
../../vhdl/test/py/vunit_out/ghdl/libraries/vunit_lib/codec-2008.o: In function `vunit_lib__codec_2008_pkg__decodeO2':
ortho:(.text+0x720): undefined reference to `std__standard__integer_vector_BUILDER'
../../vhdl/test/py/vunit_out/ghdl/libraries/vunit_lib/codec-2008.o: In function `vunit_lib__codec_2008_pkg__decodeO3':
ortho:(.text+0xc08): undefined reference to `std__standard__real_vector_BUILDER'
../../vhdl/test/py/vunit_out/ghdl/libraries/vunit_lib/codec-2008.o: In function `vunit_lib__codec_2008_pkg__decodeO4':
ortho:(.text+0x10fc): undefined reference to `std__standard__time_vector_BUILDER'
../../vhdl/test/py/vunit_out/ghdl/libraries/vunit_lib/com_messenger.o: In function `vunit_lib__com_messenger_pkg__messenger_t__subscriber_inbox_is_full':
ortho:(.text+0x20c4): undefined reference to `std__standard__boolean_vector_BUILDER'
collect2: error: ld returned 1 exit status
```

However, if I build `main.c` only and I follow the path in [Linking with foreign object files](https://ghdl.readthedocs.io/en/latest/references/ImplementationOfVHDL.html#linking-with-foreign-object-files), it works:

``` bash
$ gcc -fPIC -rdynamic -DGHDL -c ../../app/main.c -o main.o
$ ghdl -e -fPIC --std=08 -P../../vhdl/test/py/vunit_out/ghdl/libraries/vunit_lib -Wl,wrapper.o -Wl,main.o <unit>
```

I would be grateful for any hint about what might be failing when GCC is used to put everything together, instead of GHDL. Note that the procedure above has been tested on multiple platforms (x86_64 and ARM64).

---

Then, when I execute the binary, the simulation is properly started. This means that the entrypoint is `main.c`, which then calls a function defined in `wrapper.c` that executes `ghdl_main`. Precisely, one of the tasks in `main.c` is to allocate some memory, and the VHDL code retrieves those pointers with functions bind in `wrapper.c`. This works ok.

However, after some minutes (around 15), the simulation terminates with signal 11:

``` bash
$ /usr/bin/time -v ./<unit>
0xb6d01008 0xb6c00008 262144
>> ghdl_main
get_p(0): 262144
get_p(1): 32
get_p(2): 5
get_b(0): 0xb6d01008
get_b(1): 0xb6c00008
       320000000 fs - default                      -    INFO - Init test
Command terminated by signal 11
        Command being timed: "./<unit>"
        User time (seconds): 909.56
        System time (seconds): 0.23
        Percent of CPU this job got: 99%
        Elapsed (wall clock) time (h:mm:ss or m:ss): 15:09.89
        Average shared text size (kbytes): 0
        Average unshared data size (kbytes): 0
        Average stack size (kbytes): 0
        Average total size (kbytes): 0
        Maximum resident set size (kbytes): 66760
        Average resident set size (kbytes): 0
        Major (requiring I/O) page faults: 0
        Minor (reclaiming a frame) page faults: 20070
        Voluntary context switches: 1
        Involuntary context switches: 1110
        Swaps: 0
        File system inputs: 0
        File system outputs: 0
        Socket messages sent: 0
        Socket messages received: 0
        Signals delivered: 0
        Page size (bytes): 4096
        Exit status: 0
```

---

I also tried to execute the equivalent *default* VUnit testbench:

``` bash
$ python3 sim.py
...
   4525090000000 fs - default                      -    INFO - Read [5654,13]:00000000101011001001000100110110
   4525130000000 fs - default                      -    INFO - Read [5654,14]:00000000101011001001000110101000
   4525170000000 fs - default
fail (P=0 S=0 F=1 T=1) lib.tb_axis_loop.test (861.6 seconds)

==== Summary =================================
fail lib.tb_axis_loop.test (861.6 seconds)
==============================================
pass 0 of 1
fail 1 of 1
==============================================
Total time was 861.6 seconds
Elapsed time was 861.6 seconds
==============================================
Some failed!
```

Although I cannot see the termination signal provided by GHDL, I expect it to be 11 too.

I executed both simulations multiple times, and it always fails at the same point (about 910s for the binary and 860s with VUnit). So, I am thinking that there might be some memory limit issue. That would explain why it fails earlier when python3 is used.

However, I checked with `top` during the execution, and the minimum free memory is around 78000 KiB. The swap memory is not even used.

I thought that this might be related to #611. However, on the one hand, that issue should be fixed, so GHDL should report an error instead of crashing. On the other hand, `gdb` does not work:

```
(gdb) r
Starting program: /home/xilinx/vhdl-dbm/examples/axisbuffer/build/obj/test_buffer
warning: the debug information found in "/usr/lib/debug//lib/arm-linux-gnueabihf/ld-2.27.so" does not match "/lib/ld-linux-armhf.so.3" (CRC mismatch).

Program received signal SIGSEGV, Segmentation fault.
0xb6fd8f4e in ?? () from /lib/ld-linux-armhf.so.3
```

I don't know if the segmentation fault can be related to the compilation error and/or `gdb` failing. If you think that any of them should be in a separate issue, please let me know.

Note that the same compilation procedure has been used in the same platform to build an example without VUnit, and it didn't produce any error.

tgingold, 2019-02-07:
For the first question, you should add --std=08 after --list-link

tgingold, 2019-02-07:
Within gdb, does 'where' or 'bt' work ?

umarcor, 2019-02-07:
> For the first question, you should add --std=08 after --list-link

Thanks a lot. That fixed the compilation issue. I wonder why it didn't happen in order platforms or with other examples, since I use the same general script.

The execution is started but, as with the other build procedures, it fails with a segmentation fault. So, let's focus on that.

> Within gdb, does 'where' or 'bt' work ?

I can run nothing, neither binaries built with GHDL nor binaries built with GCC. System utilities such as `/bin/ls` ow `/bin/pwd` also fail. All of them return the same `the debug information found in (...) does not match (...) (CRC mismatch)` error. I can execute `where` or `bt`, but those are not the backtraces of the application, since it didn't even start:

``` bash
Reading symbols from ./buffer_ghdl...done.
(gdb) where
No stack.
(gdb) bt
No stack.
(gdb) r
Starting program: /home/xilinx/vhdl-dbm/examples/axisbuffer/build/bin/buffer_ghdl
warning: the debug information found in "/usr/lib/debug//lib/arm-linux-gnueabihf/ld-2.27.so" does not match "/lib/ld-linux-armhf.so.3" (CRC mismatch).

Program received signal SIGSEGV, Segmentation fault.
0xb6fd8f4e in ?? () from /lib/ld-linux-armhf.so.3
(gdb) bt
#0  0xb6fd8f4e in ?? () from /lib/ld-linux-armhf.so.3
#1  0xb6fd8ef8 in ?? () from /lib/ld-linux-armhf.so.3
Backtrace stopped: previous frame identical to this frame (corrupt stack?)
(gdb) where
#0  0xb6fd8f4e in ?? () from /lib/ld-linux-armhf.so.3
#1  0xb6fd8ef8 in ?? () from /lib/ld-linux-armhf.so.3
Backtrace stopped: previous frame identical to this frame (corrupt stack?)
```

I'm thinking that there is something broken in the system. Only statically linked applications work on gdb.

umarcor, 2019-02-09:
On the one hand, the crash (segmentation fault) is being addressed in #752, since it seems to be related to some of the latest commits in the master branch of VUnit. Certainly, I tried in the PYNQ with the latest (stable) release of VUnit, and it works!

On the other hand, `gdb` works ok in a freshly flashed PYNQ v2.3 SD. Therefore, I think that some of the packages that I installed broke it. It might be because of `gnat-8 libstdc++-8-dev`. Or, I might be missing some additional packages.

Anyway, the linking problem has already be solved, so I'm closing this. Thank, once again, @tgingol.

closed 2019-02-09 by umarcor