ghdl/ghdl #752: Crash on Ubuntu Bionic with LLVM but not on Debian Stretch with mcode

umarcor, 2019-02-08:
**Description**

The MWE in the attached file is executed successfully on docker image `ghdl/ghdl:vunit` (which is based on `ghdl/ghdl:buster-mcode`), but it fails on `ghdl/ghdl:ubuntu18-llvm-5.0`.

**Context**

Coming from #750. Since I could not execute `gdb` on the Zynq, I tried to execute the design on a x86-64 host, inside a docker image with the same OS (Ubuntu Bionic).

There is no `GHDL Bug occurred` log block. I think it is just a segmentation fault.

**How to reproduce?**

Extract the attached file to a folder. From there, execute:

``` bash
$(command -v winpty) docker run --rm -t \
  -v /$(pwd)://src \
  -w //src \
  ghdl/ext:vunit python3 sim.py -v
...
Compile passed

Running test: lib.tb_axis_loop.test
Running 1 tests

Starting lib.tb_axis_loop.test
Output file: vunit_out/test_output/lib.tb_axis_loop.test_260bb5c8c675e898eca5dc9024a4420ede12c0bc/output.txt
               0 fs - default                      -    INFO - Start testbench
       300000000 fs - default                      -    INFO - Init test
       350000000 fs - default                      -    INFO - Sending m_I of size 13108x20 to UUT...
      1330000000 fs - default                      -    INFO - Receiving m_O of size 13108x20 from UUT...
   5243550000000 fs - default                      -    INFO - m_I sent!
   5244530000000 fs - default                      -    INFO - m_O read!
   5244550000000 fs - default                      -    INFO - m_O saved!
   5244590000000 fs - default                      -    INFO - Test done
simulation stopped @5244590ns with status 0
pass (P=1 S=0 F=0 T=1) lib.tb_axis_loop.test (133.2 seconds)

==== Summary =================================
pass lib.tb_axis_loop.test (133.2 seconds)
==============================================
pass 1 of 1
==============================================
Total time was 133.2 seconds
Elapsed time was 133.2 seconds
==============================================
All passed!
```

``` bash
rm -rf vunit_out
```

Now, on `ghdl/ghdl:ubuntu18-llvm-5.0`:

``` bash
$(command -v winpty) docker run --rm -t \
  -v /$(pwd)://src \
  -w //src \
  ghdl/ghdl:ubuntu18-llvm-5.0 bash -c "$(cat ./run.sh)"
...
Compile passed

Running test: lib.tb_axis_loop.test
Running 1 tests

Starting lib.tb_axis_loop.test
Output file: vunit_out/test_output/lib.tb_axis_loop.test_260bb5c8c675e898eca5dc9024a4420ede12c0bc/output.txt
fail (P=0 S=0 F=1 T=1) lib.tb_axis_loop.test (49.9 seconds)

==== Summary =================================
fail lib.tb_axis_loop.test (49.9 seconds)
==============================================
pass 0 of 1
fail 1 of 1
==============================================
Total time was 49.9 seconds
Elapsed time was 49.9 seconds
==============================================
Some failed!
```

Script `run.sh` will first install VUnit, and then execute the simulation. The simulation will run for around 50s, until it fails. No output is produced at all.

[ghdl-bug.zip](https://github.com/ghdl/ghdl/files/2845514/ghdl-bug.zip)

umarcor, 2019-02-09:
After further tests with all the available `ghdl/ghdl:*` docker images, this seems to be an issue with VUnit, which has nothing to do with the backend. In `ghdl/ext:vunit` the latest VUnit release is installed with `pip3 install vunit_hdl`. In the other tests, the master branch from the git repo was cloned.

I checked the commits between the latest VUnit release and the current last commit in the master branch. I found that `551314b` is the last one that works (all the previous commits work), and `1a19233` is the first that fails (all the commits after this fail). Example that can be tested in any `ghdl/ghdl` image:

By replacing file `run.sh` in the zip file with:

``` bash
apt update -y
apt install -y python3 python3-pip git

git clone --recurse-submodule https://github.com/vunit/vunit /opt/vunit
pip3 install -r /opt/vunit/requirements.txt
export PYTHONPATH=/opt/vunit

cd /opt/vunit
# Last working commit
git reset --hard 551314bfcf9268dad7aa71c064c36673d6f40241

cd /src
python3 sim.py -v

cd /opt/vunit
# First failing commit
git reset --hard 1a1923390afef703c5f3ac580836021fc8a8e3ec

cd /src
python3 sim.py -v
```

The output is:

``` bash
...
HEAD is now at 551314b Correct misuse of os.path.join in Modelsim interface unit tests
...
Compile passed

Running test: lib.tb_axis_loop.test
Running 1 tests

Starting lib.tb_axis_loop.test
Output file: vunit_out/test_output/lib.tb_axis_loop.test_260bb5c8c675e898eca5dc9024a4420ede12c0bc/output.txt
               0 fs - default                      -    INFO - Start testbench
       300000000 fs - default                      -    INFO - Init test
       350000000 fs - default                      -    INFO - Sending m_I of size 13108x20 to UUT...
      1330000000 fs - default                      -    INFO - Receiving m_O of size 13108x20 from UUT...
   5243550000000 fs - default                      -    INFO - m_I sent!
   5244530000000 fs - default                      -    INFO - m_O read!
   5244550000000 fs - default                      -    INFO - m_O saved!
   5244590000000 fs - default                      -    INFO - Test done
simulation stopped @5244590ns with status 0
pass (P=1 S=0 F=0 T=1) lib.tb_axis_loop.test (65.7 seconds)

==== Summary =================================
pass lib.tb_axis_loop.test (65.7 seconds)
==============================================
pass 1 of 1
==============================================
Total time was 65.7 seconds
Elapsed time was 65.7 seconds
==============================================
All passed!

HEAD is now at 1a19233 Merge pull request #434 from bradleyharden/test_modelsim_fixup
...
Compile passed

Running test: lib.tb_axis_loop.test
Running 1 tests

Starting lib.tb_axis_loop.test
Output file: vunit_out/test_output/lib.tb_axis_loop.test_260bb5c8c675e898eca5dc9024a4420ede12c0bc/output.txt
fail (P=0 S=0 F=1 T=1) lib.tb_axis_loop.test (50.1 seconds)

==== Summary =================================
fail lib.tb_axis_loop.test (50.1 seconds)
==============================================
pass 0 of 1
fail 1 of 1
==============================================
Total time was 50.1 seconds
Elapsed time was 50.1 seconds
==============================================
Some failed!
```

There is a subtle difference between mcode and gcc/llvm backends. With the former, some of the messages are shown before failing. With the others, no output is shown (which is the case of the log above).

I don't know how to proceed, because commit `1a19233` is the merge of a PR (VUnit/vunit#434) that should have nothing to do with GHDL. Also, GHDL is crashing. @tgingold, if you agree, I will leave this issue open until we guess how to avoid GHDL crashing.

@kraigher, @LarsAsplund, could you kindly have a look at this issue?

tgingold, 2019-02-09:
I can reproduce the crash with the master of vunit.

tgingold, 2019-02-11:
There is a stack overflow, because a string in string_ptr_pkg is getting too big.
I will try to improve the generated code.

umarcor, 2019-02-11:
Thanks. Glad to know. Can I do anything to improve my VHDL sources so that the string doesn't get so big?

I'm intrigued about what changed in VUnit to make it fail, since the offending commit seems to have no relation with string_ptr_pkg.

tgingold, 2019-02-21:
FTR, the offending line is
```
new_ptr := new string'(1 to length => character'low);
```

umarcor, 2019-05-20:
As commented in VUnit/vunit#494, the code that breaks was introduced in VUnit/vunit#420. So, it is possible to work around it by reverting `vunit/vhdl/verification_components/src/axi_stream_master.vhd` to v4.0.8 of VUnit.

Furthermore, it is possible to avoid the crash by setting `ulimit -s unlimited`.

I think that the crash is produced when a variable is allocated which is not as large as the stack limit, but which makes the stack overflow. It seems that GHDL fails at warning about these allocations (ref #611).

Paebbels, 2025-04-27:
@tgingold anything to do for this one?

tgingold, 2025-04-27:
It is possible to work around by increasing stack limit, but I would keep it open to implement some optimizations.

dalex78, 2026-09-04:
# Issue #752 — `new string'(1 to length => …)` overflows the stack with no message

Analysis by an AI agent. The overflow is now reported instead of being a bare SIGSEGV; the underlying optimisation tgingold wants to keep this open for is not done.

## The reproducer, without the zip

The report attached `ghdl-bionic-bug.zip` — four VHDL files, a 2.2 MB CSV and a script that clones VUnit from git inside a docker image. None of that is needed either: tgingold reduced it on the thread himself.

> FTR, the offending line is
> ```
> new_ptr := new string'(1 to length => character'low);
> ```

That is VUnit's `string_ptr_pkg`, and it is the whole bug:

```vhdl
entity repro is
  generic (length : natural := 16 * 1024 * 1024);   -- 16 MB, default stack is 8
end entity;

architecture a of repro is
  type string_acc is access string;
begin
  process
    variable p : string_acc;
  begin
    p := new string'(1 to length => character'low);
    report "allocated " & integer'image (p'length);
    deallocate (p);
    wait;
  end process;
end architecture;
```

The aggregate is built in a temporary and then copied into the allocated object, and that temporary is allocated **on the stack**. The heap object is fine; it is the temporary that overflows.

## What happened

On master, with the ordinary 8 MB stack:

| backend | result |
|---|---|
| mcode | `Segmentation fault (core dumped)`, no message |
| llvm-jit | `Segmentation fault (core dumped)`, no message |
| llvm | exit 255, no message |

The reporter had already worked this out from the outside — *"it is possible to avoid the crash by setting `ulimit -s unlimited`"* — and drew the right conclusion:

> I think that the crash is produced when a variable is allocated which is not as large as the stack limit, but which makes the stack overflow. It seems that GHDL fails at warning about these allocations (ref #611).

## Why nothing was reported

GHDL already has the machinery. `Grt.Lib.Ghdl_Check_Stack_Allocation` reports

```
declaration of a too large object (N kB > --max-stack-alloc=M kB)
```

and `--max-stack-alloc` defaults to 128 kB, so an object of this size should have been caught. It was not, because the call to it is only *generated* in one place. The comment above `Trans.Chap4.Maybe_Check_Stack_Allocation` says so:

```ada
   --  Right now, this function is called only for allocation of a complex
   --  object on the stack (constant or variable).  But there are more sources
   --  of stack allocation (temporary aggregate, unbounded objects, individual
   --  assocs...)
```

**"temporary aggregate"** is exactly this case.

## What the fix does

It generates the check for two of those sources:

- `Trans.Chap4.Allocate_Complex_Object`, when allocating on the stack — this is the one the aggregate above goes through, since `string (1 to length)` is fully constrained (`length` is not static, so the object is complex);
- `Trans.Chap3.Allocate_Unbounded_Composite_Base`, when allocating on the stack — the sibling path taken when the aggregate's own subtype is *not* fully constrained.

`Maybe_Check_Stack_Allocation` is moved to the spec of `Trans.Chap4` so the second one can reach it. Nothing else changes: the check is emitted only when the size is at least the compile-time threshold (32 kB), and it only errors when the run-time limit (`--max-stack-alloc`, 128 kB by default) is exceeded.

All three backends now say:

```
error: declaration of a too large object (16384 kB > --max-stack-alloc=128 kB)
  instance: /repro/P0
  from: process work.repro(a).P0 at repro.vhd:16
ghdl:error: simulation failed
```

with the instance path and the source line, rather than dying without a word.

## What this does not do

It does not make the allocation succeed, and it does not remove the temporary. tgingold's last word on the thread is:

> It is possible to work around by increasing stack limit, but I would keep it open to implement some optimizations.

The optimisation is to build the aggregate directly in the allocated object instead of on the stack, which would make this design work with no stack at all. That is a real change to the allocator and aggregate translation and is not attempted here. What this removes is the part that is indefensible either way: a silent segfault where GHDL already had the diagnostic and simply was not emitting it.

`ulimit -s unlimited` still works, and so does `--max-stack-alloc=0` to disable the limit.

## Testing

`testsuite/gna/issue752` elaborates the reproducer and asserts that the too-large-object diagnostic appears with no internal error box, then re-runs it with a small generic and the limit disabled to check that ordinary allocations are unaffected.

On master the test fails on all three backends (segfault on mcode and llvm-jit, silent failure on llvm). With the change all three pass, and `sanity`, `gna` and `synth` pass on mcode, llvm and llvm-jit.