ghdl/ghdl #754: [question] building ghdllib GCC Back-end

arcturus140, 2019-02-13:
I was able to build mcode backend. Now, I am interested in GCC version. 

After building with `make -j2 && make install`
I need to build the library:
```
cd /path/to/ghdl/source/dir
make ghdllib
make install
```
this fails with:
```
vhdl; do \
  b=`basename $i .vhdl`; \
  if [ -f ../../ieee/v08/$b.o ]; then \
    rm -f -f $b.o; ln -s ../../ieee/v08/$b.o $b.o || exit 1; \
  fi; \
done; \
for i in synopsys/std_logic_arith.vhdl synopsys/std_logic_unsigned.vhdl synopsys/std_logic_signed.vhdl synopsys/std_logic_misc.v08 synopsys/std_logic_misc-body.v08; do \
          cmd="/usr/local/mcode/bin/ghdl -a  --std=08 -P../.. --work=ieee ../../src/$i"; \
  echo $cmd; eval $cmd || exit 1; \
done
/usr/local/mcode/bin/ghdl -a --std=08 -P../.. --work=ieee ../../src/synopsys/std_logic_arith.vhdl
/usr/local/mcode/bin/ghdl -a --std=08 -P../.. --work=ieee ../../src/synopsys/std_logic_unsigned.vhdl
/usr/local/mcode/bin/ghdl -a --std=08 -P../.. --work=ieee ../../src/synopsys/std_logic_signed.vhdl
/usr/local/mcode/bin/ghdl -a --std=08 -P../.. --work=ieee ../../src/synopsys/std_logic_misc.v08
/usr/local/mcode/bin/ghdl -a --std=08 -P../.. --work=ieee ../../src/synopsys/std_logic_misc-body.v08
cd lib/ghdl/std/v87; /usr/local/mcode/bin/ghdl --bootstrap-standard  --std=87
/usr/local/mcode/bin/ghdl:error: unknown command '--bootstrap-standard', try --help
libraries/Makefile.inc:352: recipe for target 'lib/ghdl/std/v87/std_standard.o' failed
```

I am confused why did mcode install to `/usr/local/mcode/`? I set the prefix to `/opt/ghdl/` for both gcc and GHDL:
```
cd ./
mkdir build
cd build
../configure --with-gcc=./gcc-gcc-7_4_0-release --prefix=/opt/ghdl
make copy-sources
mkdir gcc-objs; cd gcc-objs
./gcc-gcc-7_4_0-release --prefix=/usr/local --enable-languages=c,vhdl \
--disable-bootstrap --disable-lto --disable-multilib --disable-libssp \
--disable-libgomp --disable-libquadmath
make -j2 && make install
cd ..
make ghdllib
make install
```

tgingold, 2019-02-13:
You haven't used the same --prefix for gcc configure and ghdl configure.

arcturus140, 2019-02-14:
I have - not sure what I did wrong. It is working now. 

I found a mistake in the documentation, page 49 where it reads:

```
$ make -j2 && make install
$ cd /path/to/ghdl/source/dir
$ make ghdllib
$ make install
```

should be:
```
$ make -j2 && make install
$ cd /path/to/ghdl/source/dir/build
$ make ghdllib
$ make install
```

tgingold, 2019-02-15:
Fixed by your PR.

closed 2019-02-15 by tgingold