ghdl/ghdl #756: Can't instantiate unconstrained array of unconstrained type using generic length

Xiretza, 2019-02-15:
**Description**
VHDL2008 supports unconstrained arrays of unconstrained types (i.e. `array(natural range <>) of std_logic_vector`). `ghdl` seems to mostly support this already, but when using generics to define the array size, a crash occurs. Possibly related to #645.

**Expected behaviour**
Instantiation of these arrays using a generic length constraint should be successful, like with unconstrained arrays of constrained types (`sig_simple` in the example code) and unconstrained arrays of unconstrained types with constant array size (`sig_multi_(1/2)`).

Instead, when using a generic to define the *array* size (`sig_multi_3`), a segfault occurs in GCC:

```
/tmp/ghdl_test/ent.vhd: In function ‘work__test_entity__ARCH__behaviour__DECL_ELAB’:
/tmp/ghdl_test/ent.vhd:21:0: internal compiler error: Segmentation fault
  signal sig_multi_3 : type_multi_array(0 to ARR_SIZE)(7 downto 0);
 
Please submit a full bug report,
with preprocessed source if appropriate.
See <https://gcc.gnu.org/bugs/> for instructions.
ghdl:error: exec error
```

Using a generic to define the *element* size (`sig_multi_2`) works using the latest git master (though not using `v0.35`).

**Context**
- OS: Arch Linux
- Origin:
  - [x] Package manager. Repo: AUR - [ghdl-gcc-git](https://aur.archlinux.org/packages/ghdl-gcc-git/) (f8437ed)
  - [ ] Released binaries. Tarball:
  - [x] Built from sources. Commit SHA:

**How to reproduce?**
Analyze the following example code using 2008 ghdl (see `sim.sh`).

```
#>> ent.vhd
library ieee;
use ieee.std_logic_1164.all,
    ieee.numeric_std.all;

entity test_entity is
	generic (
		ARR_SIZE : natural := 4;
		VECTOR_SIZE : natural := 7
	);
end test_entity;

architecture behaviour of test_entity is
	type type_simple_array is array(natural range <>) of std_logic_vector(7 downto 0);
	type type_multi_array is array(natural range <>) of std_logic_vector;

	signal sig_simple : type_simple_array(0 to ARR_SIZE);
	signal sig_multi_1 : type_multi_array(0 to 4)(7 downto 0);
	signal sig_multi_2 : type_multi_array(0 to 4)(VECTOR_SIZE downto 0);

	-- erroring instruction below
	signal sig_multi_3 : type_multi_array(0 to ARR_SIZE)(7 downto 0);
begin
end behaviour;

#>> sim.sh
ghdl -a --std=08 ent.vhd

#>> run.sh
docker run --rm -tv /$(pwd):/src:z -w //src ghdl/ghdl:buster-mcode sh -c ./sim.sh

#>> end
```

**Checklist**
Before submitting your issue, please review the following checklist:

- [ ] [N/A] Add `GHDL Bug occurred` log block
- [x] Add a MWE
- [x] Try the latest version

Paebbels, 2019-02-16:
> Using a generic to define the _element_ size (`sig_multi_2`) works using the latest git master (though not using `v0.35`).

So is this bug fixed in the latest version, if you build from sources?  
To clarify: What backend are you using?

Xiretza, 2019-02-16:
> So is this bug fixed in the latest version, if you build from sources?

No, the latest version only allows to define the element size (`VECTOR_SIZE downto 0`) using generics, but the doing the same for the array size (`0 to ARR_SIZE`) results in a segfault in GCC.

> To clarify: What backend are you using?

GCC.

Xiretza, 2019-02-17:
I found an even better test case, this time a stripped-down version of #645:

```
library ieee;
use ieee.std_logic_1164.all;

entity test_entity is
	generic (
		n : natural := 5;
		p : natural := 8
	);
end test_entity;

architecture behaviour of test_entity is
	type t is array(natural range <>) of std_logic_vector;

	-- all of these work
	signal s_test1: t(0 to 5-1)(p-1 downto 0);
	signal s_test2: t(0 to 5-1)(8-1 downto 0);
	signal s_test3: t(0 to n-1)(p-1 downto 0);

	-- erroring instruction below
	signal s_test4: t(0 to n-1)(8-1 downto 0);
begin
end behaviour;
```

Reproduce with `ghdl -a --std=08 ent.vhd` (f8437ed8, GCC backend).

As you can see, `n` and `p` are both constant (since the generics are never changed from their default value), yet when the array is defined with a generic array size (`0 to n-1`) and a literal element size (`8-1 downto 0`), ghdl segfaults:

```
/tmp/ghdl_test/ent.vhd: In function ‘work__test_entity__ARCH__behaviour__DECL_ELAB’:
/tmp/ghdl_test/ent.vhd:20:0: internal compiler error: Segmentation fault
  signal s_test4: t(0 to n-1)(8-1 downto 0);
 
Please submit a full bug report,
with preprocessed source if appropriate.
See <https://gcc.gnu.org/bugs/> for instructions.
ghdl:error: exec error
```

Schoggy, 2019-08-21:
The VHDL code provided here now causes the same error message as in #899.
I tried the latest GHDL release, as well as the current state of the repo.
The new issue I opened might be considered a duplicate of this one.

1138-4EB, 2019-08-21:
@Schoggy, I think so. However, note that this issue is labeled as 'Backend: GCC' and 'v0.35'. Your issue is about 'Backend: mcode' and 'v.37-dev'. Therefore, labels should be updated here before closing #899.

Xiretza, 2019-11-24:
Issue still present with latest master (0cb817da), so maybe the v0.35 label should be removed.

fl4shk, 2020-03-14:
I ran into this issue when trying to use GHDL synth recently.

eine, 2020-08-05:
@Xiretza, @mankeph, @fl4shk I think that this might be fixed in latest master.

Xiretza, 2020-08-05:
Can confirm, the test cases here compile successfully on 092e788d with GCC backend.

closed 2020-08-05 by Xiretza