ghdl/ghdl #757: Crash when using to_string for real type

bradleyharden, 2019-02-17:
**Description**
Using the function `to_string` for a real value crashes GHDL

**Expected behaviour**
`to_string` should produce a string

**Context**
******************** GHDL Bug occurred ***************************
Please report this bug on https://github.com/ghdl/ghdl/issues
GHDL release: 0.36-dev (20181129-170-gf8437ed8) [Dunoon edition]
Compiled with GNAT Version: 7.3.0
Target: x86_64-linux-gnu
In directory: /home/bharden/
Command line:
/usr/local/bin/ghdl1-llvm --std=08 -P/usr/local/lib/ghdl/ieee/v08/ -P/usr/local/lib/ghdl/ -c -fpic -o test.o test.vhd
Exception TYPES.INTERNAL_ERROR raised
Exception information:
raised TYPES.INTERNAL_ERROR : errorout.adb:594
Call stack traceback locations:
0x492250 0x4a2e08 0x4ab034 0x4ab1ed 0x4ab745 0x534bce 0x549739 0x54a7ab 0x5258f8 0x526f28 0x527093 0x527ce4 0x5281ca 0x5284eb 0x558b1d 0x55f6c4 0x51d33f 0x62e9a9 0x62f561 0x630e13 0x408617 0x7f7ce5c2db95 0x4079e8 0xfffffffffffffffe
******************************************************************

- OS:
- Origin:
  - [ ] Package manager. Repo:
  - [ ] Released binaries. Tarball:
  - [x] Built from sources. Commit SHA: f725da

**How to reproduce?**
```
entity test is
end entity;

architecture tb of test is
begin
  process
  begin
    report to_string(3.0);
    wait;
  end process;
end architecture;

ghdl -a --std=08 test.vhd
```

**Checklist**
Before submitting your issue, please review the following checklist:

- [x] Add `GHDL Bug occurred` log block
- [x] Add a MWE
- [x] Try the latest version

Paebbels, 2019-02-17:
GitHub doesn't find your referenced SHA f725da. Can you confirm your hash and/or the used GHDL version.

bradleyharden, 2019-02-17:
Sorry. Fixed

bradleyharden, 2019-02-20:
I just realized, I compiled with the standard "llvm" package in Ubuntu 18.04. It looks like that is version 6. Could that be the source of the problem? Should I recompile with version 5?

1138-4EB, 2019-02-20:
> I just realized, I compiled with the standard "llvm" package in Ubuntu 18.04. It looks like that is version 6. Could that be the source of the problem? Should I recompile with version 5?

It's ok. We can reproduce it with multiple backends on multiple platforms. E.g.:

``` bash
#!/bin/sh

cat > test.vhd <<-EOF
entity test is
end entity;

architecture tb of test is
begin
  process
  begin
    report to_string(3.0);
    wait;
  end process;
end architecture;
EOF

for img in fedora28-gcc-8.1.0 fedora28-mcode fedora28-llvm ubuntu18-llvm-5.0 ubuntu18-mcode buster-gcc-7.2.0; do
  docker pull ghdl/ghdl:$img
  $(command -v winpty) docker run --rm -tv /$(pwd)://src -w //src ghdl/ghdl:$img bash -c "ghdl -a --std=08 test.vhd"
done
```

Paebbels, 2019-02-20:
Hmmm a backend and version of the backend would be great.

The backend is currently included in `ghdl -version`.

bradleyharden, 2019-02-20:
```
GHDL 0.36-dev (20181129-170-gf8437ed8) [Dunoon edition]
 Compiled with GNAT Version: 7.3.0
 llvm code generator
Written by Tristan Gingold.

Copyright (C) 2003 - 2015 Tristan Gingold.
GHDL is free software, covered by the GNU General Public License.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
```

bradleyharden, 2019-02-20:
llvm-config gives me version 6.0.0

1138-4EB, 2019-02-20:
@bradleyharden, sorry, we were side-posting here. The comments about adding the backend and version to the bug block is something that came to my mind while running the multiple tests above. We do not need your specific setup (yet), as we see that the bug is reproducible in multiple environments.

tgingold, 2019-02-21:
This issue doesn't depend on the backend.  To_String does not yet handle locally static expressions.

commit 153fdb2a (2021-09-17): testsuite/gna: add a test and close #757
closed 2021-09-17 by tgingold