ghdl/ghdl #758: Pipe content to ghdl through stdin

1138-4EB, 2019-02-20:
Is it possible to analyze a file with GHDL without actually writing it to disk? Something similar to:

``` bash
cat <<-EOF |
entity test is
end entity;

architecture tb of test is
begin
  process
  begin
    report "HELLO";
    wait;
  end process;
end architecture;
EOF
ghdl -a
```

or

``` bash
ghdl -a <<-EOF
entity test is
end entity;

architecture tb of test is
begin
  process
  begin
    report "HELLO";
    wait;
  end process;
end architecture;
EOF
```

Paebbels, 2019-02-20:
Does `cc0` support this?

1138-4EB, 2019-02-20:
`cc0` is not found. However, both `gcc` and `cc` support these:

``` bash
gcc -xc - <<-EOF
void main() { return; }
EOF
```

``` bash
cat <<-EOF | gcc -xc -
void main() { return; }
EOF
```

Note that the language needs to be specified. Otherwise, the following error is shown: `gcc: error: -E or -x required when input is from standard input`

tgingold, 2019-02-21:
No, it is not supported.  That wouldn't be easy as a file must first be analyzed, then elaborated to be run.  Why do you want that feature ?

1138-4EB, 2019-02-21:
Do you mean that you save a reference to the file during analysis and then when the elaboration command is executed you go back to the same file? I thought that once analyzed, GHDL had all the info (from the source) it needed for elaboration. If it is not like that, as you say, it would be difficult to do without GHDL making it's own copy of the file.

tgingold, 2019-02-21:
Yes.  But why do you need ghdl to accept stdin ?

1138-4EB, 2019-02-21:
> Yes. But why do you need ghdl to accept stdin ?

Usually, when I see some issue that needs triaging (e.g. #757 yesterday), I test it on multiple environments and/or I try variations of the MWE provided by the user. Since I use docker images to do so, I need to:

- Save the MWE to one or multiple files.
- Bind the folder(s) where I saved the files to the container I am about to run.
- Set the workdir in the container to the path where the external folder was mounted.
- Execute the ghdl commands that I would have executed locally.

Precisely, this is the behaviour of [1138-4EB/issue-runner](https://github.com/1138-4EB/issue-runner).

Alternatively, the content can be created directly inside the container. This allows to skip two steps, as neither binding nor chaning the workdir is required. The drawback is that, in order to edit the files and try variations, you need to use terminal tools (vim/nano), as GUI apps in the host cannot access them. Or just overwrite all of them every time.

However, if GHDL supported stdin, a group of files could be defined and executed in a single text file. Therefore, it would be possible to copy the body of an issue and paste it directly in the terminal. It would be executed inside a container, without requiring any bind and without leaving any artifact.

See these examples (I have not actually executed all of them, so there might be some bugs):

``` bash
RUN="$(command -v winpty) docker run --rm -it \
 -v /$(pwd)://src -w //src \
 ghdl/ghdl:fedora28-mcode bash -c"

fileA=$(cat <<-EOF
entity test is
end entity;
EOF
)

fileB=$(cat <<-EOF
entity test is
end entity;
EOF
)

printf $fileA > fileA.vhd
printf $fileB > fileB.vhd

# Option 1 (currently used): share files from the host

$RUN '\
  ghdl -a fileA.vhd \
  ghdl -a fileB.vhd \
  ghdl -e fileA.vhd \
  ghdl -e fileB.vhd \
'

# Note that in the following two cases the definition of RUN changes. Options `-v /$(pwd)://src -w //src` are not required here.

RUN="$(command -v winpty) docker run --rm -it \
 ghdl/ghdl:fedora28-mcode bash -c"

# Hence, we are not using files on the host anymore:

rm fileA.vhd fileB.vhd

# Option 2 (currently supported): create the files in the container

$RUN '\
  printf $fileA > fileA.vhd | ghdl -a fileA.vhd \
  printf $fileB > fileB.vhd | ghdl -a fileB.vhd \
  ghdl -e fileA.vhd \
  ghdl -e fileB.vhd \
'

# Option 3 (if stdin was supported): do not write files at all

$RUN '\
  printf $fileA | ghdl -a \
  printf $fileB | ghdl -a \
  printf $fileA | ghdl -e \
  printf $fileB | ghdl -e \
'
```

tgingold, 2019-02-21:
I see your point.  Option 2 would do it!

closed 2019-03-01 by tgingold