ghdl/ghdl #765: RTD hook seems to be broken

1138-4EB, 2019-02-23:
I wanted to check the result of #763 after merging and I see that the documentation is not being update since almost two months ago:

https://readthedocs.org/projects/ghdl/builds/

It seems not to be a problem with the build itself, because all the latest passed successfully. For some reason, builds are just not being triggered.

I tried to login (although I am not a maintainer of GHDL at RTD) and it asked my to authorize the application in my account (which I already did two years ago). I also saw that ghdl organization is not authorized.

Therefore, I think that they changed how RTD connects to GitHub (just as Travis is doing), but this time we were not notified about it.

@tgingold, @Paebbels, could you please try to check the setup, while I try to find references about possible changes on their side?

tgingold, 2019-02-23:
I haven't seen anything wrong.  I plan to trigger a release soon, we will see how it behaves.

1138-4EB, 2019-02-23:
The build passed successfully 6min ago. Did you try it manually?

tgingold, 2019-02-23:
Yes, I have just triggered a build.

1138-4EB, 2019-02-23:
I think I found the issue: https://blog.readthedocs.com/webhook-deprecations/ I am pretty sure this is the case for us:

> This is the date that GitHub will stop sending notifications through their GitHub Services. Your project might be relying on GitHub Services if your GitHub repository is using the ReadTheDocs GitHub Service and does not have another webhook to Read the Docs configured.

So, the webhook needs to be reconfigured:

> It’s possible that your project is still using these deprecated incoming webhooks if your project was configured on Read the Docs more than two years ago. If your project does not have an webhook integrations under the Integrations section of your project’s admin dashboard, you might be using these legacy incoming webhooks.

> In order to continue building automatically, projects should reconfigure their repository to use a new project webhook integration instead. For more information, see our docs on [Webhook Creation](https://docs.readthedocs.io/en/latest/webhooks.html#webhook-creation).

Paebbels, 2019-02-24:
@tgingold  As the webhook has be updated/renewed, is this issue solved?

1138-4EB, 2019-02-24:
@Paebbels it seems not to be solved. No build has been triggered automatically since yesterday, although multiple PRs were merged: https://readthedocs.org/projects/ghdl/builds/

tgingold, 2019-02-25:
Yes, I haven't yet found the way to fix it.

1138-4EB, 2019-02-25:
> Yes, I haven't yet found the way to fix it.

Please, try the following:

- In https://readthedocs.org/dashboard/ghdl-devfork/integrations/ see if you see 'GitHub incoming webhook'. Otherwise, click on 'Add integration'. You will now se a URI: `readthedocs.org/api/v2/webhook/<project-name>/<id>/`. Copy it.
- Then, go to https://github.com/ghdl/ghdl/settings/hooks and paste the address in the RTD webhook. Configure other settings as explained at https://docs.readthedocs.io/en/latest/webhooks.html#github

tgingold, 2019-02-25:
Many thanks!  I missed the copy&past of the URI.
Should be fixed now.

1138-4EB, 2019-02-25:
There should be a green button (`200`) in the Recent Activity section of https://readthedocs.org/dashboard/ghdl/integrations/ > GitHub incoming webhook . If so, we can close this. We can reopen if we see that it is not triggered with the next commit.

tgingold, 2019-02-25:
Green button is present.

closed 2019-02-25 by tgingold

1138-4EB, 2019-03-03:
@tgingold, @Paebbels, just a note to ourselves:

The hook is working properly, and the docs have been built multiple times in the last 24h. That's ok.

However, when @tgingold published a release this morning, the docs were automatically built but `v0.36-dev` was shown in [Downloading pre-built packages](https://ghdl.readthedocs.io/en/latest/getting/Releases.html#downloading-pre-built-packages). This is because Travis CI and RTD are triggered at the same time. Since RTD is faster, it retrieves the info about releases from GitHub before Travis has finished running the jobs. Hence, the latest relelease is not still pushed from Travis to GitHub releases when the table is created.

Anyway, this is just small annoyance. The docs will be fixed the next time a commit is pushed (this morning @tgingold pushed another one an hour after the release), or manually triggering the build of the docs when Travis is done. Just writing this here so I don't freak out again next time.

tgingold, 2019-03-03:
Thanks, that's useful to understand this minor annoyance.