ghdl/ghdl #771: Building GHDL error with gcc-7.4.0 and gcc-6.5.0

Zileus, 2019-02-28:
**Description**
Hi,

I'm trying to compile GHDL (20181129 and 0.36-rc1) wth gcc-7.4.0 and gcc-6.5.0 but I'm facing some troubles.

- OS: CentOS6
- Origin:
  - [ ] Package manager. Repo: 
  - [X ] Released binaries. Tarball: 20181129
  - [X] Built from sources. Commit SHA: e1bda63fb6765dcb064a24f982a6b854ae2a590f

**Shell Script**
Here is the script that I'm using (Exampl given for GCC-6.5.0) but the results are identical between both GCC version.

```
cd ghdl
sudo rm -rf  build
mkdir build
cd build
../configure --with-gcc=/home/user/PROJETS/GitHub/gcc-6.5.0/ --prefix=/usr/local
make clean
make copy-sources
sudo rm -rf gcc-objs 
mkdir gcc-objs
cd gcc-objs
/home/user/PROJETS/GitHub/gcc-6.5.0/configure --prefix=/usr/local --enable-languages=c,vhdl --disable-bootstrap --disable-lto --disable-multilib --disable-libssp --disable-libgomp --disable-libquadmath
make -j4
sudo make install
cd /home/user/PROJETS/GitHub/ghdl/build
make ghdllib
sudo make install

```

**Script Output**

```
rm -f include-fixed/README
cp /home/user/PROJETS/GitHub/gcc-6.5.0/gcc/../fixincludes/README-fixinc include-fixed/README
chmod a+r include-fixed/README
echo timestamp > stmp-int-hdrs
gcc -c -I./ -I/home/user/PROJETS/GitHub/gcc-6.5.0/gcc/vhdl -g -O2 -gnata -gnaty3befhkmr -gnatwae -I- /home/b.compin/PROJETS/GitHub/gcc-6.5.0/gcc/vhdl/nodes_meta.adb
gcc -c -I./ -I/home/user/PROJETS/GitHub/gcc-6.5.0/gcc/vhdl -g -O2 -gnata -gnaty3befhkmr -gnatwae -I- /home/b.compin/PROJETS/GitHub/gcc-6.5.0/gcc/vhdl/flists.adb
gcc -c -I./ -I/home/user/PROJETS/GitHub/gcc-6.5.0/gcc/vhdl -g -O2 -gnata -gnaty3befhkmr -gnatwae -I- /home/user/PROJETS/GitHub/gcc-6.5.0/gcc/vhdl/back_end.ads
gcc -c -I./ -I/home/user/PROJETS/GitHub/gcc-6.5.0/gcc/vhdl -g -O2 -gnata -gnaty3befhkmr -gnatwae -I- /home/user/PROJETS/GitHub/gcc-6.5.0/gcc/vhdl/disp_tree.adb
gcc -c -I./ -I/home/user/PROJETS/GitHub/gcc-6.5.0/gcc/vhdl -g -O2 -gnata -gnaty3befhkmr -gnatwae -I- /home/user/PROJETS/GitHub/gcc-6.5.0/gcc/vhdl/psl-dump_tree.adb
gcc -c -I./ -I/home/user/PROJETS/GitHub/gcc-6.5.0/gcc/vhdl -g -O2 -gnata -gnaty3befhkmr -gnatwae -I- /home/user/PROJETS/GitHub/gcc-6.5.0/gcc/vhdl/ieee.adb
gcc -c -I./ -I/home/user/PROJETS/GitHub/gcc-6.5.0/gcc/vhdl -g -O2 -gnata -gnaty3befhkmr -gnatwae -I- /home/user/PROJETS/GitHub/gcc-6.5.0/gcc/vhdl/ieee-std_logic_1164.adb
gcc -c -I./ -I/home/user/PROJETS/GitHub/gcc-6.5.0/gcc/vhdl -g -O2 -gnata -gnaty3befhkmr -gnatwae -I- /home/user/PROJETS/GitHub/gcc-6.5.0/gcc/vhdl/iir_chains.adb
gcc -c -I./ -I/home/user/PROJETS/GitHub/gcc-6.5.0/gcc/vhdl -g -O2 -gnata -gnaty3befhkmr -gnatwae -I- /home/user/PROJETS/GitHub/gcc-6.5.0/gcc/vhdl/sem_assocs.adb
gcc -c -I./ -I/home/user/PROJETS/GitHub/gcc-6.5.0/gcc/vhdl -g -O2 -gnata -gnaty3befhkmr -gnatwae -I- /home/user/PROJETS/GitHub/gcc-6.5.0/gcc/vhdl/sem_decls.adb
gcc -c -I./ -I/home/user/PROJETS/GitHub/gcc-6.5.0/gcc/vhdl -g -O2 -gnata -gnaty3befhkmr -gnatwae -I- /home/user/PROJETS/GitHub/gcc-6.5.0/gcc/vhdl/sem_expr.adb
gcc -c -I./ -I/home/user/PROJETS/GitHub/gcc-6.5.0/gcc/vhdl -g -O2 -gnata -gnaty3befhkmr -gnatwae -I- /home/b.compin/PROJETS/GitHub/gcc-6.5.0/gcc/vhdl/sem_inst.adb
gcc -c -I./ -I/home/user/PROJETS/GitHub/gcc-6.5.0/gcc/vhdl -g -O2 -gnata -gnaty3befhkmr -gnatwae -I- /home/user/PROJETS/GitHub/gcc-6.5.0/gcc/vhdl/sem_names.adb
gcc -c -I./ -I/home/user/PROJETS/GitHub/gcc-6.5.0/gcc/vhdl -g -O2 -gnata -gnaty3befhkmr -gnatwae -I- /home/user/PROJETS/GitHub/gcc-6.5.0/gcc/vhdl/sem_specs.adb
gcc -c -I/ -I/home/user/PROJETS/GitHub/gcc-6.5.0/gcc/vhdl -g -O2 -gnata -gnaty3befhkmr -gnatwae -I- /home/user/PROJETS/GitHub/gcc-6.5.0/gcc/vhdl/sem_stmts.adb
gcc -c -I./ -I/home/user/PROJETS/GitHub/gcc-6.5.0/gcc/vhdl -g -O2 -gnata -gnaty3befhkmr -gnatwae -I- /home/user/PROJETS/GitHub/gcc-6.5.0/gcc/vhdl/sem_utils.adb
gcc -c -I./ -I/home/user/PROJETS/GitHub/gcc-6.5.0/gcc/vhdl -g -O2 -gnata -gnaty3befhkmr -gnatwae -I- /home/user/PROJETS/GitHub/gcc-6.5.0/gcc/vhdl/xrefs.adb
gcc -c -I./ -I/home/user/PROJETS/GitHub/gcc-6.5.0/gcc/vhdl -g -O2 -gnata -gnaty3befhkmr -gnatwae -I- /home/user/PROJETS/GitHub/gcc-6.5.0/gcc/vhdl/disp_vhdl.adb
gcc -c -I./ -I/home/user/PROJETS/GitHub/gcc-6.5.0/gcc/vhdl -g -O2 -gnata -gnaty3befhkmr -gnatwae -I- /home/user/PROJETS/GitHub/gcc-6.5.0/gcc/vhdl/nodes_gc.adb
gcc -c -I./ -I/home/user/PROJETS/GitHub/gcc-6.5.0/gcc/vhdl -g -O2 -gnata -gnaty3befhkmr -gnatwae -I- /home/user/PROJETS/GitHub/gcc-6.5.0/gcc/vhdl/parse.adb
gcc -c -I./ -I/home/user/PROJETS/GitHub/gcc-6.5.0/gcc/vhdl -g -O2 -gnata -gnaty3befhkmr -gnatwae -I- /home/user/PROJETS/GitHub/gcc-6.5.0/gcc/vhdl/post_sems.adb
gcc -c -I./ -I/home/user/PROJETS/GitHub/gcc-6.5.0/gcc/vhdl -g -O2 -gnata -gnaty3befhkmr -gnatwae -I- /home/user/PROJETS/GitHub/gcc-6.5.0/gcc/vhdl/ortho_nodes.ads
gcc -c -I./ -I/home/user/PROJETS/GitHub/gcc-6.5.0/gcc/vhdl -g -O2 -gnata -gnaty3befhkmr -gnatwae -I- /home/user/PROJETS/GitHub/gcc-6.5.0/gcc/vhdl/trans.adb
gcc -c -I./ -I/home/user/PROJETS/GitHub/gcc-6.5.0/gcc/vhdl -g -O2 -gnata -gnaty3befhkmr -gnatwae -I- /home/user/PROJETS/GitHub/gcc-6.5.0/gcc/vhdl/trans-chap1.adb
gcc -c -I./ -I/home/user/PROJETS/GitHub/gcc-6.5.0/gcc/vhdl -g -O2 -gnata -gnaty3befhkmr -gnatwae -I- /home/user/PROJETS/GitHub/gcc-6.5.0/gcc/vhdl/trans-chap12.adb
gcc -c -I./ -I/home/user/PROJETS/GitHub/gcc-6.5.0/gcc/vhdl -g -O2 -gnata -gnaty3befhkmr -gnatwae -I- /home/user/PROJETS/GitHub/gcc-6.5.0/gcc/vhdl/trans-chap2.adb
gcc -c -I./ -I/home/user/PROJETS/GitHub/gcc-6.5.0/gcc/vhdl -g -O2 -gnata -gnaty3befhkmr -gnatwae -I- /home/user/PROJETS/GitHub/gcc-6.5.0/gcc/vhdl/trans-chap3.adb
gcc -c -I./ -I/home/user/PROJETS/GitHub/gcc-6.5.0/gcc/vhdl -g -O2 -gnata -gnaty3befhkmr -gnatwae -I- /home/user/PROJETS/GitHub/gcc-6.5.0/gcc/vhdl/trans-chap4.adb
gcc -c -I./ -I/home/user/PROJETS/GitHub/gcc-6.5.0/gcc/vhdl -g -O2 -gnata -gnaty3befhkmr -gnatwae -I- /home/user/PROJETS/GitHub/gcc-6.5.0/gcc/vhdl/trans-chap7.adb
trans-chap7.adb:181:67: warning: redundant conversion, "I" is of type "Pos32"
gnatmake: "/home/user/PROJETS/GitHub/gcc-6.5.0/gcc/vhdl/trans-chap7.adb" compilation error
make[2]: *** [ghdl1] Erreur 4
rm gcc.pod
make[2]: quittant le répertoire « /home/user/PROJETS/GitHub/ghdl/build/gcc-objs/gcc »
make[1]: *** [all-gcc] Erreur 2
make[1]: quittant le répertoire « /home/user/PROJETS/GitHub/ghdl/build/gcc-objs »
make: *** [all] Erreur 2

```

Any ideas? Thanks anyway!

tgingold, 2019-02-28:
The simplest solution is to configure using --disable-werror
I will fix the warning.

Zileus, 2019-03-01:
Even if I add the "--disable-werror" to configure, the script fails with the same previous error.

1138-4EB, 2019-03-01:
Hi @Zileus! We have a `fedora28+gcc-8.1.0` build in our CI environment that can be useful for you to debug the procedure. This is the build log: https://travis-ci.com/ghdl/docker/jobs/181498003

The build is executed in a container based on https://github.com/ghdl/docker/blob/master/dockerfiles/build/fedora28. So, these packages are required: `gcc-gnat zlib-devel make gcc-c++`.

- https://github.com/ghdl/docker/blob/gcc/.travis.yml#L11
- https://github.com/ghdl/docker/blob/gcc/travis/run.sh#L83-L93
- https://github.com/ghdl/ghdl/blob/master/dist/travis/travis-ci.sh
- https://github.com/ghdl/ghdl/blob/master/dist/travis/build.sh

Hence, you can try:

``` bash
git clone https://github.com/ghdl/ghdl
cd ghdl
./dist/travis/build.sh -b gcc-7.4.0 -p ghdl-gcc
```

And the result should be a `ghdl-gcc.tgz` file.

Zileus, 2019-03-01:
hello @1138-4EB after some tricks on the target url for gcc, I run the script "./dist/travis/build.sh -b gcc-7.4.0 -p ghdl-gcc" and it finished with the same error 

```
trans-chap7.adb:181:67: warning: redundant conversion, "I" is of type "Pos32"
gnatmake: "/home/user/PROJETS/GitHub/ghdl-0.36-rc1/build-gcc-7.4.0/gcc-srcs/gcc/vhdl/trans-chap7.adb" compilation error
make[2]: *** [ghdl1] Erreur 4
make[1]: *** [all-gcc] Erreur 2
make: *** [all] Erreur 2
travis_time:end:757071558:start=1551453440757619610,finish=1551453662052651818,duration=221295032208
```
Thanks for you help anyway!

1138-4EB, 2019-03-01:
What version of GCC are you using for compilation? 4?

Zileus, 2019-03-01:
here is the result from "gcc -v" 

```
gcc -v
Utilisation des specs internes.
Target: x86_64-redhat-linux
Configuré avec: ../configure --prefix=/usr --mandir=/usr/share/man --infodir=/usr/share/info --with-bugurl=http://bugzilla.redhat.com/bugzilla --enable-bootstrap --enable-shared --enable-threads=posix --enable-checking=release --with-system-zlib --enable-__cxa_atexit --disable-libunwind-exceptions --enable-gnu-unique-object --enable-languages=c,c++,objc,obj-c++,java,fortran,ada --enable-java-awt=gtk --disable-dssi --with-java-home=/usr/lib/jvm/java-1.5.0-gcj-1.5.0.0/jre --enable-libgcj-multifile --enable-java-maintainer-mode --with-ecj-jar=/usr/share/java/eclipse-ecj.jar --disable-libjava-multilib --with-ppl --with-cloog --with-tune=generic --with-arch_32=i686 --build=x86_64-redhat-linux
Modèle de thread: posix
gcc version 4.4.7 20120313 (Red Hat 4.4.7-23) (GCC)
```

closed 2019-03-01 by tgingold

Zileus, 2019-03-26:
Hi everyone,

I would like to known how can I specify where the GNAT binaries are located.
On my CentOs system, the GNAT version seems to be too old for compiling GHDL.

Regards,

tgingold, 2019-03-26:
Multiple answer:
* you can use: `make GNATMAKE="/path/to/gnatmake"`
* you can remove -gnatwae in Make-lang.in

commit e138506d (2019-03-01): Remove a redundant conversion.

Fix #771