ghdl/ghdl #772: Analyze error when using 'stable(time) attribute inside PSL statement

tmeissner, 2019-03-01:
**Description**
There is a analyze error when using` 'stable(time)` attribute inside PSL statement. The `'stable` attribute without parameters works without problems within PSL statements. In VHDL code, both variant with/without parameters are working without problems.

**Expected behaviour**
GHDL should analyze without error.

**Context**
Please paste the `GHDL Bug occurred` log block here. Also, provide the following information:

```
******************** GHDL Bug occurred ***************************
Please report this bug on https://github.com/ghdl/ghdl/issues
GHDL release: 0.36-rc1 (v0.36-rc1-8-ga62344e) [Dunoon edition]
Compiled with GNAT Version: 4.9.3
Target: x86_64-linux-gnu
In directory: /home/torsten/Projects/Git/bug_reports/ghdl_/
Command line:
/usr/local/bin/ghdl1-llvm --std=08 -fpsl -P/usr/local/lib/ghdl/ieee/v08/ -P/usr/local/lib/ghdl/ -c -o test.o test.vhd
Exception SYSTEM.ASSERTIONS.ASSERT_FAILURE raised
Exception information:
Exception name: SYSTEM.ASSERTIONS.ASSERT_FAILURE
Message: no field Named_Entity
Call stack traceback locations:
0x7f9b864ada81 0x460feb 0x619acf 0x60e6f8 0x60efbe 0x60ec03 0x60ece7 0x60ec03 0x60ecb9 0x60f8fc 0x609417 0x6095f8 0x636450 0x63ce50 0x4db8af 0x5faa29 0x5fb596 0x475c06 0x43023e 0x7f9b8593482e 0x42f5a7 0xfffffffffffffffe
******************************************************************
ghdl:error: compilation error
```

- OS: Ubuntu 16.04
- Origin:
  - [ ] Package manager. Repo:
  - [ ] Released binaries. Tarball:
  - [x] Built from sources. Commit SHA: ga62344e

**Additional context**

```
GHDL 0.36-rc1 (v0.36-rc1-8-ga62344e) [Dunoon edition]
 Compiled with GNAT Version: 4.9.3
 llvm code generator
Written by Tristan Gingold.

Copyright (C) 2003 - 2019 Tristan Gingold.
GHDL is free software, covered by the GNU General Public License.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
```

**How to reproduce?**
Tell us how to reproduce this issue. Please provide a Minimal Working Example (MWE), that is compatible with [issue-runner](https://github.com/1138-4EB/issue-runner). With sample code it's easier to reproduce the bug and it's much faster to fix it. For example:

```
#>> ent.vhd
library ieee;
use ieee.std_logic_1164.all;
use ieee.numeric_std.all;

entity ent is
end ent test;

architecture sim of ent is

  signal s_clk  : std_logic := '1';
  signal s_test : std_logic := '0';

begin

  s_clk <= not s_clk after 5 ns;
  s_test <= '1' after 30 ns,
            '0' after 40 ns;

  process is
  begin
    wait until rising_edge(s_clk);
    -- This works
    if s_test'stable(10 ns) then
      report "s_test stable";
    else
      report "s_test changed";
    end if;
  end process;
 
  -- This works
  -- psl assert always (s_test'stable)@rising_edge(s_clk);
 
  -- This leads to an compile error
  -- psl assert always (s_test'stable(10 ns))@rising_edge(s_clk);
 
end architecture sim;

#>> sim.sh
ghdl -a -fpsl ent.vhd
#>> end
```

Note that `run.sh` is used to execute `sim.sh` inside a docker container. Please, put your commands in `sim.sh` and just copy `run.sh` from the example. Using `ghdl/ghdl:*` docker images to run the MWEs ensures that the latest available GHDL is used.

**Checklist**
Before submitting your issue, please review the following checklist:

- [x] Add `GHDL Bug occurred` log block
- [x] Add a MWE
- [x] Try the latest version

tmeissner, 2019-03-12:
I want to mention that I see low priority with this bug. For the checks I wanted to do with the attribute I have a simple workaround using a additional register which stores the old value of the signal I want to check to be stable. All this wrapped in a VHDL block which is ignored by synthesis.

For example:

```vhdl
-- synthesis off
verification : block is
    
  signal s_data : std_logic_vector(0 to 127);
    
begin
    
  s_data <= data_o  when rising_edge(clk_i) else
                  128x"0" when reset_i = '0';

  -- psl cover valid_o and not accept_i;
  -- psl assert always (valid_o and not accept_i -> next data_o = s_data);
    
end block verification;
-- synthesis on
```

This is a simple streaming IF, which should hold data stable when valid and not yet accepted.

tgingold, 2019-05-11:
Was fixed recently.

closed 2019-05-11 by tgingold