ghdl/ghdl #775: Adding libraries to the vendors directory.

m-kru, 2019-03-08:
Is it desirable to add libraries,  that are not vendor libraries or popular libraries (like OSVVM or UVVM) to the vendors directory? 

Namely, I work with the [IPbus](https://github.com/ipbus/ipbus-firmware) and I need to create a library for myself anyway. I thought, that by the way, I could add it to the ghdl and prepare some compile scripts.

tgingold, 2019-03-09:
I am OK with adding building scripts in vendors, but another possibility is to add scripts in IPbus.  That looks better for maintainability.

m-kru, 2019-03-10:
Ok, I will make PR to the IPbus with hope that it will be accepted.

closed 2019-03-10 by m-kru

tgingold, 2019-03-10:
Another possibility is to describe IPbus using FuseSoc, so that it could be compiled on all analyzers supported by FuseSoc.