ghdl/ghdl #776: Support for applying configurations to non-top level entities

gianluca-nitti, 2019-03-10:
I'm not sure whether this is a GHDL issue or a VHDL language limitation as I'm not used to other tools; I apologize in case this is off-topic.

I'm trying to use `configuration`s to bind black-box components to actual entities; I think I managed to achieve this if the component is instantiated in the top-level entity, but not otherwise.

I'm taking [this code from nandland.com](https://www.nandland.com/vhdl/examples/example-configuration-vhdl.html) as an example:
```
library ieee;
use ieee.std_logic_1164.all;
entity HA_Entity is
  port (
    i_bit1  : in std_logic;
    i_bit2  : in std_logic;
    o_sum   : out std_logic;
    o_carry : out std_logic
    );
end HA_Entity;
architecture HA_Arch of HA_Entity is
  component HA_Comp is
    port (
      i_bit1  : in  std_logic;
      i_bit2  : in  std_logic;
      o_sum   : out std_logic;
      o_carry : out std_logic
      );
  end component HA_Comp;
begin
  HA_Inst : HA_Comp
    port map (
      i_bit1  => i_bit1,
      i_bit2  => i_bit2,
      o_sum   => o_sum,
      o_carry => o_carry);
end HA_Arch;
library ieee;
use ieee.std_logic_1164.all;
entity HA_Comp_Entity is
  port (
    i_bit1  : in std_logic;
    i_bit2  : in std_logic;
    o_sum   : out std_logic;
    o_carry : out std_logic
    );
end HA_Comp_Entity;
architecture HA_Comp_Arch_1 of HA_Comp_Entity is
begin
  o_sum   <= i_bit1 xor i_bit2;
  o_carry <= i_bit1 and i_bit2;
end HA_Comp_Arch_1;
use work.all;
configuration HA_Config of HA_Entity is
  for HA_Arch
    for HA_Inst : HA_Comp
      use entity HA_Comp_Entity(HA_Comp_Arch_1);
    end for;
  end for;
end HA_Config;
```

My first attempt was the following:
```
ghdl -a main.vhd
ghdl -e HA_Entity
```
However, it looks like no binding takes place, since I get the following warning:
```
main.vhd:32:3:warning: component instance "ha_inst" of 'ha_comp' is not bound [-Wbinding]
main.vhd:18:14:warning: (in default configuration of ha_entity(ha_arch)) [-Wbinding]
```
After some trial and error, I found out that with `ghdl -e HA_Config HA_Entity`, no warning is raised, and the built binary seems fine, since it contains the desired entity in place of the component:
```
$ ./ha_config-ha_entity --disp-tree
ha_entity [entity]
+-i_bit1 [port in]
+-i_bit2 [port in]
+-o_sum [port out]
+-o_carry [port out]
`-ha_arch [arch]
  `-ha_inst [instance]
    +-i_bit1 [port in]
    +-i_bit2 [port in]
    +-o_sum [port out]
    +-o_carry [port out]
    `-ha_comp_entity [entity]
      +-i_bit1 [port in]
      +-i_bit2 [port in]
      +-o_sum [port out]
      +-o_carry [port out]
      `-ha_comp_arch_1 [arch]
        +-P0 [process]
        `-P1 [process]
std.standard [package]
ieee.std_logic_1164 [package]
```
Is this the intended way? The main problem with it is that I'm not able to use it if the entity containing the components I want to bind is not the top-level one, i.e. if I add to the file
```
library ieee;
use ieee.std_logic_1164.all;

entity my_top is end entity;

architecture my_top_impl of my_top is
begin
        HA_Entity_instance: entity work.HA_Entity port map('0', '0');
end architecture;
```
Then I can't use `my_top` as top level and bind the component in `HA_Entity`; I tried `ghdl -e HA_Config my_top`, but without success - I get the following link error:
```
/usr/bin/ld: main.o: in function `work__my_top__ARCH__my_top_impl__ha_entity_instance__COMP_ELAB':
main.vhd:(.text+0x6f4): undefined reference to `work__ha_entity__LASTARCH__DEFAULT_CONFIG'
collect2: error: ld returned 1 exit status
ghdl:error: compilation error
```
In fact it doesn't make much sense, since in my understanding it's like trying to use a configuration for an architecture which isn't its target.

Thanks for any answer.

tgingold, 2019-03-11:
You can elaborate either an entity (+ its architecture) or a configuration.
The fact that you were allowed to specify the entity along with a configuration is not expected and I will investigate that.

In my_top, you cannot specify the configuration you want to use for ha_entity_instance (I think this is a hole in the VHDL language).   But you can instantiate configuration HA_Config instead of entity HA_Entity, or use a component and configure it.

The link error message is also unexpected.

diogratia, 2019-03-12:
If you modify my_top_impl to instantiate a component:

```vhdl
library ieee;
use ieee.std_logic_1164.all;

entity my_top is end entity;

architecture my_top_impl of my_top is
    component ha_entity is  -- ADDED component declaration
        port (
            i_bit1  : in std_logic;
            i_bit2  : in std_logic;
            o_sum   : out std_logic;
            o_carry : out std_logic
        );
    end component;
begin
       -- HA_Entity_instance: entity work.HA_Entity port map('0', '0');
HA_Entity_instance:
    ha_entity 
        port map (
            '0',
            '0'
        );
       
end architecture;
```
You can use a top level configuration:
```vhdl
configuration my_top_config of my_top is
    for my_top_impl
        for HA_Entity_instance: ha_entity
            use configuration work.ha_config;
        end for;
    end for;
end my_top_config;
```

Which 
```none
./my_top_config-my_top --disp-tree
my_top [entity]
`-my_top_impl [arch]
  `-ha_entity_instance [instance]
    +-i_bit1 [port in]
    +-i_bit2 [port in]
    +-o_sum [port out]
    +-o_carry [port out]
    `-ha_entity [entity]
      +-i_bit1 [port in]
      +-i_bit2 [port in]
      +-o_sum [port out]
      +-o_carry [port out]
      `-ha_arch [arch]
        `-ha_inst [instance]
          +-i_bit1 [port in]
          +-i_bit2 [port in]
          +-o_sum [port out]
          +-o_carry [port out]
          `-ha_comp_entity [entity]
            +-i_bit1 [port in]
            +-i_bit2 [port in]
            +-o_sum [port out]
            +-o_carry [port out]
            `-ha_comp_arch_1 [arch]
              +-P0 [process]
              `-P1 [process]
std.standard [package]
ieee.std_logic_1164 [package]
```
does the right thing.

gianluca-nitti, 2019-03-12:
Thank you both for the explanations.

I tried both solutions. @tgingold,
> you can instantiate configuration HA_Config instead of entity HA_Entity

Works, if I write the architecture this way (in case anyone has the same problem, remember to explicitly use the `configuration` keyword, otherwise analysis will fail since it will expect a component):
```
architecture my_top_impl of my_top is
begin
        HA_Entity_instance: configuration HA_Config port map('0', '0');
end architecture;
```
and then just elaborate `my_top` (`ghdl -a main.vhd && ghdl -e my_top`).

Also, I confirm @diogratia's solution works as well (only difference is you need to elaborate `my_top_config` instead of `my_top`).

tgingold, 2019-03-12:
I have improved the error message, but I cannot reproduce the link error.

commit 145aa829 (2019-03-14): vhdl: error when a secondary unit is used for a configuration.

Fix #776
closed 2019-03-14 by tgingold