ghdl/ghdl #780: Two-dimensional (2D) unconstrained array in record.

m-kru, 2019-03-13:
Does GHDL support two-dimensional (2D) unconstrained array in record types?

Take a look at the following code snippet:
```
  type t_slv_array      is array (natural range <>) of std_logic_vector;
  type t_ipbus_tranaction_header is
  record
    protocol_version : natural range 0 to 2**4  - 1;
    transaction_id   : natural range 0 to 2**12 - 1;
    words            : natural range 0 to 2**8  - 1;
    type_id          : t_ipbus_transaction_type_id;
    info_code        : t_ipbus_transaction_info_code;
  end record;

  type t_ipbus_transaction is
  record
    header : t_ipbus_tranaction_header;
    bodyy  : t_slv_array;
  end record;
```

When I try to define a signal in a design in the following way:
`signal transaction : t_ipbus_transaction(bodyy(7 downto 0)(31 downto 0));`

GHDL Bug occurs:
```
******************** GHDL Bug occurred ***************************
Please report this bug on https://github.com/ghdl/ghdl/issues
GHDL release: 0.36-rc1 (v0.36-rc1-8-ga62344e4) [Dunoon edition]
Compiled with GNAT Version: 8.2.1 20181127
Target: x86_64-pc-linux-gnu
In directory: /home/mkru/workspace/UVVM_Community_VIPs/vip_ipbus/
Command line:
/usr/local/bin/ghdl1-llvm --std=08 --workdir=work -frelaxed-rules -P/usr/local/lib/ghdl/vendors/ -P/usr/local/lib/ghdl/ieee/v08/ -P/usr/local/lib/ghdl/ -c -fpic -o work/ipbus_tvm_tb.o tb/ipbus_tvm_tb.vhd
Exception SYSTEM.ASSERTIONS.ASSERT_FAILURE raised
Exception information:
raised SYSTEM.ASSERTIONS.ASSERT_FAILURE : no field Element_Subtype_Indication
Call stack traceback locations:
0x55b943dcb7f1 0x55b943bbd134 0x55b943cb30f7 0x55b943cb3d99 0x55b943cb4c7c 0x55b943cb5208 0x55b943cc63f5 0x55b943cc9715 0x55b943ca0354 0x55b943cd2bb7 0x55b943cd9bf2 0x55b943c94c90 0x55b943db3a63 0x55b943db46bf 0x55b943db6007 0x55b943b7c2c0 0x7feaf24de221 0x55b943b7b51c 0xfffffffffffffffe
******************************************************************
```

When I define type `t_slv_array `as:
`  type t_slv_array      is array (natural range <>) of std_logic_vector(31 downto 0);`
Then define signal as:
`signal transaction : t_ipbus_transaction(bodyy(7 downto 0));`

Everything works fine.

milestone12, 2019-05-01:
Hi there,

As far as I have experienced myself, GHDL handles actual two-dimensional arrays of a constrained array type just fine, e.g.:
```vhdl
type arr_2d_t is array (natural range <>, natural range <>) of std_logic_vector(31 downto 0);
```

However, the case that @m-kru describes is that of an unconstrained array of an unconstrained array type, e.g.:
```vhdl
type arr_uu_t is array (natural range <>) of std_logic_vector;
```
Such type declarations have already caused me some headaches, and it seems as if GHDL didn't have proper support for it yet. However, as such a type declaration is syntactically correct, it also doesn't report it as "unsupported".

I'd like to contribute to (a) GHDL reporting that arrays of unconstrained types are currently unsupported, and to (b) implementing support for them in the long run.
Therefore, I'd be happy for some hints on where to start in the GHDL code base.

tgingold, 2019-05-02:
Arrays of unconstrained types are supported in general, but there might be remaining issues.

Do not hesitate to report any issue; if you want to investigate you should start with gdb.
In case of crash of ghdl, run it within gdb (be careful as 'ghdl' spawns ghdl-llvm or ghdl-gcc),
and execute 'catch exception'.
There is a .gdbinit in the sources of ghdl.

eine, 2020-08-05:
The following MWE works for me with latest master:

```vhdl
library ieee;
context ieee.ieee_std_context;

entity dut is
end entity dut;

architecture rtl of dut is

  type t_slv_array is array (natural range <>) of std_logic_vector;
  type t_ipbus_tranaction_header is
  record
    protocol_version : natural range 0 to 2**4  - 1;
    transaction_id   : natural range 0 to 2**12 - 1;
    words            : natural range 0 to 2**8  - 1;
  end record;

  type t_ipbus_transaction is
  record
    header : t_ipbus_tranaction_header;
    bodyy  : t_slv_array;
  end record;

  signal transaction : t_ipbus_transaction(bodyy(7 downto 0)(31 downto 0));

begin

end architecture rtl;
```

```sh
# ghdl -a --std=08 ent.vhd 
# ghdl -e --std=08 dut
```

@m-kru, can you please confirm?

m-kru, 2020-08-05:
@eine I do not have any code to test right now, as I have implemented things in different way. If the above works for you, I guess the issue can be closed.

closed 2020-08-05 by eine