ghdl/ghdl #784: Cannot build ghdllib with gcc back-end

stdefeber, 2019-03-21:
**Description**
I am trying to build ghdl with gcc-7.4.0. However I cannot build ghdllib.

**Expected behaviour**
What you expected to happen, and what is happening instead.

**Context**
Please paste the `GHDL Bug occurred` log block here. Also, provide the following information:

- OS: CentOs 7.5.1804
- Origin: 
  - [ ] Package manager. Repo:
  - [ ] Released binaries. Tarball:
  - [ ] Built from sources. Commit SHA: 0c79f9cfade3b5a8fc8bb24cad944e70109b86e5

**How to reproduce?**

`$ cd ghdl`
`$ mkdir build`
`$ cd build`
`$ ../configure --with-gcc=/path/to/gcc/source/dir --prefix=/usr/local`
`$ make copy-sources`
`$ mkdir gcc-objs; cd gcc-objs`
`$ /path/to/gcc/source/dir/configure --prefix=/usr/local --enable-languages=c,vhdl \`
`--disable-bootstrap --disable-lto --disable-multilib --disable-libssp \`
`--disable-libgomp --disable-libquadmath`
`$ make -j2 && make install`
`$ cd /path/to/ghdl/source/dir/build`
``
Next is where the build stops.
There is simply no rule to make ghdllib.

`$ make ghdllib `

`make:` *** No rule to make target `ghdllib'.  `Stop.`

1138-4EB, 2019-03-21:
Hi @stdefeber! As commented in [#771](https://github.com/ghdl/ghdl/issues/771#issuecomment-468666848), you can try:

``` bash
git clone https://github.com/ghdl/ghdl
cd ghdl
./dist/travis/build.sh -b gcc-7.4.0 -p ghdl-gcc
```

The result should be a `ghdl-gcc.tgz` file.

tgingold, 2019-03-21:
After `configure` for ghdl, a `Makefile` is created.  It must have a `ghdllib` target.

I don't know what could have happen, but this works on many systems.  At worst, you should post the `Makefile`.

closed 2019-05-11 by tgingold