ghdl/ghdl #785: synopsis std_logic_misc missing when using vhdl-08

dstadelm, 2019-03-22:
**Description**
During execution of a simulation of files compiled with vhdl standard 2008 the std_logic_misc seems to be missing. 

**Context**
Please paste the `GHDL Bug occurred` log block here. Also, provide the following information:
´´´
unit "std_logic_misc" not found in library "ieee"
´´´

- OS: Linux Ubuntu 18.04
- Origin:
  - [ ] Package manager. Repo:
  - [ ] Released binaries. Tarball:
  - [ x] Built from sources. Commit SHA: v0.37-dev

**Additional context**
ghdl is called through vunit

tgingold, 2019-03-23:
You should use `--ieee=synopsys` to get std_logic_misc, otherwise you only have access to the standard ieee library.

dstadelm, 2019-03-23:
Actually I've used the option. Am I correct, that the option only applies
to the analyse phase. The error occurs at the run phase. I'm using vunit as
'frontend' maybe the option gets lost somewhere on the way for some files,
I'll check that.
Regards
Dave

tgingold, 2019-03-23:
No, you should use the option for all the phases (analysis, elaboration and run).

dstadelm, 2019-03-23:
Ok, thanks for the clarification.
Regards
Dave

dstadelm, 2019-03-23:
Ok, thanks it worked. The run phase does not know about the ieee option,
but I was missing it for the elaboration phase.

Regards
Dave

tgingold, 2019-03-23:
Yes, that depends on which backend you are using.  Thanks!

closed 2019-03-23 by tgingold