ghdl/ghdl #786: Crash when assigning to array aggregate containing slices

Xiretza, 2019-03-25:
**Description**
When assigning to array aggregates containing array slices, a GHDL bug occurs.

**Expected behaviour**
I don't know if this is actually valid syntax (can't find a gratis download of the 2008 spec), but if it is, the array slices should be assigned a value. If it isn't, a more helpful error message should be generated.

**Context**
```
******************** GHDL Bug occurred ***************************
Please report this bug on https://github.com/ghdl/ghdl/issues
GHDL release: 0.37-dev (v0.36-14-g0c79f9cf) [Dunoon edition]
Compiled with GNAT Version: 8.2.1 20181127
Target: x86_64-pc-linux-gnu
In directory: /tmp/
Command line:
/usr/lib/gcc/x86_64-pc-linux-gnu/8.3.0/ghdl1 --ghdl--std=08 -P/usr/lib/ghdl/ieee/v08/ -P/usr/lib/ghdl/ -quiet -o ent.s ent.vhd
Exception CONSTRAINT_ERROR raised
Exception information:
raised CONSTRAINT_ERROR : trans-chap3.adb:3058 access check failed
Call stack traceback locations:
0x73d208 0x7618a1 0x7613ee 0x761601 0x76b99a 0x76b495 0x774f4d 0x75c340 0x745b89 0x72a393 0x728eeb 0x65686b 0xb83f3e 0x64f055 0x65b6eb 0x79d64b 0x64d541 0x7f4b5c7ad221 0x65148c 0xfffffffffffffffe
******************************************************************

Execution of /usr/lib/gcc/x86_64-pc-linux-gnu/8.3.0/ghdl1 terminated by unhandled exception
raised CONSTRAINT_ERROR : trans-chap3.adb:3058 access check failed
Call stack traceback locations:
0x73d208 0x7618a1 0x7613ee 0x761601 0x76b99a 0x76b495 0x774f4d 0x75c340 0x745b89 0x72a393 0x728eeb 0x65686b 0xb83f3e 0x64f055 0x65b6eb 0x79d64b 0x64d541 0x7f4b5c7ad221 0x65148c 0xfffffffffffffffe
ghdl:error: compilation error
```

- OS: Arch Linux
- Origin:
  - [ ] Package manager. Repo:
  - [ ] Released binaries. Tarball:
  - [x] Built from sources. Commit SHA: 0c79f9cf

**How to reproduce?**
```
#>> ent.vhd
entity ent is
end ent;

architecture a of ent is
	signal sig_x : bit_vector(2 downto 0);
	signal sig_y : bit_vector(2 downto 0);

	signal sig_z : bit_vector(3 downto 0);
begin
	-- works
	(sig_x(1), sig_x(0), sig_y(1), sig_y(0)) <= sig_z;

	-- "raised CONSTRAINT_ERROR : trans-chap3.adb:3058 access check failed"
	(sig_x(1 downto 0), sig_y(1 downto 0)) <= sig_z;
end a;

#>> sim.sh
ghdl -a --std=08 ent.vhd
ghdl --elab-run --std=08 ent

#>> run.sh
docker run --rm -tv /$(pwd):/src:z -w //src ghdl/ghdl:buster-mcode sh -c ./sim.sh

#>> end
```

**Checklist**
Before submitting your issue, please review the following checklist:

- [x] Add `GHDL Bug occurred` log block
- [x] Add a MWE
- [x] Try the latest version

1138-4EB, 2019-03-25:
It might be worth trying to mix this with #707. So: `(sig_x(my_slice), sig_y(my_slice)) <= sig_z;` where `subtype my_slice is integer range 1 downto 0;`. I don't know if it is valid, but I would expect someone to think about using it (just as I did after seeing your code).

diogratia, 2019-03-25:
Actually subtype my_slice does't work. There is a work around:

```vhdl
entity ent is
end ent;

architecture a of ent is
    signal sig_x : bit_vector(2 downto 0);
    signal sig_y : bit_vector(2 downto 0);

    signal sig_z : bit_vector(3 downto 0);
    -- ADDED pre 1318-4EB:
    -- subtype my_slice is integer range 1 downto 0;
    alias sig_x_low: bit_vector(1 downto 0) is sig_x(1 downto 0);
    alias sig_y_low: bit_vector(1 downto 0) is sig_x(1 downto 0);
begin
    -- works
    -- (sig_x(1), sig_x(0), sig_y(1), sig_y(0)) <= sig_z;

    -- "raised CONSTRAINT_ERROR : trans-chap3.adb:3058 access check failed"
    -- (sig_x(1 downto 0), sig_y(1 downto 0)) <= sig_z;
    -- DITTO:
    -- (sig_x(my_slice), sig_y(my_slice)) <= sig_z;
    (sig_x_low, sig_y_low) <= sig_z; -- WORKS
end a;
```

Note that type `bit_vector` isn't a resolved type and the first assignment (-- works) is also commented out. Leaving it in results in an error:
``` none
ghdl -r --std=08 ent
for signal: .ent(a).sig_x(1)
./ent:error: several sources for unresolved signal
./ent:error: error during elaboration
```

Using the aliases should provide a clue as to what's occurring.

There's also the -2002 and earlier (-1993) method:

```vhdl
architecture b of ent is
    signal sig_x : bit_vector(2 downto 0);
    signal sig_y : bit_vector(2 downto 0);

    signal sig_z : bit_vector(3 downto 0);
    type twobit is array (0 to 1) of bit_vector(1 downto 0);
begin
    (sig_x(1 downto 0), sig_y(1 downto 0)) <=
                  twobit'(sig_z(3 downto 2), sig_z(1 downto 0));
end architecture b;
```

which provides a matching element type for the the right hand side waveform expression and the the target aggregate association elements. This is also compatible with `--std=08`. It's a right hand side solution as opposed to the working left hand side solution in the code example.

If you were to compare IEEE Std 1076-2008 10.5.2 Simple signal assignments, 10.5.2.1 General paragraphs 3 & 4 to IEEE Std 1076-1993 8.4 Signal assignment statement paragraphs 4 & 5:

[image]

you'd see paragraph 3 has lost a restriction in -2008 that should then enable the assignment to the aggregate to work (in -1993 "The base type of the value component of each transaction produced by a waveform element on the right-hand side must be the same as the base type of the aggregate.").  

There's not a large trail of bread crumbs for the tool implementor and your code example will provide a useful test case (with the first assignment at least commented out).

tgingold, 2019-03-26:
I can reproduce the bug.  Investigating.

1138-4EB, 2019-03-26:
@diogratia thanks a lot. I must say it's always a pleasure to read your replies.

commit a15dff39 (2021-09-12): trans-chap8: fix crash for slice in target aggregate.  Fix #786
closed 2021-09-12 by tgingold