ghdl/ghdl #789: Bound check failure when working with real values

stdefeber, 2019-03-26:
**Description**
We have a proprietary module which calculates with real values.
Everywhere in the code GHDL gives bound check errors when running from start (0 ps).
The issue is not present in ghdl 0.34-dev 32-bit, source package, debian.

- OS:
- Origin:
  - [ ] Package manager. Repo:
  - [ ] Released binaries. Tarball:
  - [ ] Built from sources. Commit SHA:

tgingold, 2019-03-26:
Without reproducer, it is hard to figure out the reason.

stdefeber, 2019-04-10:
As stated it is a proprietary module, however I will try to make something which produces the same error.

stdefeber, 2019-04-11:
I have created the following failing example.

do_some_math.vhdl
-----------------------------------------------------------------------------
library ieee;
use     ieee.std_logic_1164.all;
use     ieee.numeric_std.all;

entity do_some_math is
port( 
  clk_i     : in  std_logic;
  rstn_i    : in  std_logic;
  data_out  : out real;
  data_in   : in  real;
  vref      : in  real
        
);
end entity;

architecture behav of do_some_math is

  type vector_of_real is array (natural range <>) of real;

  signal product : vector_of_real(0 to 1);
  signal sum     : real;

begin

  p_tap : process(clk_i, rstn_i)
  begin
    if (rstn_i = '0') then
      product(0) <= 0.0;
      product(1) <= 0.0;
    elsif rising_edge(clk_i) then 

      product(0) <= data_in * 0.3;
      product(1) <= product(0) * 0.1; 
    end if;
  end process p_tap; 

  sum  <= product(0) + product(1);

  data_out <= sum/1.40*vref;

end behav;
------------------------------------------------------------------------------------------

do_some_math_tb.vhdl
------------------------------------------------------------------------------------------
library ieee;
use     ieee.std_logic_1164.all;
use     ieee.numeric_std.all;

entity do_some_math_tb is
end entity;

architecture behav of do_some_math_tb is

  component do_some_math is
  port( 
    clk_i     : in  std_logic;
    rstn_i    : in  std_logic;
    data_out  : out real;
    data_in   : in  real;
    vref      : in  real
  );
  end component;

  signal clk   : std_logic;
  signal rstn  : std_logic;
  signal vref  : real;
  signal data_out : real;
  signal data_in  : real;

begin

  p_clk : process
  begin
    clk <= '0';
    wait for 10 ns;
    clk <= '1';
    wait for 10 ns;
  end process p_clk;

  p_rstn : process
  begin
    rstn <= '0';
    wait for 205 ns;
    rstn <= '1';
    wait;
  end process;      

  
  u_do_some_math : do_some_math
  port map( 
    clk_i     => clk,      -- : in  std_logic;
    rstn_i    => rstn,     -- : in  std_logic;
    data_out  => data_out, -- : out real;
    data_in   => data_in,  -- : in  real;
    vref      => vref      -- : in  real
  );

  p_stim : process
  begin
    data_in  <= 0.0;
    vref     <= 3.3;
    wait until rstn = '1';
    wait until rising_edge(clk);
    data_in <= 0.1;
    wait until rising_edge(clk);
    data_in <= 0.1;
    wait until rising_edge(clk);
    wait;
  end process;

end behav;
-----------------------------------------------------------------------------------------

Compile and run commands used
ghdl -a do_some_math.vhdl
ghdl -a do_some_math_tb.vhdl
gdhl -m do_some_math_tb
ghdl -r do_some_math_tb

diogratia, 2019-04-11:
There are a couple of report statements that you can add in a process that will explain the bounds error.

```vhdl
-- sum <= product(0) + product(1);
sum_process:
    process (product) is
    begin
        report "real'low =  " & real'image(real'low);
        report "real'high = " & real'image(real'high);
        report "real'low + real'low = " & real'image(real'low + real'low);
        sum <= product(0) + product(1);
    end process;
```
and analyzing:
```none
ghdl -a do_some_math.vhdl
do_some_math.vhdl:43:63:error: static expression violates bounds
```
The expression ` real'image(real'low + real'low);` is locally static and doesn't fit with the bounds.

You can see the bounds by commenting out the report statement containing the static expression,  analyzing, elaborating and running the testbench which gives:

```none
ghdl -r do_some_math_tb
do_some_math.vhdl:41:9:@0ms:(report note): real'low =  -1.7976931348623157e308
do_some_math.vhdl:42:9:@0ms:(report note): real'high = 1.7976931348623157e308
./do_some_math_tb:error: bound check failure at do_some_math.vhdl:43
./do_some_math_tb:error: simulation failed
```

Consider the default value real'low. Adding itself to itself gives you a value out of bounds of type real.  It's a valid error.

Provide default initial values for the real elements of your array type (e.g. (`others => 0.0`) which yields a value within the bounds of real when added to itself. 

Your previous 32 bit ghdl-0.34 was not exhibiting correct semantic behavior (here shown during both analysis and simulation).

stdefeber, 2019-04-11:
@diogratia, Thanks for your reply.

In the mean time I did some trials myself.
`  sum  <= to_real(to_float(product(0)) + to_float(product(1))); `

and 
  
` data_out <= to_real(to_float(sum)/(1.40*to_float(vref))); `

Seems to result in the correct behaviour.

diogratia, 2019-04-11:
The expression 1.4 * vref would be out of bounds if vref equal real'low as an initial default value. That could be cured by a default value for vref as well (avoiding division by 0 in the expression sum/1.40*vref the value would be non zero). 

It isn't clear where the function to_real comes from, there's one found in package float_pkg (-2008).  I take it you'd be relying on IEEE Std 754-1985 or  IEEE Std 854-1987 to avoid a bounds error. (Neither standard is mentioned in IEEE Std 1076-2008 Annex J Bibliography, 5.2.5 Floating-point types.)

It seems simpler to give vref a safe non-zero initial value.

stdefeber, 2019-04-18:
Issue is resolved.

closed 2019-04-18 by stdefeber