ghdl/ghdl #790: Option to Remove "NUMERIC_STD.TO_INTEGER: metavalue detected, returning 0"

Bamban, 2019-04-01:
**Is your feature request related to a problem? Please describe.**
Add a clear and concise description of what the problem is. E.g. *I'm always frustrated when [...]*

**Describe the solution you'd like**
Add a clear and concise description of what you want to happen.

**Describe alternatives you've considered**
Add a clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or screenshots about the feature request here.

https://sourceforge.net/p/ghdl-updates/tickets/106/

1138-4EB, 2019-04-01:
Hi @Bamban! That's documented here: https://ghdl.readthedocs.io/en/latest/using/Simulation.html#cmdoption-ghdl-assert-level. For the specific warning you mention, you might want to use `--ieee-asserts`: https://ghdl.readthedocs.io/en/latest/using/Simulation.html#cmdoption-ghdl-ieee-asserts.

Bamban, 2019-04-02:
My bad, I haven't looked closely enough!
Thanks for the pointer.

closed 2019-04-02 by Bamban