ghdl/ghdl #793: Unused, undefined foreign function makes elaboration fail

umarcor, 2019-04-07:
**Is your feature request related to a problem? Please describe.**

In the MWE below, both `get_param` and `get_addr` are defined as foreign functions (`VHPIDIRECT`). Both of them have placeholder assertions that are expected to make the execution fail at runtime, if no C implementation is provided during elaboration. However, it fails during elaboration, not at runtime.

**Describe the solution you'd like**

These are the use cases I am trying to evaluate:

- **C implementations** are **not provided and functions** are **not used** in the testbench. No error should be produced. Although I'd expect two warnings to be shown during elaboration.
- **C implementations** are **not provided and functions** are **used** in the testbench. I'd expect two warnings to be shown during elaboration and an (assertion) error to be produced at runtime.
- **C implementations** are **provided** (**no matter if functions are used** or not). No warnings should be shown during elaboration and no assertion error should be produced at runtime.

**Describe alternatives you've considered**

Note that the error during elaboration is produced by `get_addr`, because GHDL expects it to be used in `buffet_t_prot.init(i: integer)`, even though this is not true. However, `get_param` does not produce any error, because GHDL is sure about it not being used at all. I think that none of them should fail.

An alternative I have considered is to provide two different sources, one without foreign functions at all, and another one that includes them along with the 'default' version. However, I'd lke to avoid it, as this would require recompiling multiple sources that depen on this.

**Additional context**

The context of this request is VUnit/vunit#462. In order to support different implementations of certain functions, some relying on foreign functions and some not, a variable is checked in an if-then-else statement. If the value implies that the design was elaborated with C implementations, the foreign functions are used. Otherwise, they are not. Therefore, I need GHDL to let the user elaborate a design with foreign functions without an actual C implementation, as long as he/she knows that those functions will not be used.

---

``` bash
#>> run.sh

cat > pkg_c.vhd <<-EOF
package pkg_c is
  function get_param(f: integer) return integer;
    attribute foreign of get_param :
      function is "VHPIDIRECT get_param";

  type buffer_t is array(integer range <>) of integer;
  type buffer_p is access buffer_t;

  impure function get_addr(f: integer) return buffer_p;
    attribute foreign of get_addr :
      function is "VHPIDIRECT get_addr";

  type buffet_t_prot is protected
    procedure init ( i: integer );
  end protected buffet_t_prot;
end pkg_c;

package body pkg_c is
  function get_param(f: integer) return integer is begin
    assert false report "VHPI" severity failure;
  end get_param;

  impure function get_addr(f: integer) return buffer_p is begin
    assert false report "VHPI" severity failure;
  end get_addr;

  type buffet_t_prot is protected body
    variable var: buffer_p;
    procedure init ( i: integer ) is begin
      var := get_addr(i);
    end procedure;
  end protected body buffet_t_prot;
end pkg_c;
EOF

cat > tb.vhd <<-EOF
use work.pkg_c.all;

entity test is
end entity;

architecture tb of test is begin
  main: process begin
    report "HELLO" severity warning;
    wait;
  end process;
end architecture;
EOF

$(command -v winpty) docker run --rm -itv /$(pwd)://src -w //src ghdl/ghdl:buster-mcode bash -c "\
  ghdl -a --std=08 pkg_c.vhd tb.vhd &&\
  ghdl -e --std=08 test"

#>> end
```

tgingold, 2019-04-08:
Currently, the mcode backend doesn't support interfacing with external C code.  You should use the gcc or llvm backend.
You can check which backend is used by using `ghdl --version`.

umarcor, 2019-04-08:
Oh, that was a failure on my side when writting the MWE here. I found this issue with LLVM backend, and that's the backend I am most interested on ATM.

I updated the first comment to show the output with LLVM (`ghdl/ghdl:ubuntu18-llvm-5.0`) too.

tgingold, 2019-04-08:
With llvm or gcc, you can try to compile with `-ffunction-sections -Wl,--gc-sections`.
But what you are requesting is not supported.  After a package is compiled, the same object file is linked whether the function is used or not.

umarcor, 2019-04-08:
> With llvm or gcc, you can try to compile with `-ffunction-sections -Wl,--gc-sections`.

`ghdl -e --std=08 -ffunction-sections test` produces the same result and `ghdl -e --std=08 -ffunction-sections -Wl,--gc-sections test` is not recognized: `gcc: error: unrecognized command line option '--gc-sections'; did you mean '--data-sections'?`.

> But what you are requesting is not supported. After a package is compiled, the same object file is linked whether the function is used or not.

So, it is not possible in this case below to ignore `attribute foreign` and have `assert false report "VHPI" severity failure;` compiled, unless a C object containing `get_param` is provided?

What is the suggested approach? Should I use some preprocessor to comment line `attribute foreign of get_param : function is "VHPIDIRECT get_param";` conditionally?

```
cat > pkg_c.vhd <<-EOF
package pkg_c is
  function get_param(f: integer) return integer;
    attribute foreign of get_param : function is "VHPIDIRECT get_param";
end pkg_c;

package body pkg_c is
  function get_param(f: integer) return integer is begin
    assert false report "VHPI" severity failure;
  end get_param;
end pkg_c;
EOF

cat > tb.vhd <<-EOF
use work.pkg_c.all;

entity test is
end entity;

architecture tb of test is begin
  main: process
    variable t: integer;
  begin
    report "HELLO" severity warning;
	t := get_param(0);
    wait;
  end process;
end architecture;
EOF

docker run --rm -tv /$(pwd)://src -w //src ghdl/ghdl:ubuntu18-llvm-5.0 bash -c "\
  ghdl -a --std=08 pkg_c.vhd tb.vhd &&\
  ghdl -e --std=08 test && ./test"
```

tgingold, 2019-04-09:
Sorry, try with `-Wl,-Wl,--gc-sections`.

I would suggest that you always link with a C file, possibly providing dummy definitions.
You can also write FOREIGN declarations in a very specific package and use it only when needed.
But ghdl cannot know whether or not the C implementation is provided.

umarcor, 2019-04-09:
> Sorry, try with `-Wl,-Wl,--gc-sections`.

Still not working. Anyway, I will stop trying to achieve it this way, as it is a not supported approach: "*But ghdl cannot know whether or not the C implementation is provided*". I'd like to go with a more canonical solution.
 
> I would suggest that you always link with a C file, possibly providing dummy definitions.

Unfortunately, this is not possible and the point I am trying to avoid. This is because requiring a C file prevents using simulators that do not support VHPI (such as GHDL with mcode backend).

For example, in a modified version of example [array_axis_vcs](https://github.com/dbhi/vunit/blob/feat-foreign-c/examples/vhdl/array_axis_vcs) from VUnit, I am using `main.c` for the testbench where VHPI is used, and `stubs.c` for the testbench where VHPI is not used. Ideally, I could replace these lines with some option that 'disables' VHPI: https://github.com/dbhi/vunit/blob/feat-foreign-c/examples/vhdl/array_axis_vcs/run.py#L25-L28.

> You can also write FOREIGN declarations in a very specific package and use it only when needed.

I would define two declarations and a single body of the package. However, it is not possible to select one of them through a compilation option or a generic, is it? I.e., the declarations must be saved in separate files and the build script must select one of them. E.g.:

```
package extmemory_pkg is
  procedure write_byte ( id, i, v: integer ) ;
    attribute foreign of write_byte : procedure is "VHPIDIRECT write_byte";

  impure function read_byte ( id, i: integer ) return integer;
    attribute foreign of read_byte : function is "VHPIDIRECT read_byte";
end package;
```

```
package extmemory_pkg is
  procedure write_byte ( id, i, v: integer ) ;
  impure function read_byte ( id, i: integer ) return integer;
end package;
```

```
package body extmemory_pkg is
  procedure write_byte ( id, i, v: integer ) is begin
    assert false report "VHPI" severity failure;
  end;
  impure function read_byte ( id, i: integer ) return integer is begin
    assert false report "VHPI" severity failure;
  end;
end package body;
```

tgingold, 2019-04-10:
In the future, ghdl with mcode back-end should support `VHPIDIRECT` using a shared library.

For the second point, yes you need the build script to select the right files.

umarcor, 2019-04-10:
Done and tested (VUnit/vunit#470). Thank you!

closed 2019-04-10 by umarcor

umarcor, 2020-07-28:
@tgingold, while reading the LRM (2008), I found the following in *20.2.4.1 General*:

> NOTE 2—An implementation may, as part of elaboration of a 'FOREIGN attribute specification whose value is of the
form described in this subclause (20.2.4), perform certain checks, for example, that the C library exists or that the foreign model implementation functions exists.

Does it mean the checking that the foreign function implementation exists is optional but not required? I.e., may the following snippet be added to some VHDL without producing an elaboration failure if the C implementation of myfunction is not provided?

```
  function myfunction return integer is
  begin report "VHPIDIRECT myfunction" severity failure; end;
  attribute foreign of rand : function is "VHPIDIRECT myfunction";
```

tgingold, 2020-07-29:
Because it's a "may", that's a permission.

With GHDL, yes the function must exist even when not used (unless you use the mcode backend).