ghdl/ghdl #795: Crashing during analysis of vhdl entity using package with protected type

zznaki, 2019-04-09:
**Description**
Crashing during analysis of  program using package with protected type 

**Expected behaviour**
Complitation to complete either with no errors or representative syntax error message

**Context**
A modified package taken from github to read csv files is incoropoertaed into an entity in a second file. When this entity is included is a subsequent entity as a component the crash occurs.

All 3 items are compile into library work with the command line:
ghdl -a --std=02 --ieee=synopsys --workdir=. --work=work [sorucefile]

Compilation of top level items after the other items results in:

******************** GHDL Bug occurred ***************************
Please report this bug on https://github.com/ghdl/ghdl/issues
GHDL release: 0.37-dev (v0.36-26-g88830edd) [Dunoon edition]
Compiled with GNAT Version: Community 2018 (20180523-73)
Target: x86_64-apple-darwin16.7.0
In directory: [localdir]/simulation/ghdllib-llvm/
Command line:
/opt/local/bin/ghdl1-llvm --std=02 --workdir=. --work=work -P/opt/local/lib/ghdl/synopsys/v93/ -P/opt/local/lib/ghdl/ -c -o ./tbCrashExample2.o ../../stimulus/tbCrashExample2.vhd
Exception SYSTEM.ASSERTIONS.ASSERT_FAILURE raised
Exception information:
raised SYSTEM.ASSERTIONS.ASSERT_FAILURE : trans-chap6.adb:1084
Load address: 0x10bba9000
Call stack traceback locations:
0x10c669b21 0x10bda2e18 0x10bda32b3 0x10bd4bfe5 0x10bd4c0d2 0x10bd4c5a0 0x10bd4cb4d 0x10bd4d89e 0x10bd43fea 0x10bd3f499 0x10bd3fdb9 0x10bdc69c7 0x10bd0f391 0x10bdcbe96 0x10bdcc6df 0x10bdcdf9f 0x10bbab842
******************************************************************
- OS: Mac OSX 10.14.3
- Origin:
 - Built from sources. Commit SHA: 88830edd (8 April 2019 at 18:51:49 CEST)

**Additional context**
I believe the issue could be connected to https://github.com/ghdl/ghdl/issues/788 as originally I was getting this failure in similar circumstances.

**How to reproduce?**
I have produced 3 items which should go into individual files
csv_file_reader_pkg (package) 
crashexample (entity)
tbcrashexample2 (entity, toplevel)

Compilation in sequence using:
ghdl -a --std=02 --ieee=synopsys --workdir=. --work=work [sorucefile]
Will cause the crash when compiling tbcrashexample2

```
#>> tbCrashExample2.vhd
library ieee;
use ieee.std_logic_1164.all;
use ieee.numeric_std.all;
use work.all;
entity tbCrashExample2 is
  generic (
  	vectorFilename : string
  );
end entity tbCrashExample2;

architecture BEHAVORIAL of tbCrashExample2 is
	--! General Simulation variables/signals
	signal fVctrEnd			: boolean; 			--! Flag from vector reader to indicate file end
	signal vctrCmt			: string(1 to 512);	--! Current vector line comment
  	signal syncPulse		: std_logic;		--! Main Sync Signal
  	signal syncCount 		: integer;			--! Sync count
  	signal val1 			: std_logic_vector(15 downto 0);
  	signal val2 			: std_logic_vector(15 downto 0);
  	signal val3 			: std_logic_vector(15 downto 0);
begin
	cmp_TestVector : entity crashExample(BEHAVIORAL) 
		port map (  vctrFlNm 	=> vectorFilename,
[... 270 lines cut ...]
						if read_delim and read_char = '"' then
							index      := -1;
							read_delim := false;
						elsif not read_delim and read_char = ',' then
							return return_string;
						else
							return_string(index) := read_char;
							index                := index + 1;
						end if;
				end case;
				read(current_line, read_char, read_ok);
			end loop;
			return "Exit";
		end;
	end protected body;

end;

#>> sim.sh
ghdl -a --std=02 --ieee=synopsys --workdir=. --work=work csv_file_reader_pkg.vhd
ghdl -a --std=02 --ieee=synopsys --workdir=. --work=work CrashExample2.vhd
ghdl -a --std=02 --ieee=synopsys --workdir=. --work=work tbCrashExample2.vhd

#>> end
```

tgingold, 2019-04-09:
The crash is due to this line:
```
port map (  vctrFlNm 	=> vectorFilename,
```
As a workaround, you can use a generic.

diogratia, 2019-04-09:
Another workaround would be:
```vhdl
		port map (  vctrFlNm 	=> vectorFilename(1 to vectorFilename'LENGTH),
```
where the bounds are explicitly provided by the slice name.

zznaki, 2019-04-10:
Thanks for both of these - thinking about it using an unconstrained type on a port was asking for trouble.

tgingold, 2019-04-10:
It is an oversight.  Unconstrained ports work when associated with a signal, but not with an expression.

But the use of vctrFlNm as a port is unusual; it is usually a generic.

tgingold, 2021-09-07:
Not crashing anymore.

closed 2021-09-07 by tgingold