ghdl/ghdl #796: System version number in Object Files built of MacOSX incorrect

zznaki, 2019-04-10:
**Description**
While elaborating the example the example for https://github.com/ghdl/ghdl/issues/795 with the first work around the linker provides the following warning:

ld: warning: object file (crashExample2.o) was built for newer OSX version (18.2) than being linked (10.14)

This is generated for all object files used in the simulation code. 

**Expected behaviour**
As ghdl and the simulator were built on the same laptop, no warnings should be produced. 
Also there is no version of Mac OSX 18.2 so it must be picking this info up from something else.

**Context**

Command line:
 ghdl -e --std=02 --ieee=synopsys  --workdir=. --work=work tbCrashExample2

Mac OSX 10.14.3

tgingold, 2019-04-10:
What is the output of `ghdl --version` ?

zznaki, 2019-04-10:
Apologies for not supplying the info.

> ghdl --version
GHDL 0.37-dev (v0.36-26-g88830edd) [Dunoon edition]
 Compiled with GNAT Version: Community 2018 (20180523-73)
 llvm code generator
Written by Tristan Gingold.

Copyright (C) 2003 - 2019 Tristan Gingold.
GHDL is free software, covered by the GNU General Public License.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

zznaki, 2019-04-10:
I should say this is a cosmetic issue. The linker seems to function perfectly otherwise and simulations can be run,

tmeissner, 2019-04-10:
I know these messages too. They occurred first years ago. I assumed it is caused by a difference between macOS version and the version of the included Darwin kernel. But I don’t know exactly....

tgingold, 2019-04-10:
Did you build llvm by yourself ?

The interesting files in llvm are :
* lib/Support/Unix/Host.inc
* lib/MC/MCStreamer.cpp

tmeissner, 2019-04-10:
I used LLVM 3.5.1 installed with homebrew.

diogratia, 2019-04-11:
There's a bug that was fixed in llvm-3.8 where the darwin builds looked at the wrong field for determining linking constraints (see llvm [Bug 24927 - LC_VERSION_MIN_MACOSX: Darwin kernel version vs SDK version ](https://bugs.llvm.org/show_bug.cgi?id=24927) and [TargetTriple issue: LC_VERSION_MIN_MACOSX: Darwin kernel version vs SDK version](http://lists.llvm.org/pipermail/llvm-dev/2015-September/090758.html) on the lvm-devl email list).

The bug shows up when using a newer link editor (ld64) supplied by Xcode (or Xcode command line tools) which depends on the supplied information to be in the correct place.  (It's not an Apple bug.)

The above links provide a patch you could apply with a little research. 

The proper thing to do might be to use llvm-3.8 or later. If lacking a Homebrew formula there's also a binary install available from llvm.org. Their version may require a full Xcode installation, due to how it resolves SDKs by peeking into the Xcode.app bundle. I haven't verified the interaction between newer MacOS Xcode command line installations and later version llvm.org distributed binaries. My 2008 Aluminum Macbook has been obsoleted and will only run up to <strike>OS X</strike> MacOS 10.11.6. (It's lasted way longer than a plastic case notebook computer, no broken solder balls caused by logic board flexing.)

I built the [v0.36](https://github.com/ghdl/ghdl/releases/tag/v0.36) llvm binary release image [ghdl-0.36-macosx-llvm.tgz](https://github.com/ghdl/ghdl/releases/download/v0.36/ghdl-0.36-macosx-llvm.tgz) using the full Xcode install version 8.3.1  build 8C10002 - the last available for 10.11.6, clang+llvm-3.8.0-x86_64-apple-darwin binary from llvm-org and Adacore open gnat-gpl-2015-x86_64-darwin-bin binary. The release should be compatible with MacOS releases 10.9+.

zznaki, 2019-04-11:
Hi,  I used a pre-build binary of llvm 3.5.2 from llvm.org - I took it as it supported debugging.
May other new versions are listed on the GHDL building page - is there a prefered later version?
Thanks for the help, Regards

tmeissner, 2019-04-17:
I've successfully built GHDL with llvm 5.0.2 (installed with homebrew). I can confirm that the warning messages are gone :)

Thanks for your help @diogratia :)

closed 2019-05-03 by tgingold