ghdl/ghdl #797: raised CONSTRAINT_ERROR : trans-chap7.adb:3792 access check failed

umarcor, 2019-04-15:
**Description**

I am trying to get/set the bytes in a `uint8_t` array defined in C through VHPIDIRECT. If I declare an access type to `string(1 to integer'high)`, it works. However, I'd like the access type to be of an unconstrained type (`string`). This either fails or produces a crash.

**Expected behaviour**

In the MWE below, this is the expected and correct output when variables `a` are used:

``` bash
# ./run.sh
0: 10
1: 20
2: 30
3: 0
4: 0
5: 0
tb.vhd:12:5:@0ms:(report note): HELLO
0: 10
1: 20
2: 30
3: 11
4: 21
5: 31
```

When variables `b` are used, I'd expect the same behaviour, but I get an error instead:

``` bash
# ./run.sh
0: 10
1: 20
2: 30
3: 0
4: 0
5: 0
tb.vhd:12:5:@0ms:(report note): HELLO
./test:error: bound check failure at pkg_c.vhd:55
./test:error: simulation failed
```

I.e., I can retrieve a pointer to the array and assign it to an access of string, but an error is produced when trying to use the elements in the array, because the range is not defined.

**Context**

When variables `c` are used (trying to define the range), GHDL crashes:

``` bash
# ./run.sh

******************** GHDL Bug occurred ***************************
Please report this bug on https://github.com/ghdl/ghdl/issues
GHDL release: 0.36-rc1 (v0.36-rc1-4-gb4597836) [Dunoon edition]
Compiled with GNAT Version: 7.3.0
Target: x86_64-linux-gnu
In directory: /src/
Command line:
/usr/local/bin/ghdl1-llvm --std=08 -P/usr/local/lib/ghdl/ieee/v08/ -P/usr/local/lib/ghdl/ -c -fpic -o pkg_c.o pkg_c.vhd
Exception CONSTRAINT_ERROR raised
Exception information:
raised CONSTRAINT_ERROR : trans-chap7.adb:3792 access check failed
Call stack traceback locations:
0x5d7a04 0x604e79 0x604c3e 0x581fb8 0x598214 0x598361 0x620241 0x5bf2ae 0x621450 0x56e1b3 0x62f69d 0x62ff01 0x6317b3 0x408fb7 0x7f2869826b95 0x408388 0xfffffffffffffffe
******************************************************************
ghdl:error: compilation error
```

**Additional context**

The motivation to access the external buffer through `byte_vector_access_t` is to be able to manipulate an array of `type byte_vector_access_vector_t is array (natural range <>) of byte_vector_access_t` where some elements are pointers to strings allocated in the testbench (GHDL) and others are previously allocated in the main C function.

**How to reproduce?**

```
#>> run.sh
cat > pkg_c.vhd <<-EOF
package pkg_c is

  type byte_vector_access_t is access string;
  type extbuf_access_t is access string(1 to integer'high);

  impure function
  get_addr(
    id : integer
  ) return extbuf_access_t;
  attribute foreign of get_addr : function is "VHPIDIRECT get_addr";

  impure function
  get_baddr(
    id : integer
  ) return byte_vector_access_t;
  attribute foreign of get_baddr : function is "VHPIDIRECT get_baddr";
  
  procedure
  set(
    index : natural;
    value : natural
  );
  
  impure function
  get(
    index : natural
  ) return natural;
end pkg_c;

package body pkg_c is
  impure function
  get_addr(
    id : integer
  ) return extbuf_access_t is begin
    assert false report "VHPI get_addr" severity failure;
  end;
  
  impure function
  get_baddr(
    id : integer
  ) return byte_vector_access_t is begin
    assert false report "VHPI get_baddr" severity failure;
  end;
  
  procedure
  set(
    index : natural;
    value : natural
  ) is
    variable a : extbuf_access_t := get_addr(0);
    variable b : byte_vector_access_t := get_baddr(0);
    variable c : byte_vector_access_t(1 to integer'high) := get_baddr(0);
  begin
    a(index+1) := character'val(value);
    --b(index+1) := character'val(value);
    c(index+1) := character'val(value);
  end;
  
  impure function
  get(
    index : natural
  ) return natural is
    variable a : extbuf_access_t := get_addr(0);
    variable b : byte_vector_access_t := get_baddr(0);
    variable c : byte_vector_access_t(1 to integer'high) := get_baddr(0);
  begin
    return character'pos(a(index+1));
    --return character'pos(b(index+1));
    return character'pos(c(index+1));
  end;
end pkg_c;
EOF

cat > tb.vhd <<-EOF
use work.pkg_c.all;

entity test is
end entity;

architecture tb of test is
  constant block_len : natural := 3;
begin
  main: process
    variable val: integer;
  begin
    report "HELLO" severity note;
	for x in 0 to block_len-1 loop
      val := get(x);
      set(block_len+x, val+1);
    end loop;
    wait;
  end process;
end architecture;
EOF

cat > main.c <<-EOF
#include <stdio.h>
#include <stdlib.h>
#include <stdint.h>

extern int ghdl_main (int argc, char **argv);

uint8_t *D[1];
uintptr_t get_addr(uint8_t id) { return (uintptr_t)D[id]; }
uintptr_t get_baddr(uint8_t id) { return get_addr(id); }

int main(int argc, char **argv) {
  const uint32_t length = 3;
  D[0] = (uint8_t *) malloc(2*length*sizeof(uint8_t));
  if ( D[0] == NULL ) {
    perror("execution of malloc() failed!\n");
    return -1;
  }
  int i;
  for(i=0; i<length; i++) { D[0][i] = (i+1)*10; }
  for(i=0; i<2*length; i++) { printf("%d: %d\n", i, D[0][i]); }
  ghdl_main(argc, argv);
  for(i=0; i<2*length; i++) { printf("%d: %d\n", i, D[0][i]); }
  free(D[0]);
  return 0;
}
EOF

docker run --rm -tv /$(pwd)://src -w //src ghdl/ghdl:ubuntu18-llvm-5.0 bash -c "\
  ghdl -a --std=08 pkg_c.vhd tb.vhd &&\
  gcc -fPIC -rdynamic -c main.c -o main.o &&\
  ghdl -e --std=08 -Wl,main.o test && ./test"

#>> end
```

**Checklist**
Before submitting your issue, please review the following checklist:

- [x] Add `GHDL Bug occurred` log block
- [x] Add a MWE
- [x] Try the latest version

tgingold, 2019-04-15:
Don't use access to unconstrained array with VHPI; these are fat pointers whose representation is not easy to describe in C.

However, it shouldn't crash during analysis.

umarcor, 2019-04-16:
> Don't use access to unconstrained array with VHPI; these are fat pointers whose representation is not easy to describe in C.

Certainly, I don't want to access unconstrained arrays with VHPI. I understand that trying to do so is looking for problems. I will always retrieve a `uintptr_t` from C, which maps cleanly to a constrained access type. I opened this issue because, as you say, case 'c' should fail as 'b', not crash.

Regarding my specific use case, I need a type conversion between `byte_vector_access_t` and `extbuf_access_t`. But I am afraid that it is not possible with neither VHDL 1993 nor VHDL 2008 and probably not even with VHDL 2019. Not straighforward, at least. This is because `string` and `string(1 to integer'high)` are closely related, but for some reason the accesses to these types are not. Moreover, I found no mechanism to save the 'pointer' from `extbuf_access_t` to some 64 bit type and then assign that value to `byte_vector_access_t`.

> NOTE: I can confirm that GHDL throws a legit error if I try to assign a variable of type `byte_vector_access_t` to another one of type `extbuf_access_t` or vice versa.

As commented in 'additional context' above, this is the use case:

``` vhdl
type byte_vector_access_t is access string;
type extbuf_access_t is access string(1 to integer'high);
type byte_vector_access_vector_t is array (natural range <>) of byte_vector_access_t;

variable ptrs: byte_vector_access_vector_t(0 to 1) := (
  0 => new string'(1 to 1024 => character'val(0)),
  1 => get_addr(0)
);
```

So, `ptrs(0)` is an access to an 'internal' string and `ptrs(1)` is an access to an 'external' string. But the same procedures/functions can be used to manipulate them.

tgingold, 2019-04-16:
Do you really need to access to an external string or can you copy it ?

It is possible to describe a fat pointer in C (it is done in the grt run time library), but this is not portable at all.

umarcor, 2019-04-16:
I do need to access an external array of bytes, which I am describing as a string, so that each address points to a byte in C. Precisely, I am trying to avoid having the content duplicated in two identical buffers/arrays.

Currently, I can achieve the functionality I need by using two separate arrays of pointers, separate types, duplicated functions/procedures, etc. One is used to store accesses to variables of type unconstrained array and another one for externally defined arrays. For example: https://github.com/dbhi/vunit/blob/feat-external-arrays/vunit/vhdl/data_types/src/byte_vector_ptr_pkg-body-200x.vhd#L40-L71.

The original motivation that lead to the tests described above is: https://github.com/dbhi/vunit/blob/feat-external-arrays/vunit/vhdl/data_types/src/byte_vector_ptr_pkg-body-200x.vhd#L99-L109

``` vhdl
reallocate_ptrs(st.ptrs, st.ptr);
st.ptrs(st.ptr) := new string'(0 to len-1 => character'val(value));
st.ptr := st.ptr + 1;

reallocate_eptrs(st.eptrs, st.eptr);
st.eptrs(st.eptr) := get_addr(id-1);
st.eptr := st.eptr + 1;
```

As you see, the unconstrained string is constrained when the access variable is created: `new string'(0 to len-1 => character'val(value))`. So, from my naive point of view, `st.ptrs` and `st.eptrs` are both containing accesses to constrained strings, although the underlying types are different.

Nevertheless, if this is a limitation of the language and only possible workaround is to use fat pointers, I'd rather avoid it and just keep the current implementation. I was just hoping that the verbosity could be slightly reduced.

slaweksiluk, 2019-04-16:
@umarcor you might be interested in following Tristan comments https://gitter.im/ghdl1/Lobby?at=5c0fcfa228907a3c7b02aaf7
Are you interested in dynamic allocation of memory in C VHPIDIRECT subprogram and passing it to simulation as VHDL access to unconstrained array? I already have example of such a solution for string access. I can prepare minimal example for you,but be aware, as Tristan said, this very non standard. I'm not even sure if it is safe. In general it looks strange as at the end you don't call VHDL `new` at all:
```vhdl
		main : process
				variable p : string_p;
				type string_p is access string;
		begin
					p := vhpi_get;
					report p.all;
...
```

commit 898ef1b4 (2019-04-16): vhdl: fix crash on access subtype.  Fix #797
closed 2019-04-16 by tgingold