ghdl/ghdl #798: SYSTEM.ASSERTIONS.ASSERT_FAILURE

FranzForstmayr, 2019-04-16:
**Description**
I'm triggering a Bug in GHDL, during a big restructuring in my code. Unfortunately the code itself is confidential, i try to reproduce with a minimal snippet. I'm updating this issue, if i am successful.

**Context**
******************** GHDL Bug occurred ***************************
Please report this bug on https://github.com/ghdl/ghdl/issues
GHDL release: 0.36 (v0.36) [Dunoon edition]
Compiled with GNAT Version: 7.3.0
Target: x86_64-linux-gnu
In directory: /home/ecs/Workspaces/vhdl/axis-fft2/
Command line:
/usr/local/bin/ghdl1-llvm --workdir=/home/ecs/Workspaces/vhdl/axis-fft2/vunit_out/ghdl/libraries/lib --work=lib --std=08 -frelaxed-rules -P/home/ecs/Workspaces/vhdl/axis-fft2/vunit_out/ghdl/libraries/vunit_lib/ -P/home/ecs/Workspaces/vhdl/axis-fft2/vunit_out/ghdl/libraries/osvvm/ -P/home/ecs/Workspaces/vhdl/axis-fft2/vunit_out/ghdl/libraries/JSON/ -P/home/ecs/Workspaces/vhdl/axis-fft2/vunit_out/ghdl/libraries/lib/ -P/home/ecs/Workspaces/vhdl/axis-fft2/vunit_out/ghdl/libraries/PoC/ -P/home/ecs/Workspaces/vhdl/axis-fft2/vunit_out/ghdl/libraries/ro_pkg/ -P/usr/local/lib/ghdl/ieee/v08/ -P/usr/local/lib/ghdl/ -c -o /home/ecs/Workspaces/vhdl/axis-fft2/vunit_out/ghdl/libraries/lib/axis-fft.o /home/ecs/Workspaces/vhdl/axis-fft2/src/axis-fft.vhd
Exception SYSTEM.ASSERTIONS.ASSERT_FAILURE raised
Exception information:
raised SYSTEM.ASSERTIONS.ASSERT_FAILURE : no field Range_Constraint
Call stack traceback locations:
0x7f0c0efeb161 0x447701 0x4aab9f 0x4ab1ed 0x4ab745 0x534bdf 0x549739 0x53d4b5 0x53dcea 0x53e1c9 0x53e33c 0x550a4d 0x526daa 0x526d61 0x526d61 0x527093 0x527ce4 0x527d4a 0x5281a8 0x5284eb 0x558b1d 0x55f6c4 0x51d33f 0x62e9a9 0x62f561 0x630e13 0x408617 0x7f0c0e096b95 0x4079e8 0xfffffffffffffffe
******************************************************************
/usr/local/bin/ghdl: compilation error

- OS:
- Origin:
  - [ ] Package manager. Repo:
  - [ ] Released binaries. Tarball:
  - [ x] Built from sources. Commit SHA: v0.36

**Checklist**
Before submitting your issue, please review the following checklist:

- [x] Add `GHDL Bug occurred` log block
- [ ] Add a MWE
- [ ] Try the latest version

FranzForstmayr, 2019-04-24:
Sorry, I'm unable to reproduce the issue!

tgingold, 2019-04-25:
Let's close this issue then.  If you can reproduce the crash in a different context, feel free to reopen this issue.

Or you can send me the files privately.

closed 2019-04-25 by FranzForstmayr