ghdl/ghdl #799: How to do connectivity testing with ghdl

stdefeber, 2019-04-18:
I have been making modifications to our design and want to do connectivity testing.
Changes trickle down from control registers, via records, to IP blocks way down in the hierarchy. 
So basically, I want to check that a change in a control register has effect on some enable or data line in an IP block. 

- I tried using PSL, however PSL does not seem to handle vectors.
- Hierarchical names do not seem to be supported by GHDL.
- Probe or spy are not supported 

How can I get this observability ???

tmeissner, 2019-04-18:
What do you mean with „PSL doesn’t handle errors“? I have some experience with Assertion based Verification using PSL & SVA, maybe I can help.

stdefeber, 2019-04-19:
Help is much appreciated.

I really mean „PSL doesn’t handle VECTORS“.
I'd like to do something like :

-- psl DataChange: assert always ctrl_reg /= prev ctrl_reg @ rising_edge(clk) report "Data change : " & integer'image(unsigned(ctrl_reg));

Where in ctrl_reg is a 16-bit vector.

I have no write or read signal at that hierarchical level to trigger on.

tmeissner, 2019-04-19:
> -- psl DataChange: assert always ctrl_reg /= prev ctrl_reg @ rising_edge(clk) report "Data change : " & integer'image(unsigned(ctrl_reg));
>
> Where in ctrl_reg is a 16-bit vector.
> And yes.... I Know that "prev" is not a PSL command.

`prev(e, [c])` is a PSL function which works as you think, it returns the value of expression e in the previous clock cycle or c clock cycles before. So it could work, but I think that `prev()`isn't implemented by GHDL, as most of the PSL functions. The `stable()`PSL function could also be of use, but it isn't implemented either.

A possible workaround is to write some glue logic in VHDL which register the `ctrl_reg` signal and check in PSL aas above. I do something like this in a unit:

```vhdl
    -- synthesis off
    verification : block is

      signal s_data : std_logic_vector(0 to 127);

    begin

      s_data <= data_o  when rising_edge(clk_i) else
                128x"0" when reset_i = '0';

      -- psl cover accept_o;
      -- psl assert always (accept_o -> s_round = 0);
    
      -- psl cover valid_i and accept_o;
      -- psl assert always (valid_i and accept_o -> next not accept_o);

      -- psl cover valid_o;
      -- psl assert always (valid_o -> s_round = t_enc_rounds'high);

      -- psl cover valid_o and accept_i;
      -- psl assert always (valid_o and accept_i -> next not valid_o);

      -- psl cover valid_o and not accept_i;
      -- psl assert always (valid_o and not accept_i -> next valid_o);
      -- psl assert always (valid_o and not accept_i -> next data_o = s_data);

    end block verification;
    -- synthesis on
```

stdefeber, 2019-04-29:
Thanks for your suggestion.
I will try it !

closed 2019-05-03 by tgingold