ghdl/ghdl #802: Add VHDL source for the std.standard package

maehne, 2019-04-24:
**Is your feature request related to a problem? Please describe.**

I use `vhdl_ls` language server, which is developed as part of the [kraigher/rust_hdl](https://github.com/kraigher/rust_hdl/) project for online syntax checking of my VHDL sources while editing them. For full syntax checking, `vhdl_ls` needs to know the definitions also of all standard packages as well as those from the other VHDL files, which are part of the project. To this end, a project-specific [`vhdl_ls.toml`](https://github.com/kraigher/rust_hdl/) needs to created, which lists the VHDL files of all used libraries. `ghdl` conveniently installs the sources of the VHDL libraries (`std`, `ieee`, etc.), which it distributes, below `$PREFIX/lib/ghdl/src`. Only one file `standard.vhdl` is missing with the declarations from the most fundamental `std.standard` package (`integer`, `unsigned`, `positive`, `real`, `time`, `delay_length`, etc.).

**Describe the solution you'd like**

Add the VHDL source file `standard.vhdl` containing the definitions for the `std.standard` package to the `libraries/std/` folder and install it along with the other sources that constitute the `std` package. 

**Describe alternatives you've considered**

My current solution is to reference the [`standard.vhdl`](https://github.com/kraigher/rust_hdl/blob/master/example_project/vhdl_libraries/2008/std/standard.vhd) file, which is part of the example project contained in the `rust_hdl` sources. However, the location of the checked out `rust_hdl` sources may be different from system to system, whereas the default installation location of `ghdl` is the same for different OS platforms.

It would be also not convenient to store that `standard.vhdl` file as part of each VHDL project a user might be working on.

**Additional context**

None.

tgingold, 2019-04-24:
I see your point but there is a technical problem: it is not really possible to completely write std.standard in plain and correct vhdl.  There is a bootstrap issue.

You can generate the standard package using `ghdl --disp-standard` command.  It could even be possible to write `standard.vhdl` during installation (but not to use it).

OTOH, if rust_hdl implements full syntax checking, it will provide its own definition of standard and that will be integrated.  So it looks like more a rust_hdl issue than a ghdl issue.

A possible solution would be to use ghdl as a language server.  I think we aren't very far from it...

maehne, 2019-04-30:
Thanks @tgingold for your quick reply to my feature request! I understand very well the bootstrap issue that you mention. Nevertheless, even the subset, which is expressible using correct VHDL is suitable for the use case I am targeting for, i.e., making the fundamental VHDL types known to the VHDL language server `vhdl_ls` of `rust_hdl`.

Thank you for pointing me to the `--disp-standard` option! I was not aware of its existence. Its output is suitable for consumption by `vhdl_ls` even though `vhdl_ls` currently chokes on four type definitions output by `--disp-standard` (see [`rust_hdl` issue #36](https://github.com/kraigher/rust_hdl/issues/36); diagnosing these is the reason why my reply to this issue got delayed), which I hope will get resolved soon. Therefore, it would be great if in the `make install` phase, the VHDL source of `std.standard` could be written to the `$PREFIX/lib/ghdl/src/std/` directory for all three supported VHDL standards:
```sh
$ ghdl --disp-standard --std=87 > $PREFIX/lib/ghdl/src/std/standard.v87
$ ghdl --disp-standard --std=93 > $PREFIX/lib/ghdl/src/std/standard.v93
$ ghdl --disp-standard --std=08 > $PREFIX/lib/ghdl/src/std/standard.v08
```

If `ghdl` would implement the language server protocol in the future that would be even better, but for the meantime, I have to admit that already `rust_hdl` is a helpful tool avoiding me unnecessary compile + fix typos cycles.

maehne, 2019-05-01:
Fixed by merged PR #811.

closed 2019-05-01 by maehne

Ahmad-Zaklouta, 2020-02-26:
Hi! @maehne 
Sorry if this is not related to the issue. 
I have been trying to configure rust_hdl to work with neovim to avoid compilation for small error but I didn't manage. I wonder if you could share your experience on how to setup rust_hdl and configure it to make it work.

eine, 2020-02-26:
@Ahmad-Zaklouta, I think that

maehne, 2020-03-01:
As @eine said: This is not the right place to discuss your configuration issue. Anyway, I cannot really help you as I don't use (neo)vim for my editing purposes but rather Emacs.