ghdl/ghdl #803: Dynamically loading design binaries : shared libraries

umarcor, 2019-04-25:
Ref #640, #670 and #800.

This is neither a feature request nor a bug report. In case this feature is stabilized, it might be interesting to add it to the docs in the future.

As commented in [#800](https://github.com/ghdl/ghdl/issues/800#issuecomment-486468731), it is currently possible to dynamically load the executable binaries generated by GHDL under certain conditions. The following script allows to test it:

```
LLVM_VER=6.0
IMG=ubuntu:bionic

cat > test_run.sh <<-EOF
apt-get update -qq

apt-get -y install --no-install-recommends \
  ca-certificates \
  clang-$LLVM_VER \
  curl \
  gcc \
  gnat \
  llvm-$LLVM_VER-dev \
  make \
  python3 \
  python3-pip \
  zlib1g-dev

apt-get autoclean && apt-get clean && apt-get -y autoremove
update-ca-certificates

mkdir -p /tmp/ghdl && cd /tmp/ghdl
curl -fsSL https://codeload.github.com/ghdl/ghdl/tar.gz/master | tar xzf - --strip-components=1
OPT_FLAGS=-fPIC LIB_CFLAGS=-fPIC ./dist/travis/build.sh -b llvm-$LLVM_VER -p ghdl-llvm-fPIC
tar -xzf ghdl-llvm-fPIC.tgz -C /usr/local

mkdir -p /tmp/vunit && cd /tmp/vunit
curl -fsSL https://codeload.github.com/dbhi/vunit/tar.gz/dyn-ghdl | tar xzf - --strip-components=1
export PYTHONPATH=\$(pwd)
pip3 install -r requirements.txt

cd examples/vhdl/external_buffer
python3 run.py -v
EOF

$(command -v winpty) docker run --rm -it "$IMG" bash -c "$(cat test_run.sh)"
```

|  | VUNIT EXECUTION | ELABORATE ONLY | REGULAR EXECUTION | PYTHON ALLOCATION |
|---|---|---|---|---|
| `ubuntu:bionic` `VHDL 2008` | ok | ok | ok | ok |
| `ubuntu:bionic` `VHDL 1993` | ok | ok | abort | - |
| `ubuntu:xenial` `VHDL 2008` | ok | ok | fail | - |
| `ubuntu:xenial` `VHDL 1993` | ok | ok | fail | - |
| `ubuntu:trusty` `VHDL 2008` | ok | ok | fail | - |
| `ubuntu:trusty` `VHDL 1993` | ok | ok | fail | - |

- `abort` means that the `main` function is successfully executed, the simulation is successfully executed too, but it exits with `abort()`. This forces the C wrapper and Python to exit inmediately, without running any post-check.
- `fail` means that `dlopen` fails at dynamically loading the binary as if it was a shared library.

> NOTE: on `trusty` LLVM_VER=3.9 is used.

These are the ELF headers of each of the binaries:

`ubuntu:bionic` `VHDL 2008`:

```
ELF Header:
  Magic:   7f 45 4c 46 02 01 01 00 00 00 00 00 00 00 00 00
  Class:                             ELF64
  Data:                              2's complement, little endian
  Version:                           1 (current)
  OS/ABI:                            UNIX - System V
  ABI Version:                       0
  Type:                              DYN (Shared object file)
  Machine:                           Advanced Micro Devices X86-64
  Version:                           0x1
  Entry point address:               0x19fa0
  Start of program headers:          64 (bytes into file)
  Start of section headers:          2087792 (bytes into file)
  Flags:                             0x0
  Size of this header:               64 (bytes)
  Size of program headers:           56 (bytes)
  Number of program headers:         9
  Size of section headers:           64 (bytes)
  Number of section headers:         36
  Section header string table index: 35
```

`ubuntu:bionic` `VHDL 1993`:

```
ELF Header:
  Magic:   7f 45 4c 46 02 01 01 00 00 00 00 00 00 00 00 00
  Class:                             ELF64
  Data:                              2's complement, little endian
  Version:                           1 (current)
  OS/ABI:                            UNIX - System V
  ABI Version:                       0
  Type:                              DYN (Shared object file)
  Machine:                           Advanced Micro Devices X86-64
  Version:                           0x1
  Entry point address:               0x19210
  Start of program headers:          64 (bytes into file)
  Start of section headers:          1908840 (bytes into file)
  Flags:                             0x0
  Size of this header:               64 (bytes)
  Size of program headers:           56 (bytes)
  Number of program headers:         9
  Size of section headers:           64 (bytes)
  Number of section headers:         36
  Section header string table index: 35
```

`ubuntu:xenial` `VHDL 2008`:

```
ELF Header:
  Magic:   7f 45 4c 46 02 01 01 00 00 00 00 00 00 00 00 00
  Class:                             ELF64
  Data:                              2's complement, little endian
  Version:                           1 (current)
  OS/ABI:                            UNIX - System V
  ABI Version:                       0
  Type:                              EXEC (Executable file)
  Machine:                           Advanced Micro Devices X86-64
  Version:                           0x1
  Entry point address:               0x4021d0
  Start of program headers:          64 (bytes into file)
  Start of section headers:          2004768 (bytes into file)
  Flags:                             0x0
  Size of this header:               64 (bytes)
  Size of program headers:           56 (bytes)
  Number of program headers:         9
  Size of section headers:           64 (bytes)
  Number of section headers:         38
  Section header string table index: 35
```

`ubuntu:xenial` `VHDL 1993`:

```
ELF Header:
  Magic:   7f 45 4c 46 02 01 01 00 00 00 00 00 00 00 00 00
  Class:                             ELF64
  Data:                              2's complement, little endian
  Version:                           1 (current)
  OS/ABI:                            UNIX - System V
  ABI Version:                       0
  Type:                              EXEC (Executable file)
  Machine:                           Advanced Micro Devices X86-64
  Version:                           0x1
  Entry point address:               0x4021d0
  Start of program headers:          64 (bytes into file)
  Start of section headers:          1829912 (bytes into file)
  Flags:                             0x0
  Size of this header:               64 (bytes)
  Size of program headers:           56 (bytes)
  Number of program headers:         9
  Size of section headers:           64 (bytes)
  Number of section headers:         38
  Section header string table index: 35
```

`ubuntu:trusty` `VHDL 2008`:

```
ELF Header:
  Magic:   7f 45 4c 46 02 01 01 00 00 00 00 00 00 00 00 00
  Class:                             ELF64
  Data:                              2's complement, little endian
  Version:                           1 (current)
  OS/ABI:                            UNIX - System V
  ABI Version:                       0
  Type:                              EXEC (Executable file)
  Machine:                           Advanced Micro Devices X86-64
  Version:                           0x1
  Entry point address:               0x402250
  Start of program headers:          64 (bytes into file)
  Start of section headers:          1435784 (bytes into file)
  Flags:                             0x0
  Size of this header:               64 (bytes)
  Size of program headers:           56 (bytes)
  Number of program headers:         9
  Size of section headers:           64 (bytes)
  Number of section headers:         38
  Section header string table index: 35
```

`ubuntu:trusty` `VHDL 1993`:

```
ELF Header:
  Magic:   7f 45 4c 46 02 01 01 00 00 00 00 00 00 00 00 00
  Class:                             ELF64
  Data:                              2's complement, little endian
  Version:                           1 (current)
  OS/ABI:                            UNIX - System V
  ABI Version:                       0
  Type:                              EXEC (Executable file)
  Machine:                           Advanced Micro Devices X86-64
  Version:                           0x1
  Entry point address:               0x402250
  Start of program headers:          64 (bytes into file)
  Start of section headers:          1292424 (bytes into file)
  Flags:                             0x0
  Size of this header:               64 (bytes)
  Size of program headers:           56 (bytes)
  Number of program headers:         9
  Size of section headers:           64 (bytes)
  Number of section headers:         38
  Section header string table index: 35
```

umarcor, 2019-04-25:
This has nothing to do with the version of GLIBC, gcc, or any other. The different behaviour is because on some systems, such as Ubuntu 18, GCC is built with `--enable-default-pie`. As a result, all the objects are compiled as if `-fPIC -pie -fPIE` was used: https://stackoverflow.com/questions/41303584/output-of-make-is-a-shared-object-and-not-an-executable

Unfortunately, prividing the flags when GHDL is compiled as follows is not enough:

```
OPT_FLAGS="-fPIC -Wl,-pie -Wl,-fPIE" \
LIB_CFLAGS="-fPIC -Wl,-pie -Wl,-fPIE" \
./dist/travis/build.sh -b llvm -p ghdl-llvm-fPIC
```

Fortunately, it is possible to achieve it by adding flags to `GRT_FLAGS` in `src/grt/Makefile.in`. I will try to isolate the minimum changes and propose a PR.

umarcor, 2019-04-25:
With the fix proposed in #804, `OPT_FLAGS="-fPIC" LIB_CFLAGS="-fPIC" ./dist/travis/build.sh -b llvm -p ghdl-llvm-fPIC` suffices. Then, `-fPIC` must be used for analysis, and `-fPIC -Wl,-pie` for elaboration. For example, in a VUnit run script:

``` python
ui.set_compile_option("ghdl.flags", ['-fPIC'])
ui.set_sim_option("ghdl.elab_flags", ['-fPIC', '-Wl,-pie', '-Wl,-Wl,--version-script='+join(src_path, 'grt.ver')])
```

umarcor, 2019-04-26:
With #805, the syntax is simplified:

``` bash
CONFIG_OPTS="--enable-default-pie " ./dist/travis/build.sh -b llvm -p ghdl-llvm-fPIC
```

``` python
ui.set_sim_option("ghdl.elab_flags", [
    '-Wl,' + c_obj,
    '-Wl,-pie',
    '-Wl,-Wl,--version-script=' + join(src_path, 'grt.ver')
])
```

umarcor, 2021-01-24:
**Description**

This is a MWE to illustrate the issue in https://github.com/ghdl/ghdl-cosim/pull/15:

**How to reproduce?**

```vhdl
entity tb is
end entity;

architecture arch of tb is
begin
  process begin
    report "Hello!" severity failure;
    wait;
  end process;
end;

--:file: tb.vhd
```

```c
#include <dlfcn.h>
#include <stdio.h>
#include <stdlib.h>

int main(int argc, void** argv) {

  void* h = dlopen("./tb.so", RTLD_LAZY);
  if (!h){
    fprintf(stderr, "%s\n", dlerror());
    exit(1);
  }

  typedef int main_t(int, void**);

  h = dlopen("./tb.so", RTLD_LAZY);
  if (!h){
    fprintf(stderr, "%s\n", dlerror());
    exit(1);
  }

  main_t* ghdl_main = (main_t*)dlsym(h, "ghdl_main");
  if (!ghdl_main){
    fprintf(stderr, "%s\n", dlerror());
    exit(2);
  }

  printf("ghdl_main return: %d\n", ghdl_main(argc, argv));

  dlclose(h);

  return 0;

}

//:file: main.c
```

```
#!/usr/bin/env sh

gcc main.c -o main -ldl
ghdl -a tb.vhd
ghdl -e -Wl,-fPIC -Wl,-shared -Wl,-Wl,-u,ghdl_main -o tb.so tb
./main

#:file: run.sh
```

The output is the following:

```sh
# ./test.sh 
tb.vhd:7:5:@0ms:(report failure): Hello!
D:\tmp-abort\main.exe:error: report failed
in process .tb(arch).P0
D:\tmp-abort\main.exe:error: simulation failed
```

If the report in the VHDL source is changed to `note`, the output is the following:

```sh
# ./test.sh 
tb.vhd:7:5:@0ms:(report error): Hello!
ghdl_main return: 0
```

**Expected behaviour**

I would expect the output when `failure` to be:

```sh
# ./test.sh 
tb.vhd:7:5:@0ms:(report failure): Hello!
D:\tmp-abort\main.exe:error: report failed
in process .tb(arch).P0
D:\tmp-abort\main.exe:error: simulation failed
ghdl_main return: 1
```

Or any other return code other than 0. However, the problem is that the wrapper (the C program) exits immediately, so the `ghdl_main return: 1` line (or any code after ghdl_main) is not executed.

**Context**

MINGW64:

```
# ghdl --version
GHDL 1.0-dev (0.37.0.r1342.geeb25ef2) [Dunoon edition]
 Compiled with GNAT Version: 10.2.0
 llvm code generator
Written by Tristan Gingold.

Copyright (C) 2003 - 2021 Tristan Gingold.
GHDL is free software, covered by the GNU General Public License.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
```

**Additional context**

Other tests (https://github.com/ghdl/ghdl-cosim/runs/1738969727?check_suite_focus=true#step:3:1558) show that an abortion signal is generated:

```
   /usr/local/lib/python3.7/dist-packages/vunit/vhdl/core/src/stop_body_93-2002.vhd:10:5:@0ms:(report failure): Stopping simulation with status 0
  /src/vhpidirect/quickstart/wrapping/exitcb/py/vunit_out/test_output/lib.tb_abrt.all_775e0a86ed093bdf996e118af0a0791ccd72f0fd/ghdl/tb_abrt-tb:error: report failed
  in process .tb_abrt(tb).main
  SIGABRT caught 6!
  Aborted (core dumped)
```

tgingold, 2021-01-25:
There is something I don't understand.  You are using mingw64.

But this is not a Windows feature, it's a unix one:

> #include <dlfcn.h>

It is not even available on my mingw64 system.

umarcor, 2021-01-25:
😆 Did I tell that you are sometimes so annoyingly good? You are obviously correct. In fact, you can test this on Linux as-is.

Yet, `dlfcn.h` is available on MSYS2:

```sh
# pacman -F dlfcn.h
mingw32/mingw-w64-i686-dlfcn 1.2.0-1
    mingw32/include/dlfcn.h
mingw32/mingw-w64-i686-postgresql 12.4-1
    mingw32/include/postgresql/server/port/win32/dlfcn.h
mingw32/mingw-w64-i686-python-autopxd2 1.1.0-1
    mingw32/lib/python3.8/site-packages/autopxd/include/dlfcn.h
mingw64/mingw-w64-x86_64-dlfcn 1.2.0-1 [installed: 1.2.0-2]
    mingw64/include/dlfcn.h
mingw64/mingw-w64-x86_64-postgresql 12.4-1
    mingw64/include/postgresql/server/port/win32/dlfcn.h
mingw64/mingw-w64-x86_64-python-autopxd2 1.1.0-1
    mingw64/lib/python3.8/site-packages/autopxd/include/dlfcn.h
msys/msys2-runtime-devel 3.1.7-4 (msys2-devel) [installed]
    usr/include/dlfcn.h
```

Precisely, it is included in [mingw-w64-x86_64-dlfcn](https://packages.msys2.org/package/mingw-w64-x86_64-dlfcn). You can install it with `pacman -S mingw-w64-x86_64-dlfcn`. My guess is that it's a wrapper around Windows' native DLL open/close features.

FTR, this is what I use in Python for achieving the same result: https://github.com/ghdl/ghdl-cosim/blob/master/vhpidirect/shared/pycb/utils.py#L10-L34

Anyway, the code above is extracted from https://github.com/ghdl/ghdl-cosim/actions/runs/500013976. There you can see it tested on Windows and Linux, using different versions of GHDL installed through different procedures; examples include C only examples and also Python based ones. This error/bug seems to be consistent on all runs, regardless of the platform and the wrapper language. The difference is that C allows capturing Abortion signals, but Python cannot.

tgingold, 2021-01-25:
Ok, I think I now understand the issue.

umarcor, 2021-01-25:
My intuitive guess is that GHDL binaries can exit abruptely on `failure`. Note that the "crash" is not produced with reports of level `note` or `error`. However, exiting abruptely from a shared library is undesired. Therefore, this is not a bug per se. But a missing enhancement resulting from the addition of `-shared` to GHDL's elaboration options some months ago.

Yet, this issue is previous to opt `-shared` being added to GHDL. Before that, I used GCC for elaboration (through `--bind` and `--list-link`), which produced the same failure/abortion. It was not considered a bug back then because loading GHDL models dynamically was explicitly hackish.

As a result, my guess is that GHDL should handle `failure` reports similarly to `error`; thus existing but not aborting. Maybe the return value can be different for users to tell them apart.

These are the places I tried to look at:

- https://github.com/ghdl/ghdl/blob/4868294436574660552ccef50a5b0849559393de/src/grt/config/win32.c#L238
- https://github.com/ghdl/ghdl/blob/4868294436574660552ccef50a5b0849559393de/src/grt/grt-cgnatrts.c#L23

commit 0978b480 (2021-01-26): grt: add Error_NF to report final error status.  Fix #803
closed 2021-01-26 by tgingold