ghdl/ghdl #807: Crash when assigning to a procedure output that is an unconstrained record

bradleyharden, 2019-04-28:
I encountered this bug when writing an improved version of `stream_master_pkg` and `stream_slave_pkg` for VUnit. Here is an MCVE:

```
$ cat test.vhd

library ieee;
use ieee.std_logic_1164.all;

package test_pkg is

  type record_t is record
    data : std_ulogic_vector;
  end record;

  procedure test_procedure(
    variable rec : out record_t);

  function test_function
    return record_t;

  function get_data
    return std_ulogic_vector;

end package;

package body test_pkg is

  procedure test_procedure(
    variable rec : out record_t)
  is
  begin
    rec := test_function;
  end procedure;

  function test_function
    return record_t
  is
  begin
    return (data => get_data);
  end function;

  function get_data
    return std_ulogic_vector
  is
  begin
    return x"0F";
  end function;

end package body;

$ ghdl -a --std=08 test.vhd
```

The record type is intended to define the elements expected in a message. The `test_function` is supposed to represent a function that pops a record from a message. The `get_data` function mimics popping a single `std_ulogic_vector` from a message. **The `test_procedure` call to `test_function` is the source of the crash.** Commenting that line prevents the crash.

I have successfully compiled my new version of ``stream_pkg`` in Modelsim, so I'm pretty certain this is not a misuse of the language.

```
******************** GHDL Bug occurred ***************************
Please report this bug on https://github.com/ghdl/ghdl/issues
GHDL release: 0.37-dev (20181129-243-g8a88fdfc) [Dunoon edition]
Compiled with GNAT Version: 7.3.0
Target: x86_64-linux-gnu
In directory: /home/bharden/tmp/
Command line:
/usr/local/bin/ghdl1-llvm --std=08 -P/usr/local/lib/ghdl/ieee/v08/ -P/usr/local/lib/ghdl/ -c -fpic -o test.o test.vhd
Exception TYPES.INTERNAL_ERROR raised
Exception information:
raised TYPES.INTERNAL_ERROR : trans-chap7.adb:2794
Call stack traceback locations:
0x5d1a33 0x5d1b09 0x581128 0x59701c 0x597169 0x61f04f 0x5be0b6 0x62025e 0x56cff3 0x62f40b 0x630ace 0x6323e1 0x40863d 0x7f0ef4675b95 0x4079e8 0xfffffffffffffffe
******************************************************************
```

- OS: Ubuntu 18.04
- Origin:
  - [ ] Package manager. Repo:
  - [ ] Released binaries. Tarball:
  - [x] Built from sources. Commit SHA: 8a88fdf

**Checklist**
Before submitting your issue, please review the following checklist:

- [x] Add `GHDL Bug occurred` log block
- [x] Add a MWE
- [x] Try the latest version

1138-4EB, 2019-04-28:
Hi @bradleyharden!

This might be specific to GCC/LLVM backends. The code above seems to be successfully analyzed with `ghdl/ext:vunit` (mcode):

```
$(command -v winpty) docker run --rm -itv /$(pwd)://src -w //src ghdl/ext:vunit bash -c "ghdl -a --std=08 test.vhd"
```

But it fails with GCC too:

```
$(command -v winpty) docker run --rm -itv /$(pwd)://src -w //src ghdl/ghdl:fedora29-gcc-8.2.0 bash -c "ghdl -a --std=08 test.vhd"

******************** GHDL Bug occurred ***************************
Please report this bug on https://github.com/ghdl/ghdl/issues
GHDL release: 0.37-dev (v0.36-56-g27379809) [Dunoon edition]
Compiled with GNAT Version: 8.3.1 20190223 (Red Hat 8.3.1-2)
Target: x86_64-redhat-linux
In directory: /src/
Command line:
/usr/local/libexec/gcc/x86_64-pc-linux-gnu/8.2.0/ghdl1 --ghdl--std=08 -P/usr/local/lib/ghdl/ieee/v08/ -P/usr/local/lib/ghdl/ -quiet -o
 test.s test.vhd
Exception TYPES.INTERNAL_ERROR raised
Exception information:
raised TYPES.INTERNAL_ERROR : trans-chap7.adb:2794
Call stack traceback locations:
0x74a9bf 0x751d16 0x724510 0x72d50d 0x72edcd 0x71b856 0x742ab7 0x71ce31 0x70b047 0x6f3132 0x6f128b 0x630c48 0xb1aa8d 0x629b11 0x635889
 0x759a49 0x627ffa 0x7f9bae94d411 0x62be6c 0xfffffffffffffffe
******************************************************************

Execution of /usr/local/libexec/gcc/x86_64-pc-linux-gnu/8.2.0/ghdl1 terminated by unhandled exception
raised TYPES.INTERNAL_ERROR : trans-chap7.adb:2794
Call stack traceback locations:
0x74a9bf 0x751d16 0x724510 0x72d50d 0x72edcd 0x71b856 0x742ab7 0x71ce31 0x70b047 0x6f3132 0x6f128b 0x630c48 0xb1aa8d 0x629b11 0x635889
 0x759a49 0x627ffa 0x7f9bae94d411 0x62be6c 0xfffffffffffffffe
ghdl:error: compilation error
```

bradleyharden, 2019-04-28:
Using the mcode backend gets past compilation, but then it fails on elaboration.

```vhdl
library ieee;
use ieee.std_logic_1164.all;

use work.test_pkg.all;

entity test is
end entity;

architecture a of test is
begin

  process
    variable rec : record_t(data(7 downto 0));
  begin
    test_procedure(rec);
    report to_string(rec.data);
    wait;
  end process;

end architecture;
```

Using this code with

```bash
$ ghdl -e --std=08 test
```
gives
```
******************** GHDL Bug occurred ***************************
Please report this bug on https://github.com/ghdl/ghdl/issues
GHDL release: 0.37-dev (20181129-243-g8a88fdfc) [Dunoon edition]
Compiled with GNAT Version: 7.3.0
Target: x86_64-linux-gnu
In directory: /home/bharden/tmp/
Command line:
ghdl -e --std=08 test
Exception TYPES.INTERNAL_ERROR raised
Exception information:
raised TYPES.INTERNAL_ERROR : trans-chap7.adb:2794
Call stack traceback locations:
0x5561d1b18305 0x5561d1b183b2 0x5561d1adb156 0x5561d1aeacaa 0x5561d1aeadf7 0x5561d1b55f88 0x5561d1b09034 0x5561d1b570f8 0x5561d1ac9c77 0x5561d1ac7413 0x5561d1ad5086 0x5561d1b62a9c 0x5561d1aacc97 0x5561d1a3492c 0x5561d1b661fe 0x5561d186a765 0x7ff053c85b95 0x5561d1869528 0xfffffffffffffffe
******************************************************************
```

tgingold, 2019-04-29:
The assignment is not yet implemented, but this is doable...

commit 722ea848 (2019-05-03): vhdl/translate: check_composite_match: rename and handle records.

Fix #807
closed 2019-05-03 by tgingold