ghdl/ghdl #809: How to set "NO_WARNING" to true in package numeric_std when running GHDL simulation?

wm2015email, 2019-04-30:
How to set "NO_WARNING" to true in package **numeric_std** when running GHDL simulation???

The GHDL documentation is completely mute on how to do this:

FROM IEEE VHDL Documentation: "The warnings can be suppressed. IEEE Std 1076-2008 16.8.5.2 Allowable modifications para 2 (in part) The package bodies for the NUMERIC_BIT and NUMERIC_STD packages declare a constant named NO_WARNING that has the value FALSE. A user may set NO_WARNING to TRUE and reanalyze the package body to suppress warning messages generated by calls to the functions in these packages. or in modelsim.ini uncomment or set NumericStdNoWarnings = 1. They are caused by metavalues present in a signed or unsigned with their binary value being evaluated, typically at initialization. Look to your component."

Suggest:  --no_warn_numeric

tgingold, 2019-04-30:
It isn't fully mute.  See option --ieee-asserts:

.. option:: --ieee-asserts<=POLICY>

  Select how the assertions from ``ieee`` units are
  handled. `POLICY` can be ``enable`` (the default),
  ``disable`` which disables all assertions from ``ieee`` packages
  and ``disable-at-0`` which disables only at the start of simulation.

  This option can be useful to avoid assertion messages from
  ``ieee.numeric_std`` (and other ``ieee`` packages).

wm2015email, 2019-04-30:
C:\Users\user\desktop\vhdl>..\simtools\ghdl\bin\ghdl -r bus_fifo --ieee-assert=disable-at-0

ghdl.exe: error: unknown option '--ieee-assert=disable-at-0', try --help

tgingold, 2019-04-30:
According to the documentation, it's --ieee-asserts=

closed 2019-05-02 by tgingold