ghdl/ghdl #810: CONSTRAINT_ERROR exception for an invalid range in a physical type definition

maehne, 2019-05-01:
**Description**
A CONSTRAINT_ERROR exception is generated for the following invalid range in a physical type definition :
```vhdl
package my_time_pkg is

  type my_time is range -integer'low  to integer'high units
    fs;
    ps = 1000 fs;
    ns = 1000 ps;
    us = 1000 ns;
    ms = 1000 us;
    sec = 1000 ms;
    min = 60 sec;
    hr = 60 min;
  end units;

end package my_time_pkg;
```

The range should be rather `integer'low  to integer'high`.

**Expected behaviour**

`ghdl` should report an error about the invalid range.

**Context**

```
$ ghdl -a my_time_pkg.vhdl

******************** GHDL Bug occurred ***************************
Please report this bug on https://github.com/ghdl/ghdl/issues
GHDL release: 0.37-dev (v0.36-58-g0a0fc18d) [Dunoon edition]
Compiled with GNAT Version: 8.3.0
Target: x86_64-linux-gnu
In directory: /home/mht1/Test/GHDL/
Command line:
/usr/local/bin/ghdl1-llvm -P/usr/local/lib/ghdl/ieee/v93/ -P/usr/local/lib/ghdl/ -c -fpic -o my_time_pkg.o my_time_pkg.vhdl
Exception CONSTRAINT_ERROR raised
Exception information:
raised CONSTRAINT_ERROR : trans-chap3.adb:2060 range check failed
Call stack traceback locations:
0x61293a 0x61342f 0x614169 0x5b8b6d 0x5b9e72 0x5bc5b6 0x61e446 0x61e9ef 0x56c4f6 0x62dbdd 0x62f264 0x630b2d 0x409f2f 0x7fe1783a8099 0x409188 0xfffffffffffffffe
******************************************************************
ghdl:error: compilation error
```

- OS: Linux, Debian Buster AMD64
- Origin:
  - [ ] Package manager. Repo:
  - [ ] Released binaries. Tarball:
  - [X] Built from sources. Commit SHA: 0a0fc18db7e506586e1b64fbeb9bff294425f219

**Additional context**

None

**How to reproduce?**

```
#>> my_time_pkg.vhd
package my_time_pkg is

  type my_time is range -integer'low  to integer'high units
    fs;
    ps = 1000 fs;
    ns = 1000 ps;
    us = 1000 ns;
    ms = 1000 us;
    sec = 1000 ms;
    min = 60 sec;
    hr = 60 min;
  end units;

end package my_time_pkg;

#>> sim.sh
ghdl -a my_time_pkg.vhd

#>> run.sh
docker run --rm -tv /$(pwd):/src:z -w //src ghdl/ghdl:buster-mcode sh -c ./sim.sh

#>> end
```

Note that `run.sh` is used to execute `sim.sh` inside a docker container. Please, put your commands in `sim.sh` and just copy `run.sh` from the example. Using `ghdl/ghdl:*` docker images to run the MWEs ensures that the latest available GHDL is used.

**Files**

None

**Checklist**
Before submitting your issue, please review the following checklist:

- [X] Add `GHDL Bug occurred` log block
- [X] Add a MWE
- [X] Try the latest version

tgingold, 2019-05-03:
That's a corner case, but easy to fix.

closed 2019-05-04 by tgingold

maehne, 2019-05-06:
Thanks for fixing this corner case! However, now the above test case compiles without any warnings, which I would expect in the case of a null range. At least ModelSim does so for a smaller range of 8 to 7 added to my above test case, but not for the range defined for `my_time`:

This would be the new extended test case:
```vhdl
package my_time_pkg is

  type my_time is range -integer'low  to integer'high units
    fs; ps= 1000 fs; ns= 1000 ps; us= 1000 ns; -- very short
    ms= 1000 us; sec= 1000 ms; min= 60 sec; hr= 60 min; -- longer
  end units;
  type my_empty_range is range -(-8) to 7;
  type my_small_range is range -8 to 7;

end package my_time_pkg;
```
This is the output generated by ModelSim:
```
$ intelFPGA vcom my_time_pkg.vhdl
Start time: 12:26:11 on May 06,2019
vcom my_time_pkg.vhdl 
Model Technology ModelSim - Intel FPGA Edition vcom 10.5b Compiler 2016.10 Oct  5 2016
-- Loading package STANDARD
-- Compiling package my_time_pkg
** Warning: my_time_pkg.vhdl(7): (vcom-1246) Range 8 to 7 is null.
End time: 12:26:11 on May 06,2019, Elapsed time: 0:00:00
Errors: 0, Warnings: 1
```
I don't understand why neither GHDL nor ModelSim warn about the `my_time` time definition: After all, `-integer'low` should be bigger than `integer'high` (when assuming two's complement representation).

tgingold, 2019-05-07:
You mean something like:

nullrng.vhdl:3:25:warning: physical type "my_time" has a null range [-Wruntime-error]
nullrng.vhdl:7:32:warning: integer type "my_empty_range" has a null range [-Wruntime-error]

maehne, 2019-05-10:
Yes, exactly! Thanks @tgingold for adding this helpful warning and being so reactive to all these issue reports!

commit 981ccfb0 (2019-05-04): vhdl: do not crash on 64 bit null range.

Fix #810