ghdl/ghdl #816: raised SYSTEM.ASSERTIONS.ASSERT_FAILURE : canon.adb:574

umarcor, 2019-05-07:
Uncommenting any of the four commented lines in `test.vhd` produces a crash.

``` bash
******************** GHDL Bug occurred ***************************
Please report this bug on https://github.com/ghdl/ghdl/issues
GHDL release: 0.37-dev (tarball) [Dunoon edition]
Compiled with GNAT Version: 7.3.0
Target: x86_64-linux-gnu
In directory: /src/
Command line:
/usr/local/bin/ghdl1-llvm --workdir=/src/vunit_out/ghdl/libraries/lib --work=lib
 --std=08 -P/src/vunit_out/ghdl/libraries/vunit_lib/ -P/src/vunit_out/ghdl/libra
ries/lib/ -P/usr/local/lib/ghdl/ieee/v08/ -P/usr/local/lib/ghdl/ -c -fpic -o /sr
c/vunit_out/ghdl/libraries/lib/test.o /src/test.vhd
Exception SYSTEM.ASSERTIONS.ASSERT_FAILURE raised
Exception information:
raised SYSTEM.ASSERTIONS.ASSERT_FAILURE : canon.adb:574
Call stack traceback locations:
0x7f30a99a8161 0x564988 0x564a35 0x5a80f5 0x5ae2e5 0x62f68d 0x572157 0x634ae5 0x
63537f 0x636c64 0x4087b8 0x7f30a8a53b95 0x407848 0xfffffffffffffffe
******************************************************************
```

```
#>> test.vhd
library ieee;
context ieee.ieee_std_context;

library vunit_lib;
context vunit_lib.vunit_context;

entity tb_test is
  generic ( runner_cfg : string );
end entity;

architecture tb of tb_test is

  constant params: integer_vector_ptr_t := new_integer_vector_ptr(2, -1);

  constant clk_period    : time    := 20 ns;

  type time_t is array (natural range 0 to 1) of natural;

  procedure set_time(t: time_t) is begin
    set(params, 0, t(0));
    set(params, 1, t(1));
  end;

  procedure get_time(variable t: inout time_t) is begin
    t(0) := get(params, 0);
    t(1) := get(params, 1);
  end;

  impure function get_time return time_t is
    variable t: time_t;
  begin
--    t(0) := get(params, 0);
--    t(1) := get(params, 1);
    return t;
  end;

  constant null_time: time_t := (others => 0);

  impure function to_string(t: time_t) return string is begin
    return to_string(t(0)) & " " & to_string(t(1));
  end;

  signal clk: std_logic;

begin

  clk <= not clk after (clk_period/2);

  run: process(all)
    variable r: time_t;
  begin
    if rising_edge(clk) then
--      info("Time: " & to_string(get_time));
--      get_time(r);
    end if;
  end process;

  main: process
    variable t: time_t;
  begin
    test_runner_setup(runner, runner_cfg);
      get_time(t);
      if t = null_time then
        set_time((others=>natural'high));
      end if;
      get_time(t);
      info("Init test | time: " & to_string(t));
      info("Test done");
    test_runner_cleanup(runner);
    wait;
  end process;
  test_runner_watchdog(runner, 500 ms);

end architecture;

#>> main.c
#include <stdio.h>
#include <stdlib.h>
#include <stdint.h>
#include "vhpidirect_user.h"

extern int ghdl_main (int argc, char **argv);

const uint32_t length = 5;

// main is the entrypoint to the application.
int main(int argc, char **argv) {
  D[0] = (uint8_t *) malloc(4*sizeof(int32_t));
  if ( D[0] == NULL ) {
    perror("execution of malloc() failed!\n");
    return -1;
  }
  int32_t *P = (int32_t*)D[0];
  P[0] = 0;
  P[1] = 0;
  P[2] = length;
  P[3] = 32;

  return ghdl_main(argc, argv);
}

#>> vhpidirect_user.h
#include <stdint.h>

extern int ghdl_main (int argc, char **argv);

uint8_t *D[256];

void set_string_ptr(uint8_t id, uint8_t *p) { D[id] = p; }
uintptr_t get_string_ptr(uint8_t id) { return (uintptr_t)D[id]; }
void write_char(uint8_t id, uint32_t i, uint8_t v) { ((uint8_t*)D[id])[i] = v; }
uint8_t read_char(uint8_t id, uint32_t i) { return ((uint8_t*)D[id])[i]; }

void set_intvec_ptr(uint8_t id, uint8_t *p) { D[id] = p; }
uintptr_t get_intvec_ptr(uint8_t id) { return (uintptr_t)D[id]; }
void write_integer(uint8_t id, uint32_t i, int32_t v) { ((int32_t*)D[id])[i] = v; }
int32_t read_integer(uint8_t id, uint32_t i) { return ((int32_t*)D[id])[i]; }

#>> sim.sh
apt install git
git clone --recurse-submodules --branch feat-extmem http://github.com/dbhi/vunit /tmp/vunit
export PYTHONPATH=/tmp/vunit
python3 run.py -v

#>> run.sh
$(command -v winpty) docker run --rm -tv /$(pwd)://src -w //src aptman/dbhi:bionic sh -c ./sim.sh

#>> end
```

tgingold, 2019-05-07:
Ok, I was able to reproduce it, despite the missing run.py !

umarcor, 2019-05-07:
I'm sorry. I forgot to add it at first. But I added it afterwards.

closed 2019-05-08 by tgingold

umarcor, 2019-05-08:
As fast and effective as always. Thanks a lot!

commit a5b13e2e (2019-05-08): vhdl: remove too strict assertion in canon.  Fix #816