ghdl/ghdl #818: Syntax for condition operator

umarcor, 2019-05-10:
I was trying to use `if falling_edge(clk) and tg then`, where `tg` is of type `std_logic`, and GHDL complained about it. So, I did multiple tests and these are the results:

``` vhdl
1. if ?? tg then                             -- ok
2. if tg then                                -- ok

3. if falling_edge(clk) and (tg) then        -- no function declarations for operator "and"
4. if falling_edge(clk) and tg then          -- no function declarations for operator "and"
5. if falling_edge(clk) and ?? tg then       -- '??' must be the first operator of an expression
6. if falling_edge(clk) and (??tg) then      -- ok

7. if ?? falling_edge(clk) and (tg) then     -- 'then' is expected here | (found: 'and') | 'end' is expected instead of 'and'
8. if ?? falling_edge(clk) and tg then       -- 'then' is expected here | (found: 'and') | 'end' is expected instead of 'and'
9. if ?? falling_edge(clk) and ?? tg then    -- 'then' is expected here | (found: 'and') | 'end' is expected instead of 'and'
10. if ?? falling_edge(clk) and (??tg) then  -- 'then' is expected here | (found: 'and') | 'end' is expected instead of 'and'

11. if (tg) and falling_edge(clk) then       -- no function declarations for operator "and"
12. if tg and falling_edge(clk) then         -- no function declarations for operator "and"
13. if ?? tg and falling_edge(clk) then      -- 'then' is expected here | (found: 'and') | 'end' is expected instead of 'and'
14. if (??tg) and falling_edge(clk) then     -- ok

15. if ?? (tg) and falling_edge(clk) then    -- 'then' is expected here | (found: 'and') | 'end' is expected instead of 'and'
16. if ?? tg and falling_edge(clk) then      -- 'then' is expected here | (found: 'and') | 'end' is expected instead of 'and'
17. if (??tg) and falling_edge(clk) then     -- ok
```

I find 7-10 strange, but I would expect all others to work.

tgingold, 2019-05-11:
I am pretty sure ghdl is correct.  Conversion operator can only appear before a primary and as an expression.

Implicit conversion is also done only for a whole condition.

umarcor, 2019-05-11:
I now tried all of them on Aldec Riviera Pro 2017.02 (through edaplayground). Almost all of the cases produce equivalent results to GHDL, except for `5.` (`falling_edge(clk) and ?? tg`), which is accepted by Riviera Pro but not by GHDL. Is it a bug in Riviera?

Precisely, that is the case that I am seeking, as it can be extended to `falling_edge(clk) and ??(tg and sig and last)`, which is also accepted by Riviera but not by GHDL.

diogratia, 2019-05-11:
ghdl relies on (9.2.9):

>In certain circumstances, the condition operator is implicitly applied to an expression that occurs as a condition in any of the following places:  
>>...  
>>— After if or elsif in an if statement (see 10.8)  
>>...  

where the semantic description is exact. *"the condition operator is implicitly applied to an expression that occurs as a condition"*.  As a condition as opposed to as an operand of a logical operator in an expression comprising a condition.

For 5.  See (9.1) where the only way to get a unary operator and expression as the right operand of a binary logical operator is as an expression within parentheses.

For the right relation of a logical expression - relation { and relation }, relation -> shift_expression -> simple_expression -> term -> factor -> primary [** primary] (without the optional exponentiation) -> (expression), expression -> conditional_operator primary.

>Each primary has a value and a type. The only names allowed as primaries are attributes that yield values and names denoting objects or values. In the case of names denoting objects other than objects of file types or protected types, the value of the primary is the value of the object. In the case of names denoting either file objects or objects of protected types, the value of the primary is the entity denoted by the name.

Operator symbols while names aren't primaries in expressions, which likely relates to the behavior displayed by Aldec Riviera Pro  2017.02 should you treat operator symbols as primaries. (A missing semantic restriction.)

You could note
```vhdl
if falling_edge(clk) and (?? (tg and sig and last)) then
```
requires parentheses for the condition operator primary while 
```vhdl
if tg and sig and last then
```
does not because the condition is a logical expression with an implicit condition operator.

1. works without parentheses
```vhdl
if ?? tg then
```
because the primary is the name of an object.

ghdl appears to apply condition operator syntax and semantics correctly for all the shown cases.

There appears to be no relationship with support for non-static expressions as port map actuals.

1138-4EB, 2019-05-11:
Thanks for the very detailed explanation. I believe we can close this, as the syntax is clear.

umarcor, 2019-05-11:
Thanks @diogratia for all the details. I'm closing this, since the issue is clear.

commit 36b54159 (2019-05-11): vhdl-parse: improve error messages.  Fix #818
closed 2019-05-11 by umarcor