ghdl/ghdl #821: Crash on unconstrained string in record

slaweksiluk, 2019-05-16:
**Context**
```
[root@8627b5b0311f workdir]# ./run.sh 

******************** GHDL Bug occurred ***************************
Please report this bug on https://github.com/ghdl/ghdl/issues
GHDL release: 0.37-dev (v0.36-136-gd5d8471e) [Dunoon edition]
Compiled with GNAT Version: 8.3.1 20190223 (Red Hat 8.3.1-2)
Target: x86_64-redhat-linux
In directory: /workdir/
Command line:
/usr/local/bin/ghdl1-llvm -g --std=08 -P/usr/local/lib/ghdl/ieee/v08/ -P/usr/local/lib/ghdl/ -c -o a.o a.vhd
Exception CONSTRAINT_ERROR raised
Exception information:
raised CONSTRAINT_ERROR : trans.adb:1443 access check failed
Call stack traceback locations:
0x47ecab 0x47e1a7 0x5c9658 0x5c93ea 0x5706b9 0x589361 0x589541 0x60de23 0x5af4ee 0x592864 0x6173e3 0x560c24 0x61d977 0x61f047 0x62089f 0x409009 0x7f9e7f203411 0x40824c 0xfffffffffffffffe
******************************************************************
```

- OS:
- Origin:
  - [ ] Package manager. Repo:
  - [ ] Released binaries. Tarball:
  - [x] Built from sources. Commit SHA: d5d8471e

**How to reproduce?**

```
#>> ent.vhd
entity tb is
	generic (output_path : string := "");
end entity;
architecture bench of tb is
  type tb_cfg_t is record
    s1 : string;
    s2 : string;
  end record tb_cfg_t;

  impure function decode return tb_cfg_t is
  begin
	return (
		s1 => output_path&"a",
		s2 => "b"
	);
  end function decode;
begin
	proc : process begin
		std.env.finish;
	end process;
end bench;

#>> sim.sh
ghdl -a -g --std=08 ent.vhd
ghdl -e -g --std=08 tb
./tb

#>> run.sh
docker run --rm -tv /$(pwd):/src:z -w //src ghdl/ghdl:buster-mcode sh -c ./sim.sh

#>> end
```

**Checklist**
Before submitting your issue, please review the following checklist:

- [ x] Add `GHDL Bug occurred` log block
- [ x] Add a MWE
- [ x] Try the latest version

tgingold, 2019-05-17:
I can reproduce the crash, although not with the mcode backend.

eine, 2020-08-05:
This seems to be fixed in latest master. @slaweksiluk can you please confirm?

slaweksiluk, 2020-08-06:
Confirmed

closed 2020-08-06 by slaweksiluk