ghdl/ghdl #822: sysmalloc assertion error

umarcor, 2019-05-18:
**Description**

With LLVM backend, trying to allocate a variable of type `array (1 to integer'high) of integer` is successfully analyzed and elaborated, but it fails at runtime:

``` bash
# ./test
test: malloc.c:2401: sysmalloc: Assertion `(old_top == initial_top (av) && old_size == 0) || ((unsigned long) (old_size) >= MINSIZE &&
 prev_inuse (old_top) && ((unsigned long) old_end & (pagesize - 1)) == 0)' failed.
Aborted
```

With `mcode`, GHDL crashes:

```
******************** GHDL Bug occurred ***************************
Please report this bug on https://github.com/ghdl/ghdl/issues
GHDL release: 0.36-rc1 (v0.36-rc1-4-gb4597836) [Dunoon edition]
Compiled with GNAT Version: 8.2.0
Target: x86_64-linux-gnu
In directory: /src/
Command line:
ghdl -e --std=08 test
Exception CONSTRAINT_ERROR raised
Exception information:
raised CONSTRAINT_ERROR : ortho_code-exprs.adb:999 range check failed
Call stack traceback locations:
0x55f90089510d 0x55f9008ac4f3 0x55f90094372d 0x55f9009f6148 0x55f9009f62ba 0x55f900a0ca52 0x55f900a11
2a8 0x55f900a7c9a2 0x55f9009e8bfb 0x55f9009e5d67 0x55f9009f39b5 0x55f900a7fbff 0x55f9009ca4a2 0x55f90
0951232 0x55f900a829b2 0x55f90078e46d 0x7fe639edb099 0x55f90078d108 0xfffffffffffffffe
******************************************************************
```

**Additional context**

`array (1 to integer'high/4) of integer` or smaller sizes do not produce the assertion error with LLVM.

With `mcode`, `array (1 to integer'high/4) of integer` produces a different bug message, but `array (1 to integer'high/8) of integer` works:

```
******************** GHDL Bug occurred ***************************
Please report this bug on https://github.com/ghdl/ghdl/issues
GHDL release: 0.36-rc1 (v0.36-rc1-4-gb4597836) [Dunoon edition]
Compiled with GNAT Version: 8.2.0
Target: x86_64-linux-gnu
In directory: /src/
Command line:
ghdl -e --std=08 test
Exception CONSTRAINT_ERROR raised
Exception information:
raised CONSTRAINT_ERROR : ortho_code-x86-emits.adb:482 overflow check failed
Call stack traceback locations:
0x55ddaa57a7b8 0x55ddaa57a77c 0x55ddaa57aae3 0x55ddaa57be55 0x55ddaa57f0cc 0x55ddaa58031d 0x55ddaa581
007 0x55ddaa5811c3 0x55ddaa57259a 0x55ddaa58a7cb 0x55ddaa6eaad6 0x55ddaa6ef2a8 0x55ddaa75a9a2 0x55dda
a6c6bfb 0x55ddaa6c3d67 0x55ddaa6d19b5 0x55ddaa75dbff 0x55ddaa6a84a2 0x55ddaa62f232 0x55ddaa7609b2 0x5
5ddaa46c46d 0x7fda921cd099 0x55ddaa46b108 0xfffffffffffffffe
******************************************************************
```

**How to reproduce?**

```
#>> test.vhd
entity test is
end;

architecture arch of test is
  type alloc_t is array (1 to integer'high/2) of integer;
begin
    process
      variable alloc: alloc_t := (others=>0);
    begin
        wait;
    end process;
end;

#>> sim.sh
ghdl -a --std=08 test.vhd
ghdl -e --std=08 test
./test

#>> run.sh
docker run --rm -tv /$(pwd):/src:z -w //src ghdl/ghdl:buster-mcode sh -c ./sim.sh

#>> end
```

tgingold, 2019-05-18:
The size of objects is limited to 4GB I think.

umarcor, 2019-05-18:
> The size of objects is limited to 4GB I think.

The maximum size I tried to allocate is `1 to integer'high of integer`, hence `2147483648 * 4 /1024 /1024 /1024 = 8GB`. It makes sense for it to fail.

However, `1 to integer'high/2 of integer` is 4GB, and it fails too.

The maximum allowed size with LLVM seems to be `1 to integer'high/4 of integer`, 2GB. For mcode, it seems to be `1 to integer'high/8 of integer`, 1GB.

I think I am missing something, since I would expect both backends to fail with the same limits. Moreover, according to [#611](https://github.com/ghdl/ghdl/issues/611#issuecomment-445444610), GHDL might be able to produce a legit warning/error instead of crashing.