ghdl/ghdl #823: [question] purpuse of --link flag

arcturus140, 2019-05-20:
ghdl help lists a link flag: 

    --link [OPTS] UNIT [ARCH]  Link UNIT

I overlooked this option in the documentation. What is the purpose and usage for this option? Is it similar to `list-link`?

1138-4EB, 2019-05-20:
It is provided only for completeness: https://ghdl.readthedocs.io/en/latest/references/CommandReference.html#link-link

After you have analyzed the sources with `ghdl -a` you can:

- `ghdl -e`
- `ghdl --bind` + `ghdl --list-link` + `ghdl --link`
- `ghdl --bind` + `ghdl --list-link` + `ghdl -e`
- `ghdl --bind` + `ghdl --list-link` + `gcc`

closed 2019-05-20 by arcturus140

tgingold, 2019-05-20:
Note that `ghdl --list-link` does not create any file, it just dumps the files to be linked.

`ghdl -e` is equivalent to `ghdl --bind` followed by `ghdl --link`.

1138-4EB, 2019-05-20:
> Note that `ghdl --list-link` does not create any file, it just dumps the files to be linked. `ghdl -e` is equivalent to `ghdl --bind` followed by `ghdl --link`.

I misclicked and sent the first message after C&P some lines but before cleaning them. I hope it is correct now.