ghdl/ghdl #826: entity obsoleted by std_logic_1164

stfarley, 2019-05-23:
**Description**

When I try and compile th.vhd with
ghdl -a --std=08 --ieee=standard -P/projects/ghdl_libs/osvvm/v08 -frelaxed-rules -Plibs --workdir=libs --work=testbench  src/testbench/th.vhd &>> logs/testbench/th

I get this error message about entity "xyz" instantiated in tx
src/testbench/th.vhd:228:22: entity "xyz" is obsoleted by package "std_logic_1164"

Using GHDL 0.37-dev (v0.36-150-g0611b41) [Dunoon edition]
 Compiled with GNAT Version: 4.8.5 20150623 (Red Hat 4.8.5-36
 GCC back-end code generator

**Expected behaviour**
I think this should compile without issues.

- OS:Centos 7
- Origin:
  - [ ] Package manager. Repo:
  - [ ] Released binaries. Tarball:
  - [ X] Built from sources. Commit SHA:0611b41

**How to reproduce?**
Not sure how I can isolate this to a smaller Test case as it only occurs with one entity.
```
**Checklist**
Before submitting your issue, please review the following checklist:

- [? ] Add `GHDL Bug occurred` log block
- [ ?] Add a MWE
- [ Y] Try the latest version

tgingold, 2019-05-23:
Usually it is because you have updated ghdl.  Simply reanalyze the dependencies (ovvm) and your design.

stfarley, 2019-05-23:
I re-compiled osvvm after re-compiling ghdl (Switched to gcc backend) then recompiled everything in the design. Scripted flow that simply removes the compiled libraries and re-compiles everything. 

I gather this is a compile order message telling me that ieee.std_logic_1164 has been re-compiled?
I assume that this is part of the ghdl comp and install process, so any re-build of ghdl would cause it.
However it should not 

The strange thing is I use std_logic_1164 everywhere. It only fails on this one instantiation.

tgingold, 2019-05-23:
Why 'it should not' ?  This is simply the VHDL rules.  The dependencies must be analyzed before the unit.

I cannot really comment why it fails only on one instantiation (?) without a reproducer.

stfarley, 2019-05-23:
Thanks for your quick response tgingold and for your hard work on ghdl for that matter.

Anyhow I got to the bottom of it. I guess I missed the nuance that it always searches the cwd regardless of the -P switches. There were a couple of old objects there from some test compiles that it was picking up, Since I wasn't actually copileing anything to the cwd I wasn't looking there. Deleting them fixed the problem.

Anyhow Thanks Again

closed 2019-05-23 by tgingold