ghdl/ghdl #827: TYPES.INTERNAL_ERROR : trans.adb:619 with sliced port assignments containing generics

bwimm, 2019-05-24:
**Description**
GHDL crashes with an internal error when assigning entity ports to slices containing generics (see code)

**Expected behaviour**
Successful elaboration.

**Context**
```
******************** GHDL Bug occurred ***************************
Please report this bug on https://github.com/ghdl/ghdl/issues
GHDL release: 0.37-dev (v0.36-156-ge256016d) [Dunoon edition]
Compiled with GNAT Version: 7.4.0
Target: x86_64-linux-gnu
/home/<user>/
Command line:
/home/<user>/Desktop/ghdl/install/bin/ghdl1-llvm --std=08 -P/home/<user>/Desktop/ghdl/install/lib/ghdl/ -c -fpic /home/<user>/Desktop/vhdl/ent.vhd
Exception TYPES.INTERNAL_ERROR raised
Exception information:
raised TYPES.INTERNAL_ERROR : trans.adb:619
Call stack traceback locations:
0x47c90c 0x485e89 0x6061b4 0x615b96 0x615c73 0x5fce0e 0x5c65a2 0x5c68ae 0x5c6ea5 0x5c74de 0x5bf1a7 0x5bf9dc 0x59fee6 0x628287 0x56ca4d 0x62ebe4 0x630394 0x631ca7 0x408671 0x7fe9b0600b95 0x4079e8 0xfffffffffffffffe
******************************************************************
```
- OS: Ubuntu 18.04.02
- Origin:  Built from sources. Commit SHA: e256016

**How to reproduce?**
The following vhdl snippet reproduces the issue (may not be minimal but is significantly reduced from the original code that triggered the issue)

```
library IEEE;
use IEEE.std_logic_1164.all;
use IEEE.numeric_std.all;

entity B is
	generic(
		GENERIC_VALUE : positive := 2
	);
	port(
		output : out std_ulogic_vector(31 downto 0)
	);
end B;

library IEEE;
use IEEE.std_logic_1164.all;
use IEEE.numeric_std.all;

entity A is
	port(
		out_a : out unsigned(15 downto 0);
		out_b : out unsigned(15 downto 0)
	);
end A;

architecture struct of B is
begin
	a_0 : entity work.A
	port map(
		std_ulogic_vector(out_a) => output(output'high downto output'high-GENERIC_VALUE*8+1),
		std_ulogic_vector(out_b) => output(15 downto 0)
	);

end architecture;
```

diogratia, 2019-05-24:
You can change the type of port `output` to unsigned and remove the type conversions on formals `out_a` and `out_b` and the code successfully analyzes. That tells us the use of generics has no bearing on the problem, the title of the issue isn't accurate.

(With the changes you can also add a dummy architecture for A and the code will elaborate and simulate).

See IEEE Std 1076-2008 6.5.7 Association lists, 6.5.7.1 General, paragraph 8:

>Alternatively, the formal part of a named association element may be in the form of a type conversion, where the expression to be converted is the formal designator itself, if and only if the formal is an interface object, the mode of the formal is **out**, **inout**, **buffer**, or **linkage**, and if the actual is not **open**. In this case, the base type denoted by the type mark shall be the same as the base type of the corresponding actual. Such a type conversion provides for type conversion in the event that data flows from the formal to the actual. It is an error if the type of the formal is not closely related to the type of the actual. (See 9.3.6.)

ghdl appears to fail semantic evaluation here due to the formal part expression with the type conversion.

The error also occurs for --std=93c, and the above changes for changing the type of `output` also corrects the issue. (It's also legal, IEEE Std 1076-1993 4.3.2.2 Association lists, para 8.)

For patching around the issue use one or two intermediary signals of type unsigned as actuals for `out_a` and `out_b`.

ghdl could likely benefit from some test cases.

bwimm, 2019-05-25:
Thanks for the detailed reply.

I changed the title to remove the guess.
Replacing the generic with its value also made it compile successfully which is why I assumed it had something to do with it.

Intermediate signals did indeed fix the issue in the original code as well.

tgingold, 2019-05-26:
Confirmed.

Paebbels, 2025-04-27:
@tgingold is this one still open or already fixed?

tgingold, 2025-04-27:
It is not fully fixed (although working correctly with jit backends)