ghdl/ghdl #828: reverse_range attribute sometimes gives incorrect range

bradleyharden, 2019-05-27:
**Description**
In some situations, the `reverse_range` attribute does not return the correct range.

**Expected behaviour**
I have observed two different behaviors when looping over the `reverse_range` of an array. In one case, the loop was only executed once, even though the length of the array was 8. In another case, the loop executed 9 times, even though the loop was length 8.

**Context**
- OS: Ubuntu 18.04 with LLVM 6.0
- Origin:
  - [ ] Package manager. Repo:
  - [ ] Released binaries. Tarball:
  - [x] Built from sources. Commit SHA: 3fafb13

**How to reproduce?**
```vhdl
library ieee;
use ieee.std_logic_1164.all;

entity test is
  port(
    tx : out std_logic);
end entity;

architecture tb of test is
begin
  process

    procedure transmit(data : std_logic_vector;
                       signal tx : out std_logic) is

      variable norm : std_logic_vector(data'length - 1 downto 0) := data;

      procedure send(value : std_logic) is
      begin
        tx <= value;
        wait for 10 ns;
      end procedure;

    begin
      for i in norm'reverse_range loop
        send(norm(i));
        report to_string(i);
      end loop;
    end procedure;

  begin

    transmit(x"55", tx);
    wait;

  end process;
end architecture;
```

```bash
$ ghdl -a --std=08 test.vhd
$ ghdl -e --std=08 test
$ ./test
```
Observe that the loop is only executed once. If you change `reverse_range` to `range`, the loop is executed 8 times. Adding the signal parameter to the procedure was necessary to reproduce the bug. I have no idea why though.

The MCVE was developed from `uart_master` in VUnit. There, I was seeing `reverse_range` produce 9 iterations of the loop in a similar example.

**Checklist**
Before submitting your issue, please review the following checklist:

- [ ] Add `GHDL Bug occurred` log block
- [x] Add a MWE
- [x] Try the latest version

diogratia, 2019-05-28:
The test case is modified to make it revision independent (-193 through -2008) and to demonstrate whether the problems occurs in sequential statements in processes or just a subprogram (here procedure):

```vhdl
library ieee;
use ieee.std_logic_1164.all;

entity test is
  port (
    tx : out std_logic);
end entity;

architecture tb of test is
begin
    process
        -- variable boolobj: boolean; -- ADDED
        procedure transmit(data: std_logic_vector; signal tx: out std_logic) is
            variable norm: std_logic_vector(data'length - 1 downto 0) := data;

            procedure send(value: std_logic) is
            begin
                tx <= value;
                wait for 10 ns;
            end procedure;
        begin
        -- report " boolobj = " & boolean'image(boolobj); -- ADDED
            for i in norm'reverse_range loop
                send(norm(i));
                report integer'image(i); --- to_string(i); -- CHANGED
            end loop;
        end procedure;
        variable norm: std_logic_vector(7 downto 0); -- ADDED
    begin
        transmit(x"55", tx);
        for i in norm'reverse_range loop     -- ADDED Loop statement
            report "i = " & integer'image(i);
        end loop;
        wait;
    end process;
end architecture;
```
This shows the problem is isolated to subprogram execution:
```none
ghdl -a test.vhdl
test.vhdl:13:59:warning: declaration of "tx" hides port "tx" [-Whide]
ghdl -e test
ghdl -r test
test.vhdl:25:17:@10ns:(report note): 0
test.vhdl:25:17:@20ns:(report note): 1
test.vhdl:32:13:@20ns:(report note): i = 0
test.vhdl:32:13:@20ns:(report note): i = 1
test.vhdl:32:13:@20ns:(report note): i = 2
test.vhdl:32:13:@20ns:(report note): i = 3
test.vhdl:32:13:@20ns:(report note): i = 4
test.vhdl:32:13:@20ns:(report note): i = 5
test.vhdl:32:13:@20ns:(report note): i = 6
test.vhdl:32:13:@20ns:(report note): i = 7

ghdl --version
GHDL 0.36 (tarball) [Dunoon edition]
 Compiled with GNAT Version: GPL 2015 (20150428-49)
 llvm code generator
Written by Tristan Gingold.

Copyright (C) 2003 - 2019 Tristan Gingold.
GHDL is free software, covered by the GNU General Public License.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
```
Now with the boolean object declaration and report statement evaluating it uncommented:
```none
ghdl -a test.vhdl
test.vhdl:13:59:warning: declaration of "tx" hides port "tx" [-Whide]
ghdl -e test
ghdl -r test
test.vhdl:22:9:@0ms:(report note):  boolobj = false
test.vhdl:25:17:@10ns:(report note): 0
test.vhdl:25:17:@20ns:(report note): 1
test.vhdl:25:17:@30ns:(report note): 2
test.vhdl:25:17:@40ns:(report note): 3
test.vhdl:25:17:@50ns:(report note): 4
test.vhdl:25:17:@60ns:(report note): 5
test.vhdl:25:17:@70ns:(report note): 6
test.vhdl:25:17:@80ns:(report note): 7
test.vhdl:32:13:@80ns:(report note): i = 0
test.vhdl:32:13:@80ns:(report note): i = 1
test.vhdl:32:13:@80ns:(report note): i = 2
test.vhdl:32:13:@80ns:(report note): i = 3
test.vhdl:32:13:@80ns:(report note): i = 4
test.vhdl:32:13:@80ns:(report note): i = 5
test.vhdl:32:13:@80ns:(report note): i = 6
test.vhdl:32:13:@80ns:(report note): i = 7
```

Evaluating an object prior to the for loop allows the loop to perform as intended. This pretty much tells us the reverse_range isn't properly evaluating the range in the procedure call.

I tested various statements prior to the for loop in the procedure body. It appears a requirement that an object value be evaluated.

The testcase can be modified to assert a comparison between simulation time and expected time to demonstrate expected functionality when the issue is corrected.

tgingold, 2019-05-28:
I have a fix for this issue.  I will push it soon.

commit b2e6d5df (2019-05-28): trans-chap8: reverse_range returns false for is_for_loop_iterator_stable.

Fix #828
closed 2019-05-28 by tgingold