ghdl/ghdl #829: Parallel Analysis not working

mikey, 2019-05-30:
**Description**
Parallel Analysis doesn't work. This seems to be due to two ghdl instances trying to write to the library file.

**Expected behaviour**
You should be able to Analyse in parallel for use with `make -j`

**Context**
- OS: Linux (but I assume others
- Origin:
  - [ X ] Built from sources. Commit SHA: Latest upstream 5035b126d9

**Additional context**
Add any other context about the problem here. If applicable, add screenshots to help explain your problem.

**How to reproduce?**
```

% cat a.vhdl 
entity a is
end a;
architecture behav of a is
begin
end behav;
% cat b.vhdl 
entity b is
end b;
architecture behav of b is
begin
end behav;
% # Normal serial build works
% ghdl -a a.vhdl ; ghdl -a b.vhdl
% cat work-obj93.cf 
v 4
file . "b.vhdl" "a87ccd5ebb2cf7e927c48522e6c90a7cc480e9f5" "20190530004107.133":
  entity b at 1( 0) + 0 on 13;
  architecture behav of b at 3( 19) + 0 on 14;
file . "a.vhdl" "0ae1f52e3d5838da94996b793a4dbb5e8f0d311e" "20190530004107.106":
  entity a at 1( 0) + 0 on 11;
  architecture behav of a at 3( 19) + 0 on 12;
% \rm work-obj*

% # now as parallel build
% ghdl -a a.vhdl & ghdl -a b.vhdl & wait
[1] 23239
[2] 23240
[1]-  Done                    ghdl -a a.vhdl
[2]+  Done                    ghdl -a b.vhdl
% cat work-obj93.cf 
v 4
file . "b.vhdl" "a87ccd5ebb2cf7e927c48522e6c90a7cc480e9f5" "20190530004124.927":
  entity b at 1( 0) + 0 on 11;
  architecture behav of b at 3( 19) + 0 on 12;
% 
```
The second parallel build is missing the a.vhdl entry in  work-obj93.cf 

Also sometimes results in `ghdl:error: cannot update library file "work-obj93.cf"`

**Checklist**
Before submitting your issue, please review the following checklist:

- [ ] Add `GHDL Bug occurred` log block
- [ X ] Add a MWE
- [ X ] Try the latest version

tgingold, 2019-05-30:
This is an interesting problem I have already thought about.

But first, which back-end are you using ? (what is the output of `ghdl --version`)

mikey, 2019-05-30:
> But first, which back-end are you using ? (what is the output of ghdl --version)

```
% ghdl --version
GHDL 0.37-dev (v0.36-183-g5035b126d9) [Dunoon edition]
 Compiled with GNAT Version: 9.1.1 20190503 (Red Hat 9.1.1-1)
 mcode code generator
Written by Tristan Gingold.

Copyright (C) 2003 - 2019 Tristan Gingold.
GHDL is free software, covered by the GNU General Public License.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
%
```

tgingold, 2019-05-30:
As you are using the mcode back-end, you won't gain much speed by trying to parallelise.  You'd better to analyze many files in the same command: `ghdl -a f1.vhdl f2.vhdl ...`

mikey, 2019-05-30:
On my machine I do see a big benefit with parallelism. > 10x speedup below.

```

% cat ./ghdl-parallel 
#!/bin/bash

NR=1000

D=$(mktemp -d)

cd $D

for i in $(seq $NR); do
    F=a$i
    cat > $F.vhdl << EOF
entity $F is
end $F;
architecture behav of $F is
begin
end behav;
EOF

done

for i in $(seq $NR); do
    if [ -n "$1" ] ; then
	ghdl -a --work=a$i a$i.vhdl &
    else
	ghdl -a --work=a$i a$i.vhdl
    fi
done

wait

rm -rf $D
% time ./ghdl-parallel

real	0m15.920s
user	0m11.208s
sys	0m4.600s
% time ./ghdl-parallel  1

real	0m1.295s
user	0m21.243s
sys	0m8.154s
% nproc
128
%

```

tgingold, 2019-05-30:
But:
1. you haven't compared with many files on the command line
2. common vhdl files have many dependencies (like ieee.std_logic_1164)

mikey, 2019-05-30:
> 1. you haven't compared with many files on the command line

Sorry, I forgot that. Yes that does help a bunch for the mcode backend.

My 128 CPU machine is using the GCC backend so it doesn't help there but on my laptop with the mcode backend, running all in one command helps a bunch and is even faster with a single command.

FWIW updated test case:
```
#!/bin/bash

NR=1000

D=$(mktemp -d)

cd $D

for i in $(seq $NR); do
    F=a$i
    cat > $F.vhdl << EOF
entity $F is
end $F;
architecture behav of $F is
begin
end behav;
EOF

done

for i in $(seq $NR); do
    if [ -n "$1" ] ; then
	ghdl -a --work=a$i a$i.vhdl &
    else
	FILES="$FILES a$i.vhdl"
    fi
done

if [ -n "$1" ] ; then
    wait
else
    ghdl -a $FILES
fi

rm -rf $D
```
> 2 common vhdl files have many dependencies (like ieee.std_logic_1164)

A mostly static library like ieee.std_logic_1164 would already be compiled and wouldn't affect how much you could run in parallel. Other libraries could create dependencies but that would be for `make` to resolve.

tgingold, 2019-05-30:
If you are using the gcc backend on a big machine, that's a different story.  I agree with you that parallel compilation would be much faster in that case.

Right now there is no support for parallel compilation.

Here is the current idea I have to support it:
1. First sequentially analyse the files, but without generating any object code.  This will update the library file.
2. Then generate code in parallel, but without updating the library file.
3. Finally do the elaboration.

I haven't think about the integration with make.  But the parallel part could be handled by ghdl.

mikey, 2019-05-30:
Could you make the library a directory rather than a file? The you could store each entity in a different file and avoid the conflict on the write side. Obviously this would have implications on the reader side.

tgingold, 2019-05-30:
I am not convinced this is the best approach.  First it creates more dependencies on the OS (you need to read the directory to get the whole content of a library), it doesn't create an analysis order, and last you may still have conflicts (if you analyze the same file concurrently, or if an entry is deleted).

I really fear it will create more problems.

sean-anderson-seco, 2023-05-09:
For a workaround, you can serialize analysis only with something like

```
flock work-obj93.cf.lock ghdl -a ...
```

Note that you must use a separate file for the lock because GHDL uses file renaming to update the library. If you try to lock the library file, later processes will step on each others' toes because they will all be locking different files.

---

Have you considered sqlite? It supports consistent, concurrent reads/updates.

tgingold, 2023-05-09:
According to the vhdl standard, there is an order of analysis, which is contradictory to concurrent analysis.

What could be done is to separate analysis (which could be very fast) and code generation (which tends to be slow).

sean-anderson-seco, 2023-05-09:
> According to the vhdl standard, there is an order of analysis, which is contradictory to concurrent analysis.

Doesn't `--gen-depends` sufficiently specify the analysis order? If you have two modules and neither one references the other, does it matter which one is analyzed first?

tgingold, 2023-05-10:
Yes, it does matter if the two files declare an architecture some the same entity.

sean-anderson-seco, 2023-05-11:
Which is why I specified that neither module referenced the other.

tgingold, 2023-05-12:
You can have 3 files:
* e.vhd: for the entity
* a1.vhd: for one architecture
* a2.vhd: for another architecture

The order of analysis between a1.vhd and a2.vhd matters but is not defined.

sean-anderson-seco, 2023-05-12:
The order should of course be defined.

That is, in your scenario, the dependencies should be something like

```make
e: e.o a1.o a2.o
e.o: e.vhd
a1.o: a1.vhd e.o
a2.o: a2.vhd e.o a1.o
```

or the other way around. This unambiguously sets the order of analysis for these files only, while allowing parallel analysis for other files.

umarcor, 2023-05-16:
> ```
> e: e.o a1.o a2.o
> e.o: e.vhd
> a1.o: a1.vhd e.o
> a2.o: a2.vhd e.o a1.o
> ```

What should happen if the user then edits file `a1.vhd`? Should it be moved to the end, thus making it "dependent" on `a2.o`? Or should both a1 and a2 be reanalyzed/elaborated again even though they do not depend on each other?

As far as I am aware, these details are not specified in the LRM. I believe that's what Tristan means with "undefined". Of course, there is an order inside GHDL, but that's not "canonical".

This is related to the fact that sometimes "the last analyzed architecture" is picked, and that "last one" might be tool dependent unless the user specifies the order very explicitly.

sean-anderson-seco, 2023-05-16:
>   What should happen if the user then edits file `a1.vhd`? Should it be moved to the end, thus making it "dependent" on `a2.o`? Or should both a1 and a2 be reanalyzed/elaborated again even though they do not depend on each other?

The dependencies should be regenerated. So if, for example, the dependencies are stored in `deps`, then the full dependencies should look like

```make
e: e.o a1.o a2.o
e.o: e.vhd
a1.o: a1.vhd e.o
a2.o: a2.vhd e.o a1.o
deps: e.vhd a1.vhd a2.vhd
```

> As far as I am aware, these details are not specified in the LRM. I believe that's what Tristan means with "undefined". Of course, there is an order inside GHDL, but that's not "canonical".

So we are free to pick something reasonable. In fact, this works for us, since the order of analysis of unrelated sources does not matter.

> This is related to the fact that sometimes "the last analyzed architecture" is picked, and that "last one" might be tool dependent unless the user specifies the order very explicitly.

Yes, if you use `--gen-depends` the last-analyzed architecture is dependent on whatever GHDL picks (I'm sure there is a specific order but it's opaque to users). IMO if you have multiple architectures and care which one gets picked you should specify the one you want explicitly, or you should do your analysis serially.