ghdl/ghdl #831: GHDL does not build

Stefichen5, 2019-06-03:
**Description**
The current version of ghdl can not be built successfully. I run the following:

> git pull
> make clean
> ./configure --prefix=/usr/local
> make

**Expected behaviour**
Build completing successfully

**Context**
[...]
x86_64-linux-gnu-gcc-8 -c -I./ -I./src -I./src/vhdl -I./src/grt -I./src/psl -I./src/vhdl/translate -I./src/ghdldrv -I./src/ortho -I./src/ortho/mcode -gnaty3befhkmr -gnatwa -gnatf -g -gnata -gnatwe -gnatw.A -I- /home/stefan/Dokumente/ghdl/src/vhdl/vhdl-errors.adb
x86_64-linux-gnu-gcc-8 -c -I./ -I./src -I./src/vhdl -I./src/grt -I./src/psl -I./src/vhdl/translate -I./src/ghdldrv -I./src/ortho -I./src/ortho/mcode -gnaty3befhkmr -gnatwa -gnatf -g -gnata -gnatwe -gnatw.A -I- /home/stefan/Dokumente/ghdl/src/vhdl/vhdl-formatters.adb
vhdl-formatters.adb:474:07: warning: use clause for package "Types" has no effect
gnatmake: "/home/stefan/Dokumente/ghdl/src/vhdl/vhdl-formatters.adb" compilation error
make: *** [Makefile:183: ghdl_mcode] Fehler 4

- OS:
- Origin:
  - [ ] Package manager. Repo:
  - [ ] Released binaries. Tarball:
  - [X] Built from sources. Commit SHA: edfccd332a56771311fefdbfcffcf269c8026aba

**Additional context**
GNATMAKE 8.3.0

1138-4EB, 2019-06-03:
Hi @Stefichen5! Could please tell which OS and backend are you trying to use? I found latest master to fail too, but with different errors depending on the platform and backend:

- mcode:
  - stretch, buster, ubunt18, fedora28 and fedora29 pass successfully.
  - ubuntu14 and ubuntu16 fail with:

``` bash
gnatmake -o ghdl_mcode -aI../src -aI../src/vhdl -aI../src/grt -aI../src/psl -aI../src/vhdl/translate -aI../src/ghdldrv -aI../src/ortho -aI../src/ortho/mcode -gnaty3befhkmr -gnatwa -gnatf -g -gnata -gnatwe -gnatw.A ghdl_jit.adb -bargs -E -largs memsegs_c.o chkstk.o jumps.o times.o grt-cbinding.o grt-cvpi.o grt-cdynload.o fstapi.o lz4.o fastlz.o   -ldl -lm -Wl,--version-script=/work/build-mcode/../src/grt/grt.ver -Wl,--export-dynamic
gcc-4.6 -c -I./ -I../src -I../src/vhdl -I../src/grt -I../src/psl -I../src/vhdl/translate -I../src/ghdldrv -I../src/ortho -I../src/ortho/mcode -gnaty3befhkmr -gnatwa -gnatf -g -gnata -gnatwe -gnatw.A -I- /work/src/grt/grt-to_strings.adb
grt-to_strings.adb:27:09: warning: unit "Unchecked_Conversion" is not referenced
grt-to_strings.adb:28:09: warning: no entities of "Errors" are referenced
gnatmake: "/work/src/grt/grt-to_strings.adb" compilation error
make: *** [ghdl_mcode] Error 4
```

``` bash
gnatmake -o ghdl_mcode -aI../src -aI../src/vhdl -aI../src/grt -aI../src/psl -aI../src/vhdl/translate -aI../src/ghdldrv -aI../src/ortho -aI../src/ortho/mcode -gnaty3befhkmr -gnatwa -gnatf -g -gnata -gnatwe -gnatw.A ghdl_jit.adb -bargs -E -largs memsegs_c.o chkstk.o jumps.o times.o grt-cbinding.o grt-cvpi.o grt-cdynload.o fstapi.o lz4.o fastlz.o   -ldl -lm -Wl,--version-script=/work/build-mcode/../src/grt/grt.ver -Wl,--export-dynamic
gcc-4.9 -c -I./ -I../src -I../src/vhdl -I../src/grt -I../src/psl -I../src/vhdl/translate -I../src/ghdldrv -I../src/ortho -I../src/ortho/mcode -gnaty3befhkmr -gnatwa -gnatf -g -gnata -gnatwe -gnatw.A -I- /work/src/grt/grt-to_strings.adb
grt-to_strings.adb:27:09: warning: unit "Ada.Unchecked_Conversion" is not referenced
gnatmake: "/work/src/grt/grt-to_strings.adb" compilation error
Makefile:183: recipe for target 'ghdl_mcode' failed
make: *** [ghdl_mcode] Error 4
```

- llvm:
  - ubuntu14, ubuntu16 and ubuntu18 pass successfully.
  - fedora28 and fedora29 fail with:

``` bash
gnatmake -aI../src -aI../src/vhdl -aI../src/grt -aI../src/psl -aI../src/vhdl/translate -aI../src/ghdldrv -aI../src/ortho -aI../src/ortho/llvm4-nodebug -aI../src/ghdldrv -gnaty3befhkmr -gnatwa -gnatf -g -gnata -gnatwe\
 ghdl_llvm -bargs -E -largs  
gcc -c -I./ -I../src -I../src/vhdl -I../src/grt -I../src/psl -I../src/vhdl/translate -I../src/ghdldrv -I../src/ortho -I../src/ortho/llvm4-nodebug -I../src/ghdldrv -gnaty3befhkmr -gnatwa -gnatf -g -gnata -gnatwe -I- /work/src/vhdl/vhdl-formatters.adb
vhdl-formatters.adb:474:07: warning: use clause for package "Types" has no effect
gnatmake: "/work/src/vhdl/vhdl-formatters.adb" compilation error
make: *** [Makefile:321: ghdl_llvm] Error 4
```

- mcodegpl: stretch is successful but buster fails with the same error as the llvm above.

Stefichen5, 2019-06-03:
I'm sorry I forgot to fill out the OS information (I have updated it now). I use Ubuntu 19.04.
As for the backend, I did not specify anything explicitly. So I assume it will be mcode.

tgingold, 2019-06-03:
For warnings, you can always use `configure --disable-werror`.  I will fix them.

Stefichen5, 2019-06-03:
Thank you, this helped

closed 2019-06-03 by tgingold