ghdl/ghdl #833: Add 'ghdl' topic to the repository

dominiksalvet, 2019-06-04:
**Is your feature request related to a problem? Please describe.**
During the GHDL development, [`ghdl`](https://github.com/topics/ghdl) GitHub topic has been created by users who use GHDL in their repositories. It might seem weird that in the `ghdl` topic there is not present the GHDL itself.

**Describe the solution you'd like**
Add the `ghdl` topic to this repository.

**Describe alternatives you've considered**
I don't see any reason why `ghdl` topic shouldn't be associated with GHDL.

**Additional context**
I would personally propose adding more relevant GitHub topics to this repository. My suggestions would be, e.g., [`simulation`](https://github.com/topics/simulation), [`testbench`](https://github.com/topics/testbench), [`compiler`](https://github.com/topics/compiler), [`open-source`](https://github.com/topics/open-source), [`linux`](https://github.com/topics/linux), [`macos`](https://github.com/topics/macos), [`windows`](https://github.com/topics/windows), [`llvm`](https://github.com/topics/llvm), [`gcc`](https://github.com/topics/gcc) and even [`hardware`](https://github.com/topics/hardware). That would bring more people to the project and it will eventually reap the related benefits.

1138-4EB, 2019-06-04:
I added some to ghdl/docker: ghdl, dockerfiles, travis-ci, mcode, llvm, gcc, simulation, testbench, vhdl, hardware.

tgingold, 2019-06-04:
I have added most of them.  Thanks!

closed 2019-06-04 by tgingold