ghdl/ghdl #836: GDHL simulator crashes if std_logic is assigned value of (others => '0');

wm2015email, 2019-06-10:
I noticed that I can crash the GHDL simulator consistently  if I create a std_logic signal and then assign it the value of "(others => '0')" from a process.  

The main problem with this is that it crashes unexpectately during simulation leaving the user confused about where the problem occurs in the VHDL rather than flagging it as a syntax error during the compilation phase.

go2sh, 2019-06-10:
Which version are you using? My compilation failed. Can you post an example?

diogratia, 2019-06-10:
```vhdl
library ieee;
use ieee.std_logic_1164.all;

entity issue_836 is
end entity;

architecture foo of issue_836 is
    signal some_signal: std_logic;
begin
    process
    begin
        some_signal <= (others => '0');
        wait;
    end process;
end architecture;
```

```none
ghdl -a issue_836.vhdl
issue_836.vhdl:12:24:error: type enumeration subtype "std_logic" is not composite
ghdl:error: compilation error

ghdl --version
GHDL 0.36 (tarball) [Dunoon edition]
 Compiled with GNAT Version: GPL 2015 (20150428-49)
 llvm code generator
Written by Tristan Gingold.

Copyright (C) 2003 - 2019 Tristan Gingold.
GHDL is free software, covered by the GNU General Public License.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
```

Your problem can't be duplicated from your written description. Note the above code produces an error during analysis and not simulation.  The type of the aggregate is taken from context, here the signal assignment statement where semantic analysis reveals the target type is not a composite type and cannot be assigned an aggregate composite value.

Please provide a simple example replicating the problem and any command line ghdl invocations necessary to cause it. It can also helpful if you provide the ghdl version and any output as well as the crash report.

wm2015email, 2019-06-12:
**Just tried you example and it crashes on my computer...**

Here's the output:

C:\work> ghdl.exe -a --std=08 --ieee=synopsys --work=work ./issue_836.vhd
./issue_836.vhd:12:26: type enumeration subtype "std_logic" is not composite

******************** GHDL Bug occured ****************************
Please report this bug on https://github.com/tgingold/ghdl/issues
GHDL release: 0.35 (v0.35) [Dunoon edition]
Compiled with GNAT Version: 5.3.0
Target: i686-w64-mingw32
In directory: C:\Users\wpmoore\Documents\betty_phase1\fpga\tmp\
Command line:
ghdl.exe -a --std=08 --ieee=synopsys --work=work ./issue_836.vhd
Exception SYSTEM.ASSERTIONS.ASSERT_FAILURE raised
Exception information:
Exception name: SYSTEM.ASSERTIONS.ASSERT_FAILURE
Message: no field Enum_Pos
Call stack traceback locations:
0x643201 0x64343a 0x644009 0x64441c 0x64471c 0x66f8a0 0x675af8 0x530ba3 0x6351d3 0x63304d 0x4cc85b 0x4026e7 0x4013e0 0x769a8762 0x779c58eb 0x779c58bb
******************************************************************
<ERROR EXIT>

I'm using Windows 10

Window 10
Version: 1703
OS Build 15063.1805
Processor: Intel(R) Core(TM) i7=8850H CPU @ 2.6Ghz 2.59
Installed Ram: 16.0 GB
System Type: 64-bit operating system, x64-based processor

Here's the GHDL Version:

$ ghdl -v
GHDL 0.35 (v0.35) [Dunoon edition]
 Compiled with GNAT Version: 5.3.0
 mcode code generator
Written by Tristan Gingold.

Copyright (C) 2003 - 2015 Tristan Gingold.
GHDL is free software, covered by the GNU General Public License.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

1138-4EB, 2019-06-12:
@wm2015email, as @go2sh suggested, you are using a rather old version of GHDL (v0.35). Please, try a newer release: https://github.com/ghdl/ghdl/releases/tag/v0.36

closed 2019-06-16 by tgingold