ghdl/ghdl #841: Add a code of conduct

dominiksalvet, 2019-06-14:
**Is your feature request related to a problem? Please describe.**
The development of any software may lead to conflicts. It would be great to have a set of rules by which it could be independently decided what is too much and what are the consequences of that behaviour.

**Describe the solution you'd like**
Add a code of conduct or start a discussion on that topic.

**Describe alternatives you've considered**
To rely on people's common sense. :-)

**Additional context**
Personally, I thought that code of conducts are useless until recently when I figured out that if there is a chance for incoming people to behave better, it is worth it.

1138-4EB, 2019-06-14:
I am not sure about GHDL needing a CoC. This has been mostly a single-man project for almost 15 years now. I have only watched it for less than five years, but I don't remember any situation where such a procedure/mechanism/resource was actually needed. Overall, I agree with https://github.com/domgetter/NCoC for this case.

dominiksalvet, 2019-06-14:
@1138-4EB Well, that's certainly better than nothing for me. I suppose that much people do not pay attention to GitHub's [Community](https://github.com/ghdl/ghdl/community) tab but lately I found myself being quite obsessed with it. That's also the reason why I have created this issue anyway... :-)

If @tgingold agrees with that, I could add the code of conduct myself in a new pull request.

tgingold, 2019-06-15:
To be honest, I have never understood the purpose of a CoC.  For me they state the obvious: you should behave as an adult, you should be polite...

So I also agree with NCoC!

1138-4EB, 2019-06-15:
> I suppose that much people do not pay attention to GitHub's [Community](https://github.com/ghdl/ghdl/community) tab but lately I found myself being quite obsessed with it. That's also the reason why I have created this issue anyway... :-)

I do think that many of the 'Community suggestions' are useful and interesting. You can see that we added contributing guidelines, we had to [struggle](https://github.com/licensee/licensee/issues/247) to get the license properly identified, and we added bug/feature/pull request templates too. You might appreciate that the custom bug report template allows the user to easily test the bugs with multiple docker containers, so it is possible to know if the bug is present with different versions, backends, platforms. IMHO, I think we do our best to make it less painful for new users to get in touch and to be able to triage, reproduce and fix the issues ASAP.

Nevertheless, we need to evaluate the convenience of the 'Community suggestions', since those are general for all GitHub, where projects of very different sizes and contexts coexist. I think that this specific point (the CoC), is mostly useful in medium-large to very large projects, where lots of different people must coexist. I say 'must', because many of the projects that have adopted it have some people involved in them who are paid for it (by their employer, school, grant, etc.), and sometimes large companies with various interests take part. You might also see a difference between companies in Europe and in the US.

Fortunately (or unfortunately), this is not the case here. This is a really small community (around 6 contributors with more than 5 commits, and around 30 people in the chat), and the salary of none of us depends on how/what we contribute. We cannot afford to be stupid with each other, because there is no enough people to split this and make it smaller; this, our little TAZ, would disappear.

Hence, we just love being able to share knowledge with other people who are learned about VHDL, hardware design and netiquette. Although old-school, I think that netiquette, that from the 80s and 90s, is the *de facto* CoC here. I.e., we don't care about who you are, your gender, age, nationality, sexuality, family, money, etc. And yet, we respect you as much as we know and can. Note that only 1 of the 25 first contributors has an avatar image that might be an actual photo of the user. We have been talking to each other for years, and even if we lived in the same building, we probably wouldn't know. It's not what we say, but what we actualyl do.

---

> The development of any software may lead to conflicts.

Once again, this is not the case here. @tgingold is specially open to merging contributions from anyone and of any kind, as long as the technical contribution is clear and the code quality is good enough. He is also helpful with inexperienced users and he does last fixes himself, insteaf of requiring rebasing from the contributor.

Of course, we don't always agree on implementation details. I have been told 'no' multiple times. But none of them could I get angry, because either he was correct or we just didn't get to understand each other properly so we try it again some time later.

Note that I'm taking @tgingold as a reference because %99.999 of the codebase and the ownership of the project belongs to him.

> It would be great to have a set of rules by which it could be independently decided what is too much and what are the consequences of that behaviour.

I think that this is the main point here: **independently**. In this project there is no possibility to have any 'independent' group of people who could act as a jury in case of conflict. That's possible in large projects and large companies with lots of people, some of who are doing just that. Here, the same person decides which is the CoC, how to apply it, and what to do in case of conflict. This is a [*benevolent dictatorship*](https://en.wikipedia.org/wiki/Benevolent_dictator_for_life). It is so benevolent, though, that it sounds even offensive to use the term *dictatorship*.

---

Overall, I feel that NCoC is the 'modern' and 'polite' writing of http://www.catb.org/~esr/faqs/smart-questions.html.

@dominiksalvet, since you showed interest, I thought you deserved a better picture of why 'CoC' is it's actual form is not of special interest to this community. Yet, we agree on adding one, since some users (like you) might expect it. I hope it is clear now why we chose NCoC instead of other not-so-polite alternatives, such as, https://fuck-contributor-covenant.org/.

Still, if you don't agree, or you know any specific reason for us to consider some other alternative, please let us know. You might have some different experience/point of view.

dominiksalvet, 2019-06-15:
Well guys, thank you for your opinions. I have some comments though:

> For me they state the obvious: you should behave as an adult, you should be polite...

It is not obvious for everyone and not everyone knows what they state.

> I do think that many of the 'Community suggestions' are useful and interesting.

You did a great work to fulfil them! I am not sure if I have seen any other project to be so close to the maximum of the metrics... :rocket:

> we had to [struggle](https://github.com/licensee/licensee/issues/247) to get the license properly identified

Now, that's maybe the best place and time to mention it. It has been while when I found the [reuse](https://github.com/fsfe/reuse-tool) project. It manages and organises licenses and copyright. It is not only a guide/pracites but it is also a linter as well. Just type `reuse lint` in a repository and it will print unlicensed files. What's More, it is covered by [FSFE](https://fsfe.org/).

I use it in my every single software project here on GitHub and it is really great! Take a look at my [gim](https://github.com/dominiksalvet/gim) repository for instance. Reuse is compliant with [DEP-5](https://www.debian.org/doc/packaging-manuals/copyright-format/1.0/) licensing (see *debian/copyright* file). It uses an [SPDX](https://spdx.org/) license identifiers all across. You could also easily add multiple licenses into a repository. For example, take a look to [LICENSES](https://github.com/dominiksalvet/gim/tree/master/LICENSES) directory where is stored a license for used code of conduct.

I just wanted to let you know about this awesome (and underestimated) project. If you ever would think about complex yet simple solution for licensing, give it a chance.

> In this project there is no possibility to have any 'independent' group of people who could act as a jury in case of conflict.

That's true for most projects BUT this is also that problem that's got solved by including a code of conduct into the project. No one is independent as long as they do not have to proceed by a set of rules. I admit that a code of conduct does not cover any possible situation but I think it is better than nothing.

---

Well, that was my opinion but I also partially understand your point of view. And so I am preparing a PR with NCoC. If you change your mind or have any comments, let me know.

1138-4EB, 2019-06-15:
> It is not obvious for everyone and not everyone knows what they state.

I think we are getting to understand each other's point of view :D

IMHO, and from my very personal experience, if anyone needs a written CoC because otherwise the content is not obvious to them, i.e. he/she/it/put_any_pronoum_here doesn't understand what *people's common sense* means by *self, then that person is potentially problematic *per se*. This is because any CoC is just very very small subset of the *common sense*, so someone who *needs* literal descriptions might always find ways to (mis)interpret the writing. There is a saying in spanish, *wherever you go, do as/what you see*, which means that you should try to understand and try to integrate yourself when you visit some new place. I feel like CoC are mostly useful as a *weapon* for users who want the place to adapt to them, faster than the natural pace of that particular community.

At the same time, I also understand that *social context* might lead some people not to be confident enough about themselves. If having a CoC encourages these people to talk and participate, we are all with it. Should this be your main concern, please do never hesitate about bringing up your thoughts. Either in the context of the CoC, or any other enhancement to the [docs](https://ghdl.readthedocs.io).

> You did a great work to fulfil them! I am not sure if I have seen any other project to be so close to the maximum of the metrics... 🚀

Thanks! As said, we really like to use these features when we believe they add value to the project/community.

> That's true for most projects BUT this is also that problem that's got solved by including a code of conduct into the project. No one is independent as long as they do not have to proceed by a set [...]

As you see, we are already in a philosophical discussion. I'm perfectly ok with it. It's just that we should accept that there is no evident solution. But, for the sake of communication, let's exercise:

I understand that: you are assuming that having to proceed by a set of rules makes a decission independent, and, implicitly, that independence as a quality leads to fair decissions. As said above, I believe that an intelligent but not-well-intentioned person can make bad/unfair decissions on purpose, yet fulfilling a set of rules. Apart from that, we do fortunately have a context where non-independent but fair decissions are continuously being taken.

> Well, that was my opinion but I also partially understand your point of view. 

Much appreciated.

---

Regarding project [reuse](https://github.com/fsfe/reuse-tool), it looks really interesting, and it can be very useful for GHDL, since we've had some licensing/distribution conflicts in the past. However, on the one hand, I cannot have a better look at it ATM. It looks like it will need some time to fix the possible issues that might arise when using it. On the other hand, it is off-topic in this issue. Can you please cut this part of your previous comment and paste it in a new issue? I will then (re)move this part there, so we can keep the issues 'clean'.

dominiksalvet, 2019-06-15:
> IMHO, and from my very personal experience, if anyone needs a written CoC because otherwise the content is not obvious to them, i.e. he/she/it/put_any_pronoum_here doesn't understand what _people's [...]

That's a good point to think about. :thinking: 

> I understand that: you are assuming that having to proceed by a set of rules makes a decission independent, and, implicitly, that independence as a quality leads to fair decissions. As said above, I [...]

At the end of day, it is all about people. I think that there is nothing new I could say.

Guys, don't get me wrong... some time ago I was all against using code of conducts because "Well, we are not jerks, we are adults here...". I was even considering creating my own code of conduct -- "Don't be a jerk" with the same content...

But then I have started to think about it differently. What if "Well, we are not jerks, we are adults here..." is only my personal attitude? And guess what -- it seems so. We are all different a not every developer is polite to others. Paradoxically, I think that code of conduct is not as much for people who discuss it as for people who just don't care.

Now, I am really starting to philosophize yet I am going to state a really important point. One of the biggest problems on the Internet generally is close-to-zero social responsibility in my opinion. People think that they can do everything what they want to on the Internet without any consequences, as opposed to the real life. And code of conduct, for me in this case, is the way to say them "Not here."

But as I have written previously, I also understand your points. Anyway, forced code of conduct would be out of line by itself. What's more, I am new contributor to GHDL and you are here incomparably longer than me. So you surely know better what is good for the GHDL anyway.

1138-4EB, 2019-06-15:
> At the end of day, it is all about people. I think that there is nothing new I could say.
> 
> Paradoxically, I think that code of conduct is not as much for people who discuss it as for people who just don't care.

We agree. With the additional point that *we don't care* (as in *don't give s shit*) about people who *don't care* (as in *don't want*) about dicussing.

> People think that they can do everything what they want to on the Internet without any consequences, as opposed to the real life. And code of conduct, for me in this case, is the way to say them "Not here."

This is another spanish saying: *there is no greater contempt than not appreciating at all*, which is another wording for *don't feed the troll*.

> What's more, I am new contributor to GHDL and you are here incomparably longer than me. So you surely know better what is good for the GHDL anyway.

I think this is a kind of [argumentum ad verecundiam](https://en.wikipedia.org/wiki/Argument_from_authority). Just because I have been longer here, it does not mean I know better what is good for GHDL. Any of your/my opinions or arguments must hold by themselves, independently of who brings them to the dicussion. This is precisely the *essence* of *not caring* about age, religion, sex, money, experience, etc. As far as I see it, I can contribute to the dicussion with some knowledge about the context (which I tried to do early enough so you could be aware of it too). This is by no means more valuable than any other specific knowledge that you might bring.

closed 2019-06-16 by tgingold

dominiksalvet, 2019-06-16:
It looks like there is some information misinterpretation in this conversation.

> I think this is a kind of [argumentum ad verecundiam](https://en.wikipedia.org/wiki/Argument_from_authority). Just because I have been longer here, it does not mean I know better what is good for [...]

I do not want to argue with you but this is not the first time in this conversation I have feeling that you are trying to question my opinions. What you are maybe describing as my ignorance is indeed my opinion and from my point of view it is actually a considered opinion based on the current state of GHDL, which you have contributed to significantly. Also I objectively consider my experience with GHDL when replying to you.

I am not a philosopher, I am just trying to get work done. Our opinions are different and that's natural.

1138-4EB, 2019-06-16:
> I do not want to argue with you but this is not the first time in this conversation I have feeling that you are trying to question my opinions.

> What you are maybe describing as my ignorance is indeed my opinion and from my point of view it is actually a considered opinion

I'm quite sure that this is an unfortunate misunderstanding because I'm not a native english speaker. Precisely, I'm not describing your ignorance, but explicitly saying that you are not more ignorant than me; you just happen to have a different background. With the last sentence in the previous comment, I wanted to put a higher value on your opinion/knowledge than what I perceived from your wording. More precisely, the target audience was/is any other viewer that might read this in the future; not you or me explicitly.

Apart from that, questioning opinions (yours, mine, anyone's) is a mechanism of communication/discussion. It has nothing to do with *who* put the argument/opinion, but with *what* the content is. Moreover, questioning an opinion does not mean that I do personally not agree with it. It is always a good exercise to argument against one's own conviction.

So, please, accept my apologies if you felt undervalued during the conversation. That was/is very far from my purpose.

dominiksalvet, 2019-06-16:
@1138-4EB You know - I am not a native English speaker either - so it is also possible for me to incorrectly interpret some of your messages.

Thank you for your explanation. No apologies required. :smiley:

PR #842 (2019-06-16): Add a code of conduct
**Description**
I have added the [No Code Of Conduct](https://github.com/domgetter/NCoC) to the project. Resolves #841.

:rotating_light: Before submitting your PR, please read [contribute](http://ghdl.readthedocs.io/en/latest/contribute.html#fork-modify-and-pull-request) in the [Docs](http://ghdl.readthedocs.io/en/latest/index.html), and review the following checklist:

- [x] DO indicate which issues are either fixed or closed by this PR. See [GitHub Help: Closing issues using keywords](https://help.github.com/articles/closing-issues-via-commit-messages/).

**Further com [... cut]