ghdl/ghdl #843: License/copyright management - REUSE Practises

dominiksalvet, 2019-06-15:
This is an excerpt from my comment of another issue and it was suggested to me to create a new issue on this topic. So here is the text:

---

Now, that's maybe the best place and time to mention it. It has been while since I found the [reuse](https://github.com/fsfe/reuse-tool) project. It manages and organises licenses and copyright. It is not only a guide/pracites but it is also a linter as well. Just type `reuse lint` in a repository and it will print unlicensed files. What's more, it is covered by [FSFE](https://fsfe.org/).

I use it in my every single software project here on GitHub and it is really great! Take a look at my [gim](https://github.com/dominiksalvet/gim) repository for instance. Reuse is compliant with [DEP-5](https://www.debian.org/doc/packaging-manuals/copyright-format/1.0/) licensing (see [*debian/copyright*](https://github.com/dominiksalvet/gim/blob/master/debian/copyright) file). It uses an [SPDX](https://spdx.org/) license identifiers all across. You could also easily add multiple licenses into a repository. For example, take a look to [*LICENSES*](https://github.com/dominiksalvet/gim/tree/master/LICENSES) directory where is stored a license for used code of conduct.

I just wanted to let you know about this awesome (and underestimated) project. If you ever would think about complex yet simple solution for licensing, give it a chance.

---

If I could be of any help with adaptation of REUSE to GHDL, please, let me know. I would be more than happy. :smiley:

1138-4EB, 2019-06-15:
> NOTE: You can link this issue with the previous one by adding `#841`.

It looks really interesting, and it can be very useful for GHDL, since we've had some licensing/distribution conflicts in the past (precisely related to Debian). We do have multiple licenses in the repository, indeed.

Unfortunately, I cannot have a better look at it ATM. It looks like it will need some time to fix the possible issues that might arise when using it. Fortunately, it seems quite straighforward to try: `pip3 install --user fsfe-reuse`.

> If I could be of any help with adaptation of REUSE to GHDL, please, let me know. I would be more than happy. 😃

If you are willing to do so, give it a try! Just clone GHDL, install the fsfe-reuse package and run the linter on the repo. Having a specific list of things 'to do' or 'to fix' will really help us. From there on, you can tackle and cross out some of the items in the list.

Furthermore, we can add fsfe-reuse as an additional job in Travis-CI: https://github.com/ghdl/ghdl/blob/master/.travis.yml. We can either do it 'natively', or use an official [python container](https://hub.docker.com/_/python).

dominiksalvet, 2019-06-16:
@1138-4EB Ok, I will try to prepare some list but first I must warn you. REUSE works with license/copyright per each file. So to comply with REUSE, as far as I know it must be prepared license and copyright info for each file (literally each -- including, e.g., README.md), which could be impossible for some files. So this information probably will be a bit distorted in those cases.

1138-4EB, 2019-06-16:
> So to comply with REUSE, as far as I know it must be prepared license and copyright info for each file (literally each -- including, e.g., README.md), which could be impossible for some files. So this information probably will be a bit distorted in those cases

Yes, that's precisely why I said that *it looks like it will need some time to fix the possible issues that might arise*. I think that it would be great if there was some kind of 'ignore' or 'exclude' option, so we can filter by extension or location, to start with. Alternatively, we can start by running it in the `src` subdir only, so we avoid most of the doc and test sources.

Investigating if these options/possibilities exist can help us plan a gradual modification of the sources. If it is 'all or nothing', we'll have to wait until we can do it all.

> Also, I would like to state that I have zero experience with CI generally so I don't think I would be of any assistance here.

Thanks for stating it. Fortunately, this is no problem, because most of my contributions to GHDL are precisely related to the build scripts on GNU/Linux and the CI. So, I can help you with this.

You can start with creating a feature branch and replacing the `.travis.yml` file with the following:

``` yaml
os: linux
dist: xenial
language: python
python: 3.7
install: pip3 install fsfe-reuse
script:
  - reuse --version
  - reuse lint
```

Alternatively, you can copy/clone this feature branch form my fork: https://github.com/1138-4EB/ghdl/tree/feat-reuse

If you open a PR with this feature branch, or if you activate Travis for your fork of GHDL, reuse will be executed each time you push some commit(s) to GitHub. For example: https://travis-ci.org/1138-4EB/ghdl/builds/546466797

As you see in the log, reuse is properly installed and version `v0.3.4` is used. It lists a lot of the sources and at some point it fails with `The command "reuse lint" exited with 67`. So, unless `67` has some meaning which I am not aware about, there might be some bug in reuse. The first task in the list might be to investigate it.

dominiksalvet, 2019-06-16:
Well, for REUSE adaptation steps, I will be using my REUSE compliant repository [gim](https://github.com/dominiksalvet/gim) as a reference and I will trying to show examples on it. I hope it's not a problem but there are not many REUSE compliant projects (yet). Also, REUSE project is going through some quite significant changes now, so I hope nothing will break during the adaptation.

What I recommend to be done:
* [ ] Create **COPYING.md.license** with an [SPDX license identifier](https://spdx.org/licenses/) - [example](https://github.com/dominiksalvet/gim/blob/master/LICENSE.txt.license)
  * It is also possible to include the license identifier as a header to the license text or name the license file itself as the license identifier but GitHub may not recognise the license in those cases.
  * Also, I recommend that COPYING.md file should contain a license, which is used the most in the project as GitHub will show it as a representative license.
* [ ] Create **LICENSES** directory, which includes all other licenses used in this repository - [example](https://github.com/dominiksalvet/gim/tree/master/LICENSES)
  * In this case it's better to name each file containing a license as its identifier. If it is required to use a license that does not have an SPDX identifier, simple [example is here](https://github.com/dominiksalvet/uvod_do_vhdl/tree/master/LICENSES).
* [ ] Rewrite all **existing license/copyright notices** in individual files for REUSE to recognise them - [example](https://github.com/dominiksalvet/gim/blob/master/bin/gim)
  * It is required to include not only the copyright notice but also a license identifier as well. Either is missing -> `reuse lint` will print the name of the file.
  * If it is required to include multiple copyright notices (multiple copyright owners), this [simple example](https://github.com/dominiksalvet/gim/blob/master/debian/copyright) demonstrates it.
* [ ] **Add a license/copyright notice** to all source files without it - [example](https://github.com/dominiksalvet/gim/blob/master/bin/gim) (the same as the previous)
  * Now, this would be probably the hardest part. It is required to check each source file (not including documentation, MD files, ...) listed from `reuse lint` output and create a new copyright/license notice in those files. If you don't mind not including a copyright/license notice in each source file (and I am encouraging you against it), you can go to the next point and license source files as stated below.
* [ ] Create **debian/copyright** file and license everything else - [example](https://github.com/dominiksalvet/gim/blob/master/debian/copyright)
  * I personally find this file the most suitable for documentation-related files (MD files, images, ...) but in this particular case it may temporary serve as a license placeholder during the REUSE adaptation because it also supports a subset of regular expressions.

As I have mentioned, It looks like some changes will be implemented in a newer version of REUSE, so please consider it before starting to work on it. Namely, see [Usage section](https://github.com/fsfe/reuse-tool#usage) of its README file.

I am not a REUSE developer, so there might be something missing or incorrectly interpreted. In this case, let me know or ask me. Also, if you want to create a list better suited to your planning or needs, feel free to do so.

dominiksalvet, 2019-06-16:
One little note; maybe I should have stated it sooner.

> If I could be of any help with adaptation of REUSE to GHDL, please, let me know. I would be more than happy. :smiley:

I don't want to implement it myself but I am here to help if there will be any questions. I hope it is not a problem.

dominiksalvet, 2019-06-17:
> Thanks for stating it. Fortunately, this is no problem, because most of my contributions to GHDL are precisely related to the build scripts on GNU/Linux and the CI. So, I can help you with this.

I already have (briefly) learnt it for purposed of my [gim](https://github.com/dominiksalvet/gim) repository. Still don't know how to change python version while having `language: shell` though - `python: <version>` doesn't work in that case. Even Google didn't help me.

> So, unless `67` has some meaning which I am not aware about, there might be some bug in reuse. The first task in the list might be to investigate it.

It looks like the exit code represents number of unlicensed files. The question is how many files can it indicate. I think that it is 67 because you certainly must have more unlicensed files (i.e., missing copyright notice or license notice) than this number in the whole GHDL project.

You can test it by cloning my gim repository and modifying *debian/copyright* file by removing its items.

tgingold, 2019-06-18:
It is possible to select which files must have SPDX ?  I am OK with adding SPDX to files we have written, much less to imported files.

dominiksalvet, 2019-06-18:
@tgingold As far as I know, all files present in a repository must be licensed. That's generally not a problem as long as you can specify the author/s (i.e., copyright) and license of each file, which in case of imported libraries should be pretty easy as I suspect you wouldn't import libraries that you don't know authors and licenses of, would you? :D

Is this the answer to your question? I am not sure, maybe I misunderstand what you meant by SPDX.

1138-4EB, 2019-06-18:
> Also, I recommend that COPYING.md file should contain a license, which is used the most in the project as GitHub will show it as a representative license.

This is already the case, isn't it?

> Create LICENSES directory, which includes all other licenses used in this repository 

This is something we would need to do. It would be useful, though. We cannot have two valid license files at the root of the repo. ATM, the license for the documentation is at `doc/COPYING_DOC.md`. Other licenses are explained in https://ghdl.readthedocs.io/en/latest/licenses.html

> Rewrite all existing license/copyright notices in individual files for REUSE to recognise them.

So, as you see @dominiksalvet, it is not possible to modify some of the sources, even to add a SPDX identifier. Precisely, the sources of `numeric_*` *"may be distributed without change*". Note that these files are licensed; we fulfil that. But they have their own notice format which we cannot modify. I have read that IEEE is considering to distribute future versions of the libraries with a different license in order to avoid this kind of conflicts can be avoided in the future. But this is what we have for now. Hence, we need a different method which satisfies the requirements of 'reuse'.

> Add a license/copyright notice to all source files without it

This is probably fulfilled for all the sources in `src`. But I don't think it is true for the docs and the tests. Is there any 'exclude' mechanism in the reuse cli?

> Create debian/copyright file and license everything else - example
>
>    I personally find this file the most suitable for documentation-related files (MD files, images, ...) but in this particular [...]

I believe this would be the way to go. Not only as a temporary resource, but as a permanent method. We could use this to specify the license of the files that we cannot modify.

Do you know if this subdir can be directly used when packaging for Debian? If so, do you know the procedure? This is because some of the libraries that are distributed with GHDL are not free software/hardware. Some options in our build scripts allow to build tarballs without these libraries, it would be strange to have a `debian` subdir with some content that is not compatible with the [DFSG](https://en.wikipedia.org/wiki/Debian_Free_Software_Guidelines). We would probably end having two files: `debian/copyright` and `debian/copyrigth_dfsg` or `debian/copyrigth_nodfsg`.

> As I have mentioned, It looks like some changes will be implemented in a newer version of REUSE, so please consider it before starting to work on it. 

Let's hope they improve the UX slightly. It would be great to know why files do not pass, instead of just having a list of non-valid files.

> Still don't know how to change python version while having language: shell though - python: <version> doesn't work in that case. Even Google didn't help me.

NOTE: `language: shell` is not listed: https://docs.travis-ci.com/user/languages/. It is an alias of `generic` or `minimal`: https://docs.travis-ci.com/user/languages/minimal-and-generic/

Field 'language' has higher priority than 'python'. With 'language', you are selecting the main shell/terminal that you are going to use. With 'python' you just select the version of the language. So, if you set 'language: shell', field 'python' is just ignored. Indeed, Python should not even be installed/available in the shell/job: https://docs.travis-ci.com/user/languages/minimal-and-generic/#generic

If you want a shell, just because you want to do other things apart from running a Python script, you should set 'language: python' and use the `before_script` and `after_script` or `after_success` field. Alternatively, you can set 'language: shell' the install the specific version of Python that you want through `apt`. See: https://docs.travis-ci.com/user/languages/python/#using-tox-as-the-build-script. You can also use a Docker container which provides both a bash shell and a python shell: https://hub.docker.com/_/python

dominiksalvet, 2019-06-18:
> This is already the case, isn't it?

I think so but still I wanted to point out to demonstrate how to integrate the main license with GitHUb while organizing multiple licenses.

> So, as you see @dominiksalvet, it is not possible to modify some of the sources, even to add a SPDX identifier. Precisely, the sources of `numeric_*` _"may be distributed without change_".

No problem, as you pointed out later in your comment, one of the options is the *debian/copyright* file. Another option is to include copyright/license notice in a separate file. For each *filename* create *filename.license* with the notice and REUSE will recognise it.

> Is there any 'exclude' mechanism in the reuse cli?

I am not aware of any except for ignoring *debian/copyright* contents `--ignore-debian`.

> Do you know if this subdir can be directly used when packaging for Debian? If so, do you know the procedure?

I don't know much about it from my experience but I guess [it's possible](https://www.debian.org/doc/packaging-manuals/copyright-format/1.0/).

> Let's hope they improve the UX slightly. It would be great to know why files do not pass, instead of just having a list of non-valid files.

Frankly, I do not consider it a problem. Each file is included in the output as long as it does not contain copyright notice or license (license identifier or license text preceding by a header).

> Indeed, Python should not even be installed/available in the shell/job: https://docs.travis-ci.com/user/languages/minimal-and-generic/#generic

That's not true as far as I know - 'The generic image contains everything from minimal, and also the usual databases, services and language runtimes:' - and minimal includes Python.

@1138-4EB Nevertheless, so far I have everything I wanted from Travis CI. This is my [configuration](https://github.com/dominiksalvet/gim/blob/master/.travis.yml). You also helped me, so thank you. :+1:

1138-4EB, 2019-06-18:
> Another option is to include copyright/license notice in a separate file. For each _filename_ create _filename.license_ with the notice and REUSE will recognise it.

I believe this is ok too, at least for some specific files.

> Frankly, I do not consider it a problem. Each file is included in the output as long as it does not contain copyright notice or license (license identifier or license text preceding by a header).

I believe that the conditions that make the file to be included are more than two:

- There is neither a notice nor a file.
- There is no notice, and there is a valid file which contains:
  - Unrecognized content.
  - Valid field name, but the value is not valid/known.
  - Valid field name, but it is duplicated and the contents match.
  - Valid field name, but it is duplicated and the contents do not match.
- There is a notice but it is not valid because the SPDX has any of the problems listed in the previous point.
- ...

Not having any feedback about it can be problematic. For example:

- Reuse lists a file as not being compilant.
- Without opening it, I add a `*.license` file specifying that the license is X.
- Reuse accepts it and passes.
- In a few months someone opens the file and finds out that there is a copyright notice in it, which does not match the `*.license` file, but license Y.
- What should we do? Which is the 'real' license? Is it an unwantedly dual-licensed source?

Overall, I feel like the tool in it's current state does not avoid a user having to manually open and read the header of each file. It can be very useful for projects that start from zero, tho, as in this case the tool lets you know how much work you have ToDo.

Nevertheless, I believe this is because it is in a very early state of development yet.

> That's not true as far as I know - 'The generic image contains everything from minimal, and also the usual databases, services and language runtimes:' - and minimal includes Python.

You are correct. I checked the 'generic' section, where other languages are listed, but I missed the section about 'minimal'.
 
> @1138-4EB Nevertheless, so far I have everything I wanted from Travis CI. This is my [configuration](https://github.com/dominiksalvet/gim/blob/master/.travis.yml) [](/dominiksalvet/gim/blob/HEAD@{2019-06-18T09:42:12Z}/.travis.yml). You also helped me, so thank you. 👍

You are welcome.

The file looks good and clean. You could problably merge the last two items in the installation jobs: `echo "$(sudo gim status github.com/dominiksalvet/gim)" | grep -q -v 'nothing'` or `sudo gim status github.com/dominiksalvet/gim | grep -q -v 'nothing'`.

dominiksalvet, 2019-06-18:
> Not having any feedback about it can be problematic. For example:
>
> * Reuse lists a file as not being compilant.
> * Without opening it, I add a `*.license` file specifying that the license is X.
> * [...]

I believe in some similar cases there are also warnings. But it certainly does not cover all possible problems right now, e.g., as you described, dual licensing.

> Nevertheless, I believe this is because it is in a very early state of development yet.

I have always used REUSE from an early state of development, so I don't have any experience with transforming a project (of GHDL size) to REUSE licensing. :(

> The file looks good and clean. You could problably merge the last two items in the installation jobs: `echo "$(sudo gim status github.com/dominiksalvet/gim)" | grep -q -v 'nothing'` or `sudo gim status github.com/dominiksalvet/gim | grep -q -v 'nothing'`.

This is intentional; `gim status` can fail and pipe wouldn't catch it.

dominiksalvet, 2020-04-28:
Hello guys, hope you are doing fine! :wave: Just a quick update from my side to this topic...

I am no longer using [reuse](https://github.com/fsfe/reuse-tool) in my repositories. The reason is simple. Although its implementation of per-file licenses is decent and straightforward, I have come to the conclusion that the whole per-file licenses mechanism is overwhelming for me in the first place. I have simplified a lot of things in my repositories since last time and this was a part of that proces. Now, do not get me wrong. Reuse was and probably is great at doing what it does but I just don't need it - that's all. :smile: 

Nevertheless, it was reuse which helped me to establish my current copyright+URL header notices to each source file as seen for example in [GitPack](https://github.com/dominiksalvet/gitpack/blob/master/src/gitpack), the successor to [gim](https://github.com/dominiksalvet/gim). Now people immediately know who is the copyright owner (and the license provider as well) and where to go for further details. That really makes sense to me and it is very simple and flexible as well. So reuse was definitely beneficial for me in this regard! :rocket:

If you are not interested to incorporate reuse licensing into GHDL anymore, feel free to close this issue. If you are considering reuse licensing, I will not be able to help with it unfortunately as I have not been using it for some time.

tgingold, 2020-04-29:
Ok, I close this issue.  We could reopen it if someone wants to implement REUSE practices.

closed 2020-04-29 by tgingold