ghdl/ghdl #848: Enhancements to python/libghdl

1138-4EB, 2019-06-17:
This issue is to keep track of the enhancements to python/libghdl, as commented in (https://github.com/ghdl/ghdl/pull/840#issuecomment-502487771).

- [ ] Linking libghdl fails in Mac OS because of `ld: library not found for -lgnat-2017`.
- [ ] `python/libghdl` might need a dummy `conf.py`.
- [ ] Make `conf.py` contain `__version__` only.
- [ ] Other mechanism to find where GHDL is it installed.
- [ ] dlopen/LoadLibrary.

tgingold, 2019-06-18:
+ moving setup.py to python/ if needed.  I have to investigate --upgrade.

1138-4EB, 2019-06-18:
> * moving setup.py to python/ if needed.  I have to investigate --upgrade.

Maybe @Paebbels or @kraigher, who are more proficient with Python, can help us understand: https://github.com/ghdl/ghdl-language-server/commit/de0c58f0cc2458d32c0b940648512ba93815c009

tgingold, 2019-06-20:
I have moved setup.py in python/, changed the method to find the standard libraries and made a standalone Makefile rule to build the py tar.  I plan  to automatically create the config.py file.

1138-4EB, 2019-06-20:
> I have moved setup.py in python/, changed the method to find the standard libraries and made a standalone Makefile rule to build the py tar.

This might be interpreted as `libghdl`, `pnodes` and `xtools` being three items/modules of the same package. I.e.:

```
# Before
curl -fsSL https://codeload.github.com/ghdl/ghdl/tar.gz/master | tar xzf - --strip-components=2 ghdl-master/python/libghdl
pip3 install .

# Now
curl -fsSL https://codeload.github.com/ghdl/ghdl/tar.gz/master | tar xzf - --strip-components=1 ghdl-master/python
# optionally rm -rf pnodes xtools
pip3 install .
```

Is this expected?

> I plan to automatically create the config.py file.

Do you mean as an additional makefile rule instead of doing it in `configure`?

I'm thinking that an alternative would be to define the current version in `config.py` and make `configure` read it (just need to `sed` the content between `'`). This would allow us to have a `config.py` checked in the SCM (git), so the package is directly downloadable/installable. It is always possible to add a `--version` option to configure, in order to override the value.

Related to this, does it make sense to have [`version.tmp`](https://github.com/ghdl/ghdl/blob/master/Makefile.in#L159-L167) as a makefile rule, instead of doing it in `configure` as an additional item in `subst_files` and/or merging the logic with the previous paragraph?

tgingold, 2019-06-21:
Only libghdl is installed by setup.py.  If this is confusing, we could add a new directory.  I was never able to use 'pip install -e' with setup.py within libghdl.

Also the version could be stored in a config.py, this is not neutral.  If one day we have a binding in Lua (for example), what will we do ?  On the other hand, we could check-in config.py

Yes, it make sense to have `version.tmp` so that the version changes in case of local modification.

1138-4EB, 2019-06-22:
> Only libghdl is installed by setup.py. If this is confusing, we could add a new directory. I was never able to use 'pip install -e' with setup.py within libghdl.

The point is that, if we add a subdirectory, I think we would be in the same context as the language-server repo. And that didn't work for you neither.
 
> Also the version could be stored in a config.py, this is not neutral. If one day we have a binding in Lua (for example), what will we do ? On the other hand, we could check-in config.py

As commented, it is possible to add a `--version` option to `configure`, in order to overwrite the version in `config.py`, the Lua config file, any other place.

With the current procedure, `config.py` needs not to be modified at all, unless a new release is published. Therefore, it seems unnecessary to generate it in each build, as we do now. I think that any mechanism that avoids this will be ok, be it in the configure script, in the makefile, etc.
 
> Yes, it make sense to have `version.tmp` so that the version changes in case of local modification.

I didn't mean to remove the rule that generates the `version.tmp`, but to move the logic inside it. Currently, `ghdl_version` is defined in the configure script, but then in the makefile rule git is used to extract the hash. I think that both values should be defined in the same place. This would allow to make configure update `config.py`, the Lua config file, any other, etc. using both the version and the hash.

tgingold, 2019-06-22:
Generating config.py when the version is changed is an option I am considering.

But we cannot generate version.tmp by configure, as the version is changed if the sources are modified.  But configure is run once.

tgingold, 2019-07-10:
I am closing this issue.  I haven't seen any particular problem with mac os x.

closed 2019-07-10 by tgingold

1138-4EB, 2019-07-10:
> I am closing this issue. I haven't seen any particular problem with mac os x.

Did you actually try it 'manually' on Mac OS? If you are relying on Travis for this, note that I explicitly disabled libghdl on the mac job: https://github.com/ghdl/ghdl/blob/master/dist/travis/travis-ci.sh#L61

This is the result before I applied that fix: https://travis-ci.org/ghdl/ghdl/builds/546502695

This is the current result (after I applied the fix): https://travis-ci.org/ghdl/ghdl/builds/546516197

tgingold, 2019-07-11:
Still failing.  I suspect there is something wrong with the gnat compiler in the image.  Is it worth investigating ?

1138-4EB, 2019-07-11:
I thought that you would be able to identify the issue quickly. If it needs more time, it is up to you. I believe that macOS users will be interested on having libghdl (and ghdl-ls). But, since those are experimental features yet, we can wait.

Let's keep this issue closed and work on #867. If we get it to pass at some point, we can merge it. Otherwise, we will keep it disabled on macOS.