ghdl/ghdl #850: Add support for LLVM 8

1138-4EB, 2019-06-18:
As read in https://aur.archlinux.org/packages/ghdl-llvm-git/, LLVM is not supported yet:

- https://github.com/ghdl/ghdl/blob/master/configure#L254
- https://github.com/ghdl/ghdl/blob/master/dist/travis/build.sh#L115

closed 2019-06-22 by tgingold

1138-4EB, 2019-06-22:
@tgingold, did you check whether `build.sh` needs to be fixed accordingly?

tgingold, 2019-06-22:
Indeed, it should.

1138-4EB, 2019-06-22:
I just checked it, and I think that no change is necessary. On Ubuntu 19.04 LLVM 8 is used, but both `llvm-config` and `clang++` exist. Hence, I think it's all ok already.

Actually, the same happens on Ubuntu 18.04 with LLVM 6, on Debian Stretch with LLVM 3.8 and with Debian Buster and LLVM 7.

Therefore, we can probably remove https://github.com/ghdl/ghdl/blob/master/dist/travis/build.sh#L95-L118. In the end, we are testing each version in a different container, so there is no need to select the version explicitly in the build script. What do you think?

commit 7328eae0 (2019-06-22): Add support for llvm-8.  Close #850